Koenkk/Z-Stack-firmware #204: Request: Support for bin-update via USB for ITEAD pre-flashed controller

stenleo, 2020-08-31:
Hi!
I bought a couple of pre-flashed ITEAD controllers which work great. Only issue is that the flash is rather old and I have some devices only working with newer version.
The controllers have bootload functionality and should be possible to update via USB only. Following a guide for it it mentions something about changing a setting to MAKE_CRC_SHDW to make it work. 

Tried flashing the device with the supplied bin file. It seemed to work ok but the controller failed to start later. I am guessing it has something to do with the setting above.

I tried to get the environment set up to see if I could compile with the change above but I didn't succeed. I am also not used to working in this type of development.

Would it be possible to supply a bin together with the project that works for flashing the ITEAD via USB out of the box?

puddly, 2020-09-09:
I implemented this functionality in [zigpy-znp](https://github.com/zha-ng/zigpy-znp/#firmware) and have been able to successfully flash both Z-Stack 3.0.1 and Z-Stack Home 1.2 `.bin` files via bootloader. You may want to reset NVRAM and clear the device state after doing so (the `-r` option) because this isn't done by default when upgrading the firmware.

stenleo, 2020-09-29:
Thanks puddly!
I did see that and I have just tried to make it work with zigpy, though I get lost to be honest. I am running Home Assistant supervised and can't figure out how to get that to work. It says on the zigpy page that zigpy-znp should come installed with HA? Can't seem to find any way to run the py commands in that environment to find out though..

stenleo, 2020-09-29:
> I implemented this functionality in [zigpy-znp](https://github.com/zha-ng/zigpy-znp/#firmware) and have been able to successfully flash both Z-Stack 3.0.1 and Z-Stack Home 1.2 `.bin` files via the bootloader. You may want to reset NVRAM and clear the device state after doing so (the `-r` option) because this isn't done by default when upgrading the firmware.

Actually, ignore my other comment. I managed to set zigpy_znp up on another raspberry and can run the scripts. I am able to do the flash_read with the 20190608 default firware for CC2531, it goes through all percentages and finishes. However, when I try flash write I get  "Firmware CRC is incorrect. Expected 0x2010, got 0x69CC. 
I took the official z-stack firmware, is it a bad version?

puddly, 2020-09-29:
Are you trying to write back the same firmware that you read? Or a fresh .bin file (without a SBL)?

stenleo, 2020-09-29:
My bad, I did indeed try to write back the same firmware. I redid it, skipping the read part and it worked fine!
I did not find a way to reset NVRAM like you wrote above, couldn't see how I should use the -r option. Does seem to start up anyway though, is it really needed?

puddly, 2020-09-29:
If it works then it works. All the option does is delete the two NVIDs that Z2M/zigpy-znp use to keep track of whether or not the network has been formed, which would cause it to be re-formed.

stenleo, 2020-09-29:
Yep, working fine in HA with zigbee2mqtt integration. Thanks for the help!

closed 2020-11-06 by github-actions[bot]