Koenkk/Z-Stack-firmware #206: CC2652P firmware modification request

mercenaruss, 2020-09-15:
Ebyte provided new board with CC2652P chip - E72-2G4M20S1E ,but firmware is for Launch.
http://www.ebyte.com/en/product-view-news.aspx?id=1002

According to datasheet is another truth table and another DIO used.
[Capture](https://user-images.githubusercontent.com/5508130/93263743-1b1cff00-f79e-11ea-874f-f2f95cb27224.PNG)
Please assist with a firmware where this is implemented.
Russian Community will be thankful.

mercenaruss, 2020-09-16:
Following my request , i used Code Composer to try to build myself.
And when changing everything i get next error:
Please see function 'rfDriverCallbackAntennaSwitching' in 'ti_drivers_config.c'. The antenna switching functionality must be implemented by the user.
Unfortunately i don't have enough knowledge to make this even with a code snippet.

    /* ======== PLEASE READ THIS ========
    *
    * This function is declared weak for the application to override it,
    * and the undefined function call below is intended to catch the
    * developer's attention at link time.
    * A new definition of 'rfDriverCallbackAntennaSwitching' is required.
    *
    * Please copy this function definition to create your own, but make
    * sure to remove '__attribute__((weak))' for your definition and
    * remove the below function declaration and function call.
    * 
    * To handle the events listed in '.globalEventMask', please see the
    * help text provided in 'rfDriverCallback' above. 
    * 
    * For an example on how to interact with the selected antenna pins,
    * please see the code snippet below:
    * 
    *   --- Code snippet begin ---
    * 
    * // > Initialize pins as output
    * static PIN_Handle antennaPins;
    * static PIN_State antennaState;
    * 
    * PIN_Config antennaConfig[] = {
    *   CONFIG_RF_24GHZ | PIN_GPIO_OUTPUT_EN | PIN_GPIO_LOW | PIN_PUSHPULL | PIN_DRVSTR_MAX,
    *   CONFIG_RF_HIGH_PA | PIN_GPIO_OUTPUT_EN | PIN_GPIO_LOW | PIN_PUSHPULL | PIN_DRVSTR_MAX,
    *   PIN_TERMINATE
    * };
    * antennaPins = PIN_open(&antennaState, antennaConfig);
    *
    * // > Set pin output value manually
    * PINCC26XX_setOutputValue(CONFIG_RF_24GHZ, 0); // Low
    * PINCC26XX_setOutputValue(CONFIG_RF_24GHZ, 1); // High
    * 
    * // > Mux pin to be driven/controlled by the RF Core
    * // > (RFC GPIO0 is here only used as an example)
    * PINCC26XX_setMux(antennaPins, CONFIG_RF_24GHZ, PINCC26XX_MUX_RFC_GPO0);
    * 
    * // > Mux pin to be controlled manually (i.e. release RF Core control)
    * PINCC26XX_setMux(antennaPins, CONFIG_RF_24GHZ, PINCC26XX_MUX_GPIO);
    * 
    *   --- Code snippet end ---

Koenkk, 2020-09-16:
I would suggest opening a question here: https://e2e.ti.com/support/wireless-connectivity/zigbee-and-thread/f/158 . There is much more knowledge there.

closed 2020-09-18 by mercenaruss