Koenkk/Z-Stack-firmware #210: Hardware Discussion: Standardisation of BSL and RF Switch pins for CC1352/CC2652 based designs

omerk, 2020-09-26:
With the number of open-source hardware boards increasing around the newer TI chips, I think the time has come to agree on and standardise the BSL and RF Switch control pins (for -P parts) as otherwise @Koenkk's firmware repository is going to end up with very many variants of the same thing that is undoubtedly going to confuse new users trying to pick the right firmware for their boards.

This is a summary of the current situation and a proposal to try and lock these down for the future to reduce firmware support complexity going forward.

## BSL 

The beauty of BSL (the ROM serial bootloader) is that unless it is disabled in CCFG, it allows you to burn new firmware to your devices over serial without needing an external programmer. One caveat: Given than an external pin is involved in this special boot mode, you can accidentally "disable" BSL by picking a firmware image that moves the BSL trigger to a different physical pin than what your board might have.

Unfortunately TI documentation is confusing and has errors (see [here](https://e2e.ti.com/support/wireless-connectivity/bluetooth/f/538/t/871984?LAUNCHXL-CC26X2R1-Bootloader-backdoor-enable-pin) and [here](https://e2e.ti.com/support/wireless-connectivity/other-wireless/f/667/p/940094/3472312#3472312)).

Looking at the boards in the wild and ones I am currently working on as @electrolama projects, here's a summary of the current situation:

On CC2652R/RB, `DIO_13` is the BSL backdoor enable pin used:

  - TI LAUNCHXL-CC26X2R1 (BTN1)
  - @electrolama zzh
  - @slaesh cc2652rb usb dev stick
  - (anything else, please comment below)

On CC1352P/CC2652P, this pin appears to be `DIO_15`:
  - Launchpad LAUNCHXL-CC1352P-2 (BTN1)
  - @electrolama zoe2, zzhp and zzhp-lc
  - @mercenaruss Zigbeer E79 Stick
  - (anything else, please comment below)

**Proposal:** Let's lock this down and use `DIO_13` for CC2652R/RB and `DIO_15` for CC1352P/CC2652P as the BSL backdoor enable pin for any hardware that is aiming to make use of this repository for firmware support.

### "Auto-BSL"

On boards with USB-UART adapters that expose the extra control signals, following arrangement is used on some of the boards noted above:

  - Computer side DTR - WMCU BSL pin
  - Computer side RTS - WMCU RESET

Purpose of this is to get the device into BSL mode by cleverly toggling these pins in a special sequence so that automatic firmware updates could be done with, say, zigbee2mqtt without the need to unplug the coordinator board used.

**Proposal:** Let's lock this down and use the DTR-BSL/RTS-RESET mapping for any hardware that is aiming to make use of this repository for firmware support.

## RF Switch pins for -P devices

This is a complex issue without an easy answer/proposal.

On -P devices, RF switches are used to switch between different RF paths. See [this](https://e2e.ti.com/support/wireless-connectivity/zigbee-and-thread/f/158/t/894423) and [this](https://e2e.ti.com/support/wireless-connectivity/bluetooth/f/538/t/882838?tisearch=e2e-sitesearch&keymatch=cc2652p) for the nitty gritty.

For example, this is how LAUNCHXL-CC1352P-2 deals with it:

[image]
[image]

Popular module maker Ebyte has two modules, E72 (CC2652P):

[image]

...and E79 (CC1352P):

[image]

On @electrolama zzhp (CC2652P):

[image]

So already we have a variety of pins being used for RF Switch control with some clashes. For instance, zzhp `DIO_6` is the inverse of the Ebyte switch control (from what I can gather, I am waiting on a shipment of Ebyte modules to verify this). If that is indeed the case, I am happy to modify zzhp design so that the control signal matches the behaviour of Ebyte modules (so they can share the same firmware).

Generalising this is not going to work as nicely as the BSL proposal, for now at least for the @electrolama boards I am working on I am going to mimic Ebyte modules as they appear to be used by other designs as well.

Perhaps we can solve this by having a Launchpad firmware and "everything else" that utilises `DIO_5` and `DIO_6`, @Koenkk?

## Comments and additions welcome!

Please tag/add/mention any hardware developers/projects that might be working on similar hardware.

mercenaruss, 2020-09-26:
Following @omerk request we agree with him,many DYI devices now based on Ebyte modules,what are very cheap and easy to get:
E79-400DM2005S(CC1352P),E79-900DM2005S(CC1352P),E72-2G4M20S1E(CC2652P), all of them have different Truth Table(@omerk show them upper) what is obviously different from standard LAUNCHPAD.
@egony and @palko are developing/testing some hardware based on this modules from EByte, firmware is giving us headache to adjust for this modules.
We all agree with @omerk that is required to standardize and make a firmware for this modules.

BSL - DIO_15

> Perhaps we can solve this by having a Launchpad firmware and "everything else" that utilises DIO_5 and DIO_6, @Koenkk?
We agree with this and think is the best solution for DYI devices and Sticks based on EByte modules.

> DIO_7 is a common LED pin on the open-source boards, active high (Pin state 1 or high: LED On). Perhaps it can be the standard LED for features like #202 ?

About this i think is better to have 2 LED on DIO_7 and DIO_8

mercenaruss, 2020-10-11:
@omerk better to make a full list of all pins/DIO's used for our hardware.

omerk, 2020-11-07:
Here's a first stab at the list of adapters and hardware details:

<table>
<thead>
  <tr>
    <th>Adapter</th>
    <th>TI Chip/Module Used</th>
    <th>Firmware to Flash</th>
    <th>BSL Trigger Pin (1)</th>
    <th>Auto-BSL (2)</th>
    <th>RF Switch Control Pins (3)</th>
    <th>LED(s)</th>
  </tr>
</thead>
<tbody>
  <tr>
    <td>TI LAUNCHXL-CC26xR1</td>
    <td>CC2652R</td>
    <td>CC2652R<br></td>
    <td>DIO_13</td>
    <td>No</td>
    <td>N/A</td>
    <td>DIO_6 (Red)<br>DIO_7 (Green)<br></td>
  </tr>
  <tr>
    <td>TI LAUNCHXL-CC1352P1</td>
    <td>CC1352P</td>
    <td>CC1352P_CC2652P_LAUNCHPAD</td>
    <td>DIO_15</td>
    <td>No</td>
    <td>DIO_28: 2.4Ghz<br>DIO_29: 20dBm PA<br>DIO_30: Sub-1GHz</td>
    <td>DIO_6 (Red)<br>DIO_7 (Green)<br></td>
  </tr>
  <tr>
    <td>TI LAUNCHXL-CC1352P-2</td>
    <td>CC1352P</td>
    <td>CC1352P_CC2652P_LAUNCHPAD</td>
    <td>DIO_15</td>
    <td>No</td>
    <td>DIO_28: 2.4Ghz<br>DIO_29: 20dBm PA<br>DIO_30: Sub-1GHz</td>
    <td>DIO_6 (Red)<br>DIO_7 (Green)<br></td>
  </tr>
  <tr>
    <td>TI LAUNCHXL-CC1352P-4</td>
    <td>CC1352P</td>
    <td>CC1352P_CC2652P_LAUNCHPAD</td>
    <td>DIO_15</td>
    <td>No</td>
    <td>DIO_28: 2.4Ghz<br>DIO_29: 20dBm PA<br>DIO_30: Sub-1GHz</td>
    <td>DIO_6 (Red)<br>DIO_7 (Green)<br></td>
  </tr>
  <tr>
    <td>Electrolama zzh</td>
    <td>CC2652R</td>
    <td>CC2652R</td>
    <td>DIO_13</td>
    <td>No</td>
    <td>N/A</td>
    <td>DIO_7 (Pink)</td>
  </tr>
  <tr>
    <td>Electrolama zzhp-lite</td>
    <td>CC2652P<br>(Ebyte E72)<br></td>
    <td>CC1352P_CC2652P_DIO5-6</td>
    <td>DIO_15</td>
    <td>Yes</td>
    <td>DIO_5: 20dBm PA ??<br>DIO_6: 2.4GHz ??<br></td>
    <td>DIO_7 (Pink)</td>
  </tr>
  <tr>
    <td>Electrolama zzhp</td>
    <td>CC2652P</td>
    <td>CC1352P_CC2652P_DIO5-6</td>
    <td>DIO_15</td>
    <td>Yes</td>
    <td>DIO_5: 20dBm PA ??<br>DIO_6: 2.4GHz ??</td>
    <td>DIO_7 (Pink)</td>
  </tr>
  <tr>
    <td>Electrolama zoe2</td>
    <td>CC1352P<br>(Ebyte E79)<br></td>
    <td>CC1352P_CC2652P_DIO5-6</td>
    <td>DIO_15</td>
    <td>No</td>
    <td>DIO_5: 20dBm PA ??<br>DIO_6: 2.4GHz ??</td>
    <td>DIO_7 (Pink)</td>
  </tr>
  <tr>
    <td>slaesh's CC2652RB stick</td>
    <td>CC2652RB</td>
    <td>CC2652RB</td>
    <td>DIO_13</td>
    <td>Yes</td>
    <td>N/A</td>
    <td>DIO_7</td>
  </tr>
  <tr>
    <td>mercenaruss ZigStar<br></td>
    <td>CC2652P<br>(RFStar ?)<br></td>
    <td>CC1352P_CC2652P_DIO5-6?</td>
    <td>DIO_15</td>
    <td>No?</td>
    <td>?</td>
    <td>DIO_7<br>DIO_8<br></td>
  </tr>
  <tr>
    <td>mercenaruss Zigbeer E79 stick</td>
    <td>CC1352P<br>(Ebyte E79)</td>
    <td>CC1352P_CC2652P_DIO5-6</td>
    <td>DIO_15</td>
    <td>No?</td>
    <td>DIO_5: 20dBm PA ??<br>DIO_6: 2.4GHz ??</td>
    <td>DIO_7<br>DIO_8<br></td>
  </tr>
</tbody>
</table>

Note 1: Pull this pin down on power up to trigger BSL mode
Note 2: Toggle RTS (RESET) and DTR (BSL Trigger) to put the device in BSL mode
Note 3: Pin mentioned high, all others low

@mercenaruss Can you please have a look and add missing details for your adapters? Also, ISTR you mentioning that you had some success with the firmware for Ebyte modules, can you add a bit more info here? Thanks!

/cc: @Koenkk @sjorge

mercenaruss, 2020-11-07:
Name: ZigStar
Chip: CC2652P
Model: RFSTAR RF-BM-2652P2
BSL Trigger Pin: DIO_15
Auto-BSL: Not in Main Gerber, but will be available for CH340C version
RF Switch Control Pins: DIO_28: 2.4Ghz , DIO_29: 20dBm PA
LED(s): DIO_7(Green), DIO_8(Red)

Koenkk, 2020-11-10:
Added new firmwares: https://github.com/Koenkk/Z-Stack-firmware/tree/develop/coordinator/Z-Stack_3.x.0/bin

mercenaruss, 2020-11-13:
@Koenkk I think existing firmware CC1352P2_CC2652P_launchpad_20201103.zip is not suitable for sticks/coordinators where amplifier is required. In my personal opinion I think is required 2 firmware's:
1. CC1352P2_CC2652P_launchpad - where amplifier is not enabled
2. CC1352P2_CC2652P_launchpad_amplifier - with 20 dbm amplifier

or is possible i do a mistake here.

Koenkk, 2020-11-13:
@mercenaruss what would be the differences between these firmwares?

mercenaruss, 2020-11-13:
> @mercenaruss what would be the differences between these firmwares?

Enabled amplifier???
P.S. Maybe i dont understand something

Koenkk, 2020-11-13:
@mercenaruss I've just published new firmwares: https://github.com/Koenkk/Z-Stack-firmware/tree/develop/coordinator/Z-Stack_3.x.0/bin

Defuault tx power is 20dbm now (up from 0dbm). It can be lowered by adding the following to the `configuration.yaml`.

```
experimental:
  transmit_power: 5
```

mercenaruss, 2020-11-13:
@Koenkk a big thank you!!! Testing is ongoing!

closed 2020-12-22 by github-actions[bot]

sjorge, 2020-12-22:
Small update, the zzhp-lite compatible firmware seems to be working fine.

Need to give it a good stress test but ran into some unrelated issues. Hope to look at it soon.