Koenkk/Z-Stack-firmware #211: CC2538 as router.

1igeza, 2020-09-28:
Does anybody use CC2538 as router? I'm using [https://github.com/jethome-ru/zigbee-firmware/tree/master/router/cc2538_cc2592](url) firmware, it seems to work ok, but I can't get routing table. When I sniff traffic I see:
```
ZigBee Device Profile, Routing Table Response, Status: Not Supported
    Sequence Number: 65
    Status: Not Supported (132)
    Table Size: 0
    Index: 0
    Table Count: 0
```
Is there a way to get routing table? Will be CC2538 with router firmware added to zigbee2mqtt?

Xsev1, 2020-10-18:
I would like to use an second cc2538 as router too. I tested 2530 and 2531 (coordinator and router) but the CC2538 is much much better, so a router with this hardware would be great.

@Koenkk  can you say please, is there in near future a routing firmware in planing for this zigbee modul?
Best regards

Koenkk, 2020-10-18:
@Xsev1 I don't maintain the CC2538 firmware.

@kirovilya I've seen another cc2538 repo firmware popup (forgot how it was named). Would it be better to point to that and remove the CC2538 firmwares from this repo.

Hacker-CB, 2020-10-18:
@Koenkk its our repo. (https://github.com/jethome-ru/zigbee-firmware)
We are maintain alternative corrdinator f/w for 2538. Also we made router f/w for 2538.
l think that we will make f/w for 2562 later.

closed 2020-11-25 by github-actions[bot]