Koenkk/Z-Stack-firmware #216: Z-Stack 3.0.2 Issues and fixes.

mdeweerd, 2020-10-09:
As far as I can see the know issues and fixes for the Z-Stack 3.0.2 are not integrated in the binaries proposed in this repository?

https://e2e.ti.com/support/wireless-connectivity/zigbee-and-thread/f/158/t/902631

Koenkk, 2020-10-09:
I've checked and only the points related to the `MAXMEMHEAP` for CC2531/CC2530 are integrated.

mdeweerd, 2020-10-10:
Ok, I'll have to make a build as I seem to have issue number 8.

mdeweerd, 2020-10-10:
I reviewed the patches (firmware_CC2531_CC2530.patch)

The following patch is not commented.

https://github.com/Koenkk/Z-Stack-firmware/blob/d0e58b6132e7363a933a51d027f208450abc6b99/coordinator/Z-Stack_3.0.x/firmware_CC2531_CC2530.patch#L126

Il also suggest that such specific changes are subject to the existance of a "#define" such as "ZIGBEE_MQTT" or something like that.

mdeweerd, 2020-10-10:
# Summary

Most items on 
[TI ZSTACK 3.0.2 Patches](https://e2e.ti.com/support/wireless-connectivity/zigbee-and-thread/f/158/t/902631)
were already applied in the ZStack 3.0.2 .

CC2531/ZNP-without-SBL fails with a link error after applying the patches and instructions from this repository.
The issue is with the "CHECKSUM", maybe you already know how to fix it.

# Compiling

I patch the `tools/znp.bat` file to launch `znp.js` :
```
[...]
wscript znp.js %2
```

as `START` does not start the correct interpreter on my Win10 machine.

Then, I create all binaries from the command line (`BUILDZNP`) with a script like this (from the CC253x directory):
```
"C:\Program Files (x86)\IAR Systems\Embedded Workbench 8.1\common\bin\IarBuild.exe" CC2531.ewp -build "ZNP-with-SBL"
"C:\Program Files (x86)\IAR Systems\Embedded Workbench 8.1\common\bin\IarBuild.exe" CC2531.ewp -build "ZNP-without-SBL"
"C:\Program Files (x86)\IAR Systems\Embedded Workbench 8.1\common\bin\IarBuild.exe" CC2530.ewp -build "ZNP-with-SBL"
"C:\Program Files (x86)\IAR Systems\Embedded Workbench 8.1\common\bin\IarBuild.exe" CC2530.ewp -build "ZNP-without-SBL"
```

I use these scripts in all cases

## Original Z-STACK 3.0.2

Compiling the unchanged Z-STACK works.
The produced hex and bin sizes are the same as the original, the content changes

## Modified Z-STACK 3.0.2

I applied patches from `https://github.com/Koenkk/Z-Stack-firmware/tree/master/coordinator/Z-Stack_3.0.x` and changed the configurations as indicated - those are reflected in the `*.ewp` files. 

The CC2530 compilations (hex and binary) work fine.

The CC2531/ZNP-with-SBL compilation also succeeds, CC2531/ZNP-without-SBL fails with a link error:
,  compilations fail during link:
```
Error[e16]: Segment CHECKSUM (size: 0x4 align: 0) is too long for segment definition. At least 0x2 more bytes needed. The problem occurred while processing the segment placement command "-Z(CODE)CHECKSUM=0x2090-0x2091", where at the moment of placement the available memory ranges were "CODE:2090-2091"
   Reserved ranges relevant to this placement:
```

After applying the patches proposed by TI,  CC2531/ZNP-without-SBL  still fails with the same error.

# Review of TI's changes

The items below are not in the patches provided in this repository.  Item 9 is implicitally applied.  The BDB F&B functionnality (19) is not enabled in this repository and does not impact the ZNP.  21 may note berelevant either, so only the first item may need to be integrated for the ZNP.

I applied:
8. ZDAPP_UPDATE_NWK_NV_TIME change

9. Increase MAXMEMHEAP to 2800 in CC2530.ewp .
   Increased INT_HEAP_LEN from 2048 to 2800 in TI2530DB/OnBoard.h

15. Errors due to BDB header build types

    Changed f8wCoord.cfg accordingly - headers seemed changed already.
    Changed EndDevice-Pro.lib and Router-Pro.lib in ewp files.
	Changed all '#if ( ZSTACK_ROUTER_BUILD ) || defined ( ZBIT )' (nwk_globals.*)

19. Memory Leak in the F&B Function of BDB
    Added bdb_zclSimpleDescClusterListClean call
	
Skipped: 20. Large number of Address Manager Entries Results in Long ZC Startup Times

21. Multiple attribute reporting issue

Koenkk, 2020-10-11:
For that compile error try reducing the `MAXMEMHEAP` a bit

mdeweerd, 2020-10-11:
I set MAXMEMHEAP to 2800, but that did not help initially - I had to fix something else first.

It turns out that it's my bad, I got the "FIRMWARE_SBL" define correct for the CC2530 compilations, but wrong for the CC2531 compilations.

As a result two extra bytes were requested for the CHECKSUM as it is being added outside the source code when compiling for  "CC2531 - ZNP-without-SBL".

After fixing that, I got the error:
"Error[e104]: Failed to fit all segments into specified ranges. Problem discovered in segment XDATA_N. Unable to place 2 block(s) (0x200 byte(s) total) in 0x131 byte(s) of m
emory. The problem occurred while processing the segment placement command "-P(XDATA)XDATA_N=_XDATA_START-_XDATA_END", where at the moment of placement the available memory
 ranges were "XDATA:1dcf-1eff"
Error while running Linker
"

I had to reduce the MAXMEMHEAP to 2995 (208 bytes less, or 0xd0 , about 0x200-0x131=0xCF) .

So the MAXMEMHEAP limit varies with FIRMWARE_SBL .

So your hint was anticipating the "next" error.  Thanks!

closed 2020-11-18 by github-actions[bot]