Koenkk/Z-Stack-firmware #218: Innr RB 245-2 Smart Lamp problems when trying to pair

DerReparator, 2020-10-13:
I have problems getting 2 newly boght Innr Smart Lapms to work with my zigbee2mqtt setup.

In the logfile of my zigbee2mqtt installation (running on a Raspberry Pi with the following Zigbee USB-module: [),](https://www.amazon.de/ITSTUFF-USB-Stick-ioBroker-openHAB-Firmware/dp/B07ZZ88BWZ/ref=sr_1_6?__mk_de_DE=%C3%85M%C3%85%C5%BD%C3%95%C3%91&dchild=1&keywords=zigbee+usb&qid=1602618284&sr=8-6) I get the following entries:

`Oct 13 20:41:35 pihole npm[1427]: Zigbee2MQTT:info  2020-10-13 20:41:35: Device '0x00158d0004fdbd0e' joined
Oct 13 20:41:35 pihole npm[1427]: Zigbee2MQTT:info  2020-10-13 20:41:35: Starting interview of '0x00158d0004fdbd0e'
Oct 13 20:41:35 pihole npm[1427]: Zigbee2MQTT:info  2020-10-13 20:41:35: MQTT publish: topic 'Wg/zigbee2mqtt/bridge/log', payload '{"message":{"friendly_name":"0x00158d0004fdbd0e"},"type":"device_connected"}'
Oct 13 20:41:35 pihole npm[1427]: Zigbee2MQTT:info  2020-10-13 20:41:35: MQTT publish: topic 'Wg/zigbee2mqtt/bridge/log', payload '{"message":"interview_started","meta":{"friendly_name":"0x00158d0004fdbd0e"},"type":"pairing"}'`

I already tried deleting the devices' entries from the database.db file as described in [this GitHub issue](https://github.com/Koenkk/Z-Stack-firmware/issues/62). But that didn't work out either.

When I send the MQTT message to switch the bulb on or off (coming from an openHab MQTT switch) running on another Raspberry Pi, I get the following on my zigbee2mqtt console:

`Oct 13 20:47:07 pihole npm[1427]: Zigbee2MQTT:warn  2020-10-13 20:47:07: Device with modelID 'undefined' is not supported.
Oct 13 20:47:07 pihole npm[1427]: Zigbee2MQTT:warn  2020-10-13 20:47:07: Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html
Oct 13 20:47:08 pihole npm[1427]: Zigbee2MQTT:warn  2020-10-13 20:47:08: Device with modelID 'undefined' is not supported.
Oct 13 20:47:08 pihole npm[1427]: Zigbee2MQTT:warn  2020-10-13 20:47:08: Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html`

Here are the MQTT messages sent:

`Client mosqsub|656-raspberrypi received PUBLISH (d0, q0, r0, m0, 'Wg/zigbee2mqtt/0x00158d0004fdbd0e/set/state', ... (2 bytes))
ON
Client mosqsub|656-raspberrypi received PUBLISH (d0, q0, r0, m0, 'Wg/zigbee2mqtt/0x00158d0004fdbd0e/set/state', ... (3 bytes))
OFF`

Thanks in advance for any help!

Koenkk, 2020-10-14:
Did you check: https://www.zigbee2mqtt.io/information/FAQ.html#interview-fails ?

DerReparator, 2020-10-15:
Now I did. Moving the Coordinator and the bulb closer together didn't help. A reset via touchlink yielded the following [output (on pastebin)](https://pastebin.com/pMZvGzfs). This Touchlink action did result in an internal zigbee2MQTT error.

I assigned a `friendly_name` to the bulb mentioned above in the meantime.

But the bulb keeps in `type: "pairing"`.

Starting herdsman in Debug mode yields the following after a restart of the zigbee2MQTT service:

`Oct 15 10:58:20 pihole npm[23433]: Zigbee2MQTT:info  2020-10-15 10:58:20: Starting Zigbee2MQTT version 1.15.0 (commit #ed8b4e5)
Oct 15 10:58:20 pihole npm[23433]: Zigbee2MQTT:info  2020-10-15 10:58:20: Starting zigbee-herdsman...
Oct 15 10:58:20 pihole npm[23433]: Zigbee2MQTT:debug 2020-10-15 10:58:20: Using zigbee-herdsman with settings: '{"adapter":{"concurrent":null},"backupPath":"/opt/zigbee2mqtt/data/coordinator_backup.json","databaseBackupPath":"/opt/zigbee2mqtt/data/database.db.backup","databasePath":"/opt/zigbee2mqtt/data/database.db","network":{"channelList":[11],"extendedPanID":[221,221,221,221,221,221,221,221],"networkKey":"HIDDEN","panID":6754},"serialPort":{"path":"/dev/ttyACM0"}}'
Oct 15 10:58:27 pihole npm[23433]: Zigbee2MQTT:info  2020-10-15 10:58:27: zigbee-herdsman started
Oct 15 10:58:27 pihole npm[23433]: Zigbee2MQTT:info  2020-10-15 10:58:27: Coordinator firmware version: '{"meta":{"maintrel":3,"majorrel":2,"minorrel":6,"product":0,"revision":20190608,"transportrev":2},"type":"zStack12"}'
Oct 15 10:58:27 pihole npm[23433]: Zigbee2MQTT:debug 2020-10-15 10:58:27: Zigbee network parameters: {"channel":11,"extendedPanID":"0xdddddddddddddddd","panID":6754}
Oct 15 10:58:27 pihole npm[23433]: Zigbee2MQTT:info  2020-10-15 10:58:27: Currently 0 devices are joined:
Oct 15 10:58:27 pihole npm[23433]: Zigbee2MQTT:warn  2020-10-15 10:58:27: `permit_join` set to  `true` in configuration.yaml.
Oct 15 10:58:27 pihole npm[23433]: Zigbee2MQTT:warn  2020-10-15 10:58:27: Allowing new devices to join.
Oct 15 10:58:27 pihole npm[23433]: Zigbee2MQTT:warn  2020-10-15 10:58:27: Set `permit_join` to `false` once you joined all devices.
Oct 15 10:58:27 pihole npm[23433]: Zigbee2MQTT:info  2020-10-15 10:58:27: Zigbee: allowing new devices to join.
Oct 15 10:58:28 pihole npm[23433]: Zigbee2MQTT:info  2020-10-15 10:58:28: Connecting to MQTT server at mqtt://192.168.178.26
Oct 15 10:58:31 pihole npm[23433]: Zigbee2MQTT:info  2020-10-15 10:58:31: Connected to MQTT server
Oct 15 10:58:31 pihole npm[23433]: Zigbee2MQTT:info  2020-10-15 10:58:31: MQTT publish: topic 'Wg/zigbee2mqtt/bridge/state', payload 'online'
Oct 15 10:58:33 pihole npm[23433]: Zigbee2MQTT:info  2020-10-15 10:58:33: Device 'PhilNachttischE14' joined
Oct 15 10:58:34 pihole npm[23433]: Zigbee2MQTT:info  2020-10-15 10:58:34: Starting interview of 'PhilNachttischE14'
Oct 15 10:58:34 pihole npm[23433]: Zigbee2MQTT:info  2020-10-15 10:58:34: MQTT publish: topic 'Wg/zigbee2mqtt/bridge/log', payload '{"message":{"friendly_name":"PhilNachttischE14"},"type":"device_connected"}'
Oct 15 10:58:34 pihole npm[23433]: Zigbee2MQTT:info  2020-10-15 10:58:34: MQTT publish: topic 'Wg/zigbee2mqtt/bridge/log', payload '{"message":"interview_started","meta":{"friendly_name":"PhilNachttischE14"},"type":"pairing"}'
Oct 15 10:58:34 pihole npm[23433]: Zigbee2MQTT:info  2020-10-15 10:58:34: MQTT publish: topic 'Wg/zigbee2mqtt/bridge/config', payload '{"commit":"ed8b4e5","coordinator":{"meta":{"maintrel":3,"majorrel":2,"minorrel":6,"product":0,"revision":20190608,"transportrev":2},"type":"zStack12"},"log_level":"debug","network":{"channel":11,"extendedPanID":"0xdddddddddddddddd","panID":6754},"permit_join":true,"version":"1.15.0"}'`

Koenkk, 2020-10-15:
Sorry, touchlink is currently broken in 1.15.0, switch to the dev branch: https://www.zigbee2mqtt.io/how_tos/how-to-switch-to-dev-branch.html OR reset via touchlink using the new api: https://github.com/Koenkk/zigbee2mqtt/issues/3281 / https://github.com/Koenkk/zigbee2mqtt.io/blob/new_api/docs/information/mqtt_topics_and_message_structure.md#zigbee2mqttbridgeinfo

DerReparator, 2020-10-15:
Thanks for your quick response! The Touchlink reset apparently went through this time. Unfortunately, the lamp still refuses to pair correctly:

> Oct 15 19:12:31 pihole npm[1262]: Zigbee2MQTT:debug 2020-10-15 19:12:31: Received MQTT message on 'Wg/zigbee2mqtt/bridge/config/touchlink/factory_reset' with data ''
Oct 15 19:12:32 pihole npm[1262]: Zigbee2MQTT:info  2020-10-15 19:12:32: Starting touchlink factory reset...
Oct 15 19:12:32 pihole npm[1262]: Zigbee2MQTT:info  2020-10-15 19:12:32: MQTT publish: topic 'Wg/zigbee2mqtt/bridge/log', payload '{"message":"reset_started","meta":{"status":"started"},"type":"touchlink"}'
Oct 15 19:12:37 pihole npm[1262]: Zigbee2MQTT:info  2020-10-15 19:12:37: Successfully factory reset device through Touchlink
Oct 15 19:12:37 pihole npm[1262]: Zigbee2MQTT:info  2020-10-15 19:12:37: MQTT publish: topic 'Wg/zigbee2mqtt/bridge/log', payload '{"message":"reset_success","meta":{"status":"success"},"type":"touchlink"}'
Oct 15 19:12:42 pihole npm[1262]: Zigbee2MQTT:error 2020-10-15 19:12:42: Failed to interview 'PhilNachttischE14', device has not successfully been paired
Oct 15 19:12:42 pihole npm[1262]: Zigbee2MQTT:info  2020-10-15 19:12:42: MQTT publish: topic 'Wg/zigbee2mqtt/bridge/log', payload '{"message":"interview_failed","meta":{"friendly_name":"PhilNachttischE14"},"type":"pairing"}'
Oct 15 19:12:43 pihole npm[1262]: Zigbee2MQTT:info  2020-10-15 19:12:43: Starting interview of 'PhilNachttischE14'
Oct 15 19:12:43 pihole npm[1262]: Zigbee2MQTT:info  2020-10-15 19:12:43: MQTT publish: topic 'Wg/zigbee2mqtt/bridge/log', payload '{"message":"interview_started","meta":{"friendly_name":"PhilNachttischE14"},"type":"pairing"}'
Oct 15 19:15:01 pihole npm[1262]: Zigbee2MQTT:debug 2020-10-15 19:15:01: Saving state to file /opt/zigbee2mqtt/data/state.json
Oct 15 19:15:01 pihole npm[1262]: Zigbee2MQTT:error 2020-10-15 19:15:01: Failed to interview 'PhilNachttischE14', device has not successfully been paired
Oct 15 19:15:01 pihole npm[1262]: Zigbee2MQTT:info  2020-10-15 19:15:01: MQTT publish: topic 'Wg/zigbee2mqtt/bridge/log', payload '{"message":"interview_failed","meta":{"friendly_name":"PhilNachttischE14"},"type":"pairing"}'
Oct 15 19:15:03 pihole npm[1262]: Zigbee2MQTT:info  2020-10-15 19:15:03: Starting interview of 'PhilNachttischE14'
Oct 15 19:15:03 pihole npm[1262]: Zigbee2MQTT:info  2020-10-15 19:15:03: MQTT publish: topic 'Wg/zigbee2mqtt/bridge/log', payload '{"message":"interview_started","meta":{"friendly_name":"PhilNachttischE14"},"type":"pairing"}'
Oct 15 19:15:29 pihole npm[1262]: Zigbee2MQTT:debug 2020-10-15 19:15:29: Received MQTT message on 'Wg/zigbee2mqtt/PhilNachttischE14/set/state' with data 'ON'
Oct 15 19:15:29 pihole npm[1262]: Zigbee2MQTT:warn  2020-10-15 19:15:29: Device with modelID 'undefined' is not supported.
Oct 15 19:15:29 pihole npm[1262]: Zigbee2MQTT:warn  2020-10-15 19:15:29: Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html
Oct 15 19:15:30 pihole npm[1262]: Zigbee2MQTT:debug 2020-10-15 19:15:30: Received MQTT message on 'Wg/zigbee2mqtt/PhilNachttischE14/set/state' with data 'OFF'
Oct 15 19:15:30 pihole npm[1262]: Zigbee2MQTT:warn  2020-10-15 19:15:30: Device with modelID 'undefined' is not supported.
Oct 15 19:15:30 pihole npm[1262]: Zigbee2MQTT:warn  2020-10-15 19:15:30: Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html
Oct 15 19:15:30 pihole npm[1262]: Zigbee2MQTT:debug 2020-10-15 19:15:30: Received MQTT message on 'Wg/zigbee2mqtt/PhilNachttischE14/set/state' with data 'ON'
Oct 15 19:15:30 pihole npm[1262]: Zigbee2MQTT:warn  2020-10-15 19:15:30: Device with modelID 'undefined' is not supported.

Koenkk, 2020-10-15:
Can you provide the herdsman debug logging of the pairing?

To enable herdsman debug logging, see https://www.zigbee2mqtt.io/information/debug.html#zigbee-herdsman-debug-logging

DerReparator, 2020-10-24:
Herdsman yielded the following output: [Pastebin](https://pastebin.com/7uXYmTpL). From what I can tell, it still doesn't recognize the type of lamp I have?

Koenkk, 2020-10-24:
I see you stop Zigbee2MQTT before the interview finishes, give it some time.

DerReparator, 2020-10-24:
Okay, now I think there's a useful error message at the end of this logfile: [Pastebin](https://pastebin.com/kb2Q4GkP).
Interesting lines:
>  zigbee-herdsman:controller:device:log Interview - failed for device '0x00158d0004fdbd0e' with error 'Error: Interview failed because can not get node descriptor ('0x00158d0004fdbd0e')
  zigbee-herdsman:controller:device:log     at Device.<anonymous> (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/controller/model/device.js:348:23)
  zigbee-herdsman:controller:device:log     at Generator.throw (<anonymous>)
  zigbee-herdsman:controller:device:log     at rejected (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/controller/model/device.js:25:65)' +1ms
  zigbee-herdsman:controller:error Interview failed for '0x00158d0004fdbd0e with error 'Error: Interview failed because can not get node descriptor ('0x00158d0004fdbd0e')' +2m
Zigbee2MQTT:error 2020-10-24 18:10:37: Failed to interview '0x00158d0004fdbd0e', device has not successfully been paired
Zigbee2MQTT:info  2020-10-24 18:10:37: MQTT publish: topic 'Wg/zigbee2mqtt/bridge/log', payload '{"message":"interview_failed","meta":{"friendly_name":"0x00158d0004fdbd0e"},"type":"pairing"}'

Koenkk, 2020-10-24:
@DerReparator it seems the device does not join the network at all. Can you check: https://www.zigbee2mqtt.io/information/FAQ.html#why-does-my-device-not-or-fail-to-pair

DerReparator, 2020-11-25:
The issue was fixed a few days ago by pressing random buttons on the (unfortunately undocumented) ZigBee adapter I use. One of the buttons said "Debug". I don't know what exactly this changed.

But now I could pair the Innr bulbs and another Ikea ZigBee lamp a friend gave me without problems.

closed 2020-12-27 by DerReparator