Koenkk/Z-Stack-firmware #222: Router firmware is very noisy on MQTT

jeroenvermeulen, 2020-10-23:
I first want to thank @Koenkk and the rest of the contributors for this great project.

I have a [CC2531 with a big antenna](https://www.aliexpress.com/item/4000439492385.html) which I use as router in combination with a [Sonoff Bridge running Tasmota](https://zigbee.blakadder.com/Sonoff_ZBBridge.html).

**My problem is:**
The CC2531 is sending an update over MQTT every second on topic `tele/zbbridge_E2FABC/8ED0`, reporting:
```
{"0x8ED0":{"Device":"0x8ED0","Name":"UsbRouterAnt","Power":0,"Endpoint":8,"LinkQuality":21}}
```
This makes it quite noisy, and I have a feeling it slows down other MQTT communication.

It this something fixable of configurable?

alsa-kode, 2020-11-02:
+1
Same issue here. Not a big deal but it is sort of inconvenient sometimes.

ptvoinfo, 2020-11-03:
@jeroenvermeulen @alsa-kode 
Please, try the following firmware. There you may configure the reporting interval or disable it at all.
https://ptvo.info/zigbee-switch-configurable-firmware-v2-210/
The firmware for CC2530 without oscillator should work for CC2531.

closed 2020-12-11 by github-actions[bot]