Koenkk/Z-Stack-firmware #223: After firmwareupdate on slaesh’s CC2652RB stick I need to repair all devices

aveseli, 2020-10-24:
I've done a firmware update on slaesh’s CC2652RB stick. Unfortunately I need to re-pair all my devices.

Update from CC2652RB_20200805.hex to np_CC2652RB_20200715.bin.

I followed the instructions on https://slae.sh/projects/cc2652/#flashing.

The command use to update the firmware:
./cc2538-bsl.py -p /dev/cu.usbserial-1410 -evw znp_CC2652RB_LAUNCHXL_tirtos_ccs.hex

Is it required to re-pair all devices after fw update?
What am I doing wrong?

Koenkk, 2020-10-25:
this is not required as long there is a coordinator_backup.json in the Zigbee2mqtt data directory (which is created automatically)

cndian, 2020-10-26:
Where did you put the firmware update file? Because I want to update my Slaesh also with the method you did.

aveseli, 2020-11-06:
I updated to the latest firmware yesterday, some devices are working and some are not working anymore.

problematic devices:
- [AC0251100NJ/AC0251700NJ](https://www.zigbee2mqtt.io/devices/AC0251100NJ_AC0251700NJ.html)
- [LED1624G9](https://www.zigbee2mqtt.io/devices/LED1624G9.html)

Which information do you need and how can I gather them?

This is the error message in zigbee2mqtt when trying to communicate with the ikea bulb:
`Nov 06 11:02:00 raspberrypi npm[3801]: Zigbee2MQTT:error 2020-11-06 11:02:00: Publish 'set' 'brightness' to 'og_buero_nachtlicht' failed: 'Error: Command 0x000d6ffffe48f7cd/1 genLevelCtrl.moveToLevelWithOnOff({"level":254,"transtime":10}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null}) failed (Data request failed with error: 'No network route' (205))'
Nov 06 11:02:00 raspberrypi npm[3801]: Zigbee2MQTT:info  2020-11-06 11:02:00: MQTT publish: topic 'zigbee2mqtt3/bridge/log', payload '{"message":"Publish 'set' 'brightness' to 'og_buero_nachtlicht' failed: 'Error: Command 0x000d6ffffe48f7cd/1 genLevelCtrl.moveToLevelWithOnOff({\"level\":254,\"transtime\":10}, {\"timeout\":10000,\"disableResponse\":false,\"disableRecovery\":false,\"disableDefaultResponse\":false,\"direction\":0,\"srcEndpoint\":null,\"reservedBits\":0,\"manufacturerCode\":null,\"transactionSequenceNumber\":null}) failed (Data request failed with error: 'No network route' (205))'","meta":{"friendly_name":"og_buero_nachtlicht"},"type":"zigbee_publish_error"}'`

aveseli, 2020-11-06:
> Where did you put the firmware update file? Because I want to update my Slaesh also with the method you did.

You just need to download and unzip it. It can be anywhere on your machine as log as you provide the path in command.

Koenkk, 2020-11-06:
@aveseli the backup looks to be restored correctly. However after reflashing the coordinator needs to build up routes again. If this for some reason fails for a device, you can try to repower that device.

aveseli, 2020-11-06:
I have repowered the ikea bulb and it works now. even the osram remote is working now. I just discovered that another device is not working though, ikea soundcontroller.

closed 2020-12-14 by github-actions[bot]