Koenkk/Z-Stack-firmware #224: remove device while device trying to join insecurely

Arminhosseini, 2020-10-26:
I have a problem related to cc2530 firmware. 
when some devices trying to join many times they try to join with an insecure update device so the coordinator sending remove device and force the device to leave so I searched that issue in the TI forum and find it's problem is like below:
occurs if BDB_DEFAULT_TC_REQUIRE_KEY_EXCHANGE is set to TRUE and pre-Zigbee 3.0 Spec devices attempt to join without sending a TCLK update.
and I want to ask you could you please change cc230 firmware that has been used for zigbee2mqtt coordinator so it let's insecure joining.
regards.

Koenkk, 2020-10-26:
I've checked but this is already set to `FALSE`:

[image]

As TP2_LEAGCY_ZC is defined (https://github.com/Koenkk/Z-Stack-firmware/blob/f2a7aa17fba716c72dd46d4821328f7daeb1e238/coordinator/Z-Stack_3.0.x/firmware_CC2531_CC2530.patch#L195)

Arminhosseini, 2020-10-28:
ok, I will check this patch and I will let you know.
thank's

Arminhosseini, 2020-10-28:
oooh, I'm not using z-stack3 I'm using zstack ha 1.2

Koenkk, 2020-10-28:
1.2 does not have `BDB_DEFAULT_TC_REQUIRE_KEY_EXCHANGE`

closed 2020-12-05 by github-actions[bot]