Koenkk/Z-Stack-firmware #227: error woth latest firmware for CC1352P_2

mpuff, 2020-11-01:
after flashing the CC1352P_2_20201026.zip, i cant controll any devices and after a reboot zigbee2mqtt was not starting anymoe.
Got the error:

Nov 01 17:07:19 zigbee2mqttpi01 npm[635]: Zigbee2MQTT:error 2020-11-01 17:07:19: Error while starting zigbee-herdsman
Nov 01 17:07:19 zigbee2mqttpi01 npm[635]: Zigbee2MQTT:error 2020-11-01 17:07:19: Failed to start zigbee
Nov 01 17:07:19 zigbee2mqttpi01 npm[635]: Zigbee2MQTT:error 2020-11-01 17:07:19: Exiting...
Nov 01 17:07:19 zigbee2mqttpi01 npm[635]: Zigbee2MQTT:error 2020-11-01 17:07:19: Error: Failed to connect to the adapter (Error: SRSP - SYS - ping after 6000ms)
Nov 01 17:07:19 zigbee2mqttpi01 npm[635]: at ZStackAdapter.<anonymous> (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/adapter/z-stack/adapter/zStackAdapter.js:94:31)
Nov 01 17:07:19 zigbee2mqttpi01 npm[635]: at Generator.throw (<anonymous>)
Nov 01 17:07:19 zigbee2mqttpi01 npm[635]: at rejected (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/adapter/z-stack/adapter/zStackAdapter.js:25:65)
Nov 01 17:07:19 zigbee2mqttpi01 npm[635]: npm ERR! code ELIFECYCLE
Nov 01 17:07:19 zigbee2mqttpi01 npm[635]: npm ERR! errno 1
Nov 01 17:07:19 zigbee2mqttpi01 npm[635]: npm ERR! zigbee2mqtt@1.16.0 start: `node index.js`
Nov 01 17:07:19 zigbee2mqttpi01 npm[635]: npm ERR! Exit status 1
Nov 01 17:07:19 zigbee2mqttpi01 npm[635]: npm ERR!
Nov 01 17:07:19 zigbee2mqttpi01 npm[635]: npm ERR! Failed at the zigbee2mqtt@1.16.0 start script.
Nov 01 17:07:19 zigbee2mqttpi01 npm[635]: npm ERR! This is probably not a problem with npm. There is likely additional logging output above.
Nov 01 17:07:19 zigbee2mqttpi01 npm[635]: npm ERR! A complete log of this run can be found in:
Nov 01 17:07:19 zigbee2mqttpi01 npm[635]: npm ERR!     /home/pi/.npm/_logs/2020-11-01T16_07_19_329Z-debug.log
Nov 01 17:07:19 zigbee2mqttpi01 systemd[1]: zigbee2mqtt.service: main process exited, code=exited, status=1/FAILURE
Nov 01 17:07:19 zigbee2mqttpi01 systemd[1]: Unit zigbee2mqtt.service entered failed state.
Nov 01 17:07:19 zigbee2mqttpi01 systemd[1]: zigbee2mqtt.service holdoff time over, scheduling restart.
Nov 01 17:07:19 zigbee2mqttpi01 systemd[1]: Stopping zigbee2mqtt...

tpftpf, 2020-11-02:
I could flash mine without a problem.

Koenkk, 2020-11-02:
Probably a port/usb communication problem not related to the firmware. Check https://www.zigbee2mqtt.io/information/FAQ.html#help-zigbee2mqtt-fails-to-start (the issue probably occurred because you have to disconnected the adapter).

OzGav, 2020-11-09:
I might have the same issue. I can flash the "source routing" firmware (which I'm currently using) no problem but if I try this new firmware I just get a red flashing light and no adapter detected. Is that what you are experiencing @mpuff

closed 2020-12-18 by github-actions[bot]