Koenkk/Z-Stack-firmware #228: New firmware for CC1352P/CC2652P devices that are not using Launchpad RF Control pins

omerk, 2020-11-07:
As discussed in #210, there are some modules and boards that do not follow the same RF switch arrangement as the TI Launchpad designs and these will need a new firmware.

Why do we need a switch in the first place? There are two possible RF paths for CC2652P and three for CC1352P:

  1) 2.4Ghz: Standard 5dBm for TX and RX
  2) 20dBm PA : 20dBm TX only
  3) Sub-1GHz (CC1352P only)

The RF Switch allows for the same antenna to be used with different RF paths.

Popular module maker Ebyte has two modules, E72 (CC2652P):

[image]

...and E79 (CC1352P):

[image]

A little ambiguous especially for E79 so might need a bit of trial and error 

Discussion on how these GPIOs can be changed in the ZNP firmware can be found in this E2E post: https://e2e.ti.com/support/wireless-connectivity/zigbee-and-thread/f/158/t/894423

/cc @mercenaruss @sjorge @Koenkk

Koenkk, 2020-11-10:
Firmware is available: https://github.com/Koenkk/Z-Stack-firmware/tree/develop/coordinator/Z-Stack_3.x.0/bin

sjorge, 2020-11-12:
@Koenkk I'm trying to follow COMPILE.MD to build the image for the zzhp-lite so I can tweak the RF values if needed, one should be on the way to me for testing... but it is unclear how to produce the `_other_` variant of the firmware?

sjorge, 2020-11-12:
I cleared up the COMPILE.MD a bit, but still having a ton of issues following the instructions, like applying the patch now fails on znp.syscfg... I think because the 100 -> 1000 -> 100 change? (Also i was using windows, which seems to be a huge mistake? does it only work on linux?)

sjorge, 2020-11-12:
Better results using macOS, but it still fails but on a different error with the patch now... one that makes sense

```
[axion :: sjorge][~/Documents/zzhp-lite]
[●]$ grep -r com.ti.ccstudio.buildDefinitions.core.OPT_CODEGEN_VERSION .
./firmware.patch:-							<option id="com.ti.ccstudio.buildDefinitions.core.OPT_CODEGEN_VERSION.360529634" superClass="com.ti.ccstudio.buildDefinitions.core.OPT_CODEGEN_VERSION" value="20.2.1.LTS" valueType="string"/>
./firmware.patch:+							<option id="com.ti.ccstudio.buildDefinitions.core.OPT_CODEGEN_VERSION.360529634" name="Compiler version" superClass="com.ti.ccstudio.buildDefinitions.core.OPT_CODEGEN_VERSION" value="20.2.1.LTS" valueType="string"/>
./firmware.patch:+							<option id="com.ti.ccstudio.buildDefinitions.core.OPT_CODEGEN_VERSION.573075251" name="Compiler version" superClass="com.ti.ccstudio.buildDefinitions.core.OPT_CODEGEN_VERSION" value="20.2.2.LTS" valueType="string"/>
./znp_CC2652RB_LAUNCHXL_tirtos_ccs/.cproject:							<option id="com.ti.ccstudio.buildDefinitions.core.OPT_CODEGEN_VERSION.1149369988" superClass="com.ti.ccstudio.buildDefinitions.core.OPT_CODEGEN_VERSION" value="20.2.1.LTS" valueType="string"/>
./znp_CC1352P_2_LAUNCHXL_tirtos_ccs/.cproject:							<option id="com.ti.ccstudio.buildDefinitions.core.OPT_CODEGEN_VERSION.635435663" superClass="com.ti.ccstudio.buildDefinitions.core.OPT_CODEGEN_VERSION" value="20.2.1.LTS" valueType="string"/>
./znp_CC26X2R1_LAUNCHXL_tirtos_ccs/.cproject:							<option id="com.ti.ccstudio.buildDefinitions.core.OPT_CODEGEN_VERSION.1307066957" superClass="com.ti.ccstudio.buildDefinitions.core.OPT_CODEGEN_VERSION" value="20.2.1.LTS" valueType="string"/>
```

com.ti.ccstudio.buildDefinitions.core.OPT_CODEGEN_VERSION.360529634 (patch)
com.ti.ccstudio.buildDefinitions.core.OPT_CODEGEN_VERSION.635435663 (code)

But I am using the listed SDK
```
[axion :: sjorge][~/Documents]
[●]$ ls -d /Applications/ti/simplelink_*
/Applications/ti/simplelink_cc13x2_26x2_sdk_4_30_00_54/
```

Koenkk, 2020-11-12:
Seems that we cannot patch the .cproject automatically (as it will differ even with the same SDK users). I will update the compile.md with the manual instructions.

sjorge, 2020-11-12:
Can we have .cproject and the other file as is?

~ sjorge

Koenkk, 2020-11-13:
@sjorge preferably not, I don't know if these files are platform/ccs version dependent. I've updated the instructions https://github.com/Koenkk/Z-Stack-firmware/blob/develop/coordinator/Z-Stack_3.x.0/COMPILE.md

mercenaruss, 2020-11-13:
@sjorge you are off-top here. Please open another issue if you fail to build,here is another tematic.

sjorge, 2020-11-13:
@mercenaruss i disagree that being unable to reproduce the build mentioned here is off-topic.

Tweaking settings and rebuilding the firmware are part of the optimization and one should not depend on koenkk to do so.

Koenkk, 2020-11-13:
@mercenaruss I don't see why it's off-topic and I liked the fact to @sjorge provided feedback to my instructions (which found out to only be working for me).

I've published a new firmware for this which sets the default transmit power from 0 to 20 dbm (big thanks to @egony). https://github.com/Koenkk/Z-Stack-firmware/blob/develop/coordinator/Z-Stack_3.x.0/bin/CC1352P2_CC2652P_other_20201113.zip

mercenaruss, 2020-11-13:
@Koenkk feedback is all the time good.
Is hard to followup with some issues if discussion is about another things. Anyway thanks to @sjorge .

closed 2020-12-22 by github-actions[bot]