Koenkk/Z-Stack-firmware #232: CC2530 with latest 3.0.X firmware freezes on BUFFER_FULL error

emc02, 2020-11-13:
Platform: zigbee2mqtt on Raspberry Pi3
problem: CC2530 freezes and can only be resettet by removing it and attach again (needs to be powered down completely, a reboot of the pi is not enough, so I think the CC2530 crashes)
way to reproduce: pair with [Immax 07046L](https://www.zigbee2mqtt.io/devices/07046L.html) and after successful pairing press the buttons >5x in a row to produce high traffic.
The device always sends the commands 2 times and after about 3 seconds again 2 times.

log: see attached file
[LOG.txt](https://github.com/Koenkk/Z-Stack-firmware/files/5537173/LOG.txt)

A workaround is not to press the button that often in a row, but my children don't always do what I want :-D

Koenkk, 2020-11-14:
I don't recommend the zigbee 3.0 firmware for the CC2530, the CC2530 is not very powerful and the 3.0 firmware is more heavy than the 1.2 firmware. I recommend https://github.com/Koenkk/Z-Stack-firmware/tree/master/coordinator/Z-Stack_Home_1.2/bin/source_routing instead but this will require repairing.

emc02, 2020-11-14:
Ok, thanks, after reading a few articles about cc2530 and the z-stack versions I also thinking that ver 1.2 would be better. Are all zigbee devices compatible with version 1.2?

Koenkk, 2020-11-14:
For the recommended coordinator check: https://www.zigbee2mqtt.io/getting_started/what_do_i_need.html#supported-zigbee-adapter

Panid cannot be made static on z-stack 3.0 coordinators since those won't start when a pan id is already in use.

emc02, 2020-11-15:
do you know any coordinator-device that can have a static pan id?

Koenkk, 2020-11-16:
@emc02 zstack 1.2 with the CC2531 can have a static pan id

emc02, 2020-11-16:
@Koenkk ok, perfect! Thanks!
you write "can have" under which circumstances?
I found the option in zigbee2mqtt config file (advanced -> pan_id), would that be the right way for static pan_id?

Koenkk, 2020-11-16:
advanced -> pan_id is correct indeed, also goes for the CC2530.

orrpan, 2020-12-09:
I've had the same error with CC2530+CC2591 v3.0.x gpio wiring to RPI after updating to 1.16.x. So I guess I'll go back to v1.2.

I have the same question as @emc02 
> Are all zigbee devices compatible with version 1.2?
@Koenkk

Nefarious19, 2020-12-09:
hello guys. Can you tell me FW you are providing suports standard ZNP protocol?

closed 2021-01-16 by github-actions[bot]