Koenkk/Z-Stack-firmware #235: ZZH adapter reports low LQI values

Anjerlaan, 2020-11-16:
Hi,

Since a month I migrated from CC2531 (zstack 1.2) to a ZZH (zStack3x0 20200925) adapter.
Z2M version: 1.15.0 
System: RPi 3b+ with Raspbian Buster

I thought that the LQI values of a ZZH adapter would be much higher than the CC2531?
Here is a screenshot of the experimental frontend:
 
[image]

Last week I had to force-remove and /re-configure the first device of the list above.
When the Device was succesfully interviewed, I saw a LQI value of approx 221, but an hour later the value was already below 80.

The Coordinator is located in the "Meter cupboard" at the ground floor, and the first device in the list above is approx. 4 meters away also at the ground floor. There is only a thin wall (not concrete) between them.

Are my values normal of is something wrong?

Koenkk, 2020-11-17:
These values look normal to me (I have similar ones)

Anjerlaan, 2020-11-17:
> These values look normal to me (I have similar ones)

Ok thanx!

closed 2020-11-18 by Koenkk