Koenkk/Z-Stack-firmware #236: CC1352P E79 Dev firmware: Error: SRSP - AF - register after 6000ms

public-radiation, 2020-11-18:
I have a E79 ebytes module that i solderd cJtag cables to and a FTDI module on (long time a go, end of 2019). I have been looking at the latest version of the firmware, now that there is support coming for the E79. But I can't seem to get it to work with the latest firmware on dev. Do I need to update my module wiring?

The firmware version that did "work": (CC1352P_2_20200417.hex)
`Zigbee2MQTT:debug 2020-11-18 20:49:32: Loaded state from file /opt/zigbee2mqtt/data/state.json
Zigbee2MQTT:info  2020-11-18 20:49:32: Logging to console and directory: '/opt/zigbee2mqtt/data/log/2020-11-18.20-49-32' filename: log.txt
Zigbee2MQTT:info  2020-11-18 20:49:32: Starting Zigbee2MQTT version 1.16.1-dev (commit #1e9553f)
Zigbee2MQTT:info  2020-11-18 20:49:32: Starting zigbee-herdsman (0.13.29)
Zigbee2MQTT:debug 2020-11-18 20:49:32: Using zigbee-herdsman with settings: '{"adapter":{"concurrent":16,"delay":null},"backupPath":"/opt/zigbee2mqtt/data/coordinator_backup.json","databaseBackupPath":"/opt/zigbee2mqtt/data/database.db.backup","databasePath":"/opt/zigbee2mqtt/data/database.db","network":{"channelList":[25],"extendedPanID":[56,159,38,199,254,128,54,3],"networkKey":"HIDDEN","panID":6773},"serialPort":{"baudRate":115200,"path":"/dev/serial/by-id/usb-FTDI_USB__-__Serial-if00-port0","rtscts":false}}'
Zigbee2MQTT:info  2020-11-18 20:49:35: zigbee-herdsman started
Zigbee2MQTT:info  2020-11-18 20:49:35: Coordinator firmware version: '{"meta":{"maintrel":1,"majorrel":2,"minorrel":7,"product":1,"revision":20201026,"transportrev":2},"type":"zStack3x0"}'
Zigbee2MQTT:debug 2020-11-18 20:49:35: Zigbee network parameters: {"channel":25,"extendedPanID":"0x033680fec7269f38","panID":6773}
Zigbee2MQTT:info  2020-11-18 20:49:35: Currently 0 devices are joined:
Zigbee2MQTT:warn  2020-11-18 20:49:35: `permit_join` set to  `true` in configuration.yaml.
Zigbee2MQTT:warn  2020-11-18 20:49:35: Allowing new devices to join.
Zigbee2MQTT:warn  2020-11-18 20:49:35: Set `permit_join` to `false` once you joined all devices.
Zigbee2MQTT:info  2020-11-18 20:49:35: Zigbee: allowing new devices to join.
Zigbee2MQTT:info  2020-11-18 20:49:35: Connecting to MQTT server at mqtt://localhost
Zigbee2MQTT:debug 2020-11-18 20:49:35: Using MQTT anonymous login
Zigbee2MQTT:info  2020-11-18 20:49:35: Connected to MQTT server
Zigbee2MQTT:info  2020-11-18 20:49:35: MQTT publish: topic 'zigbee2mqtt/bridge/state', payload 'online'
Zigbee2MQTT:info  2020-11-18 20:49:35: MQTT publish: topic 'zigbee2mqtt/bridge/config', payload '{"commit":"1e9553f","coordinator":{"meta":{"maintrel":1,"majorrel":2,"minorrel":7,"product":1,"revision":20201026,"transportrev":2},"type":"zStack3x0"},"log_level":"debug","network":{"channel":25,"extendedPanID":"0x033680fec7269f38","panID":6773},"permit_join":true,"version":"1.16.1-dev"}'
`

Dev version that doesn't work (CC1352P2_CC2652P_other_20201113.hex):
`
node index.js
Zigbee2MQTT:debug 2020-11-18 20:50:24: Loaded state from file /opt/zigbee2mqtt/data/state.json
Zigbee2MQTT:info  2020-11-18 20:50:24: Logging to console and directory: '/opt/zigbee2mqtt/data/log/2020-11-18.20-50-24' filename: log.txt
Zigbee2MQTT:info  2020-11-18 20:50:24: Starting Zigbee2MQTT version 1.16.1-dev (commit #1e9553f)
Zigbee2MQTT:info  2020-11-18 20:50:24: Starting zigbee-herdsman (0.13.29)
Zigbee2MQTT:debug 2020-11-18 20:50:24: Using zigbee-herdsman with settings: '{"adapter":{"concurrent":16,"delay":null},"backupPath":"/opt/zigbee2mqtt/data/coordinator_backup.json","databaseBackupPath":"/opt/zigbee2mqtt/data/database.db.backup","databasePath":"/opt/zigbee2mqtt/data/database.db","network":{"channelList":[25],"extendedPanID":[56,159,38,199,254,128,54,3],"networkKey":"HIDDEN","panID":6773},"serialPort":{"baudRate":115200,"path":"/dev/serial/by-id/usb-FTDI_USB__-__Serial-if00-port0","rtscts":false}}'
Zigbee2MQTT:error 2020-11-18 20:50:35: Error while starting zigbee-herdsman
Zigbee2MQTT:error 2020-11-18 20:50:35: Failed to start zigbee
Zigbee2MQTT:error 2020-11-18 20:50:35: Exiting...
Zigbee2MQTT:error 2020-11-18 20:50:35: Error: SRSP - AF - register after 6000ms
    at Timeout.waiter.timer.setTimeout [as _onTimeout] (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/utils/waitress.js:46:35)
    at ontimeout (timers.js:436:11)
    at tryOnTimeout (timers.js:300:5)
    at listOnTimeout (timers.js:263:5)
    at Timer.processTimers (timers.js:223:10)
`

Any help is welcome!

Koenkk, 2020-11-19:
If CC1352P_2_20201026 worked you should use the launchpad version. `SRSP - AF - register after 6000ms` indicates that the adapter started but the connection failed while starting.

public-radiation, 2020-11-25:
Hi. I did check the launch pad version (last block in the comment). But will look more why my modules are misbehaving. Maybe its more senstive now and that is causing issues on my end. Thanks anyway, just wanted to be sure i didn't chase my own tail.

closed 2021-01-02 by github-actions[bot]