Koenkk/Z-Stack-firmware #240: More errror

SargonofAssyria, 2020-12-03:


Koenkk, 2020-12-03:
Does downgrading resolve the issues?

SargonofAssyria, 2020-12-03:
I Will check tomorrow.

SargonofAssyria, 2020-12-04:
It looks like it's not the firmware, because today with firmware 20201026 I got this error.
`(Data request failed with error: 'MAC channel access failure' (225))`
I did not notice that kind of errors before, so it should be something else what changed.
I only added a SEA801, which showed up as a SEA802. That's the only change to the network.
Do you have an idea what might cause these errors?

closed 2020-12-04 by SargonofAssyria

njordr, 2020-12-20:
@SargonofAssyria I have the same error (MAC channel access failure), how did you fix it?

njordr, 2020-12-20:
For the other ones who have the same issue, the solution for me was in this other post https://github.com/Koenkk/zigbee2mqtt/issues/2437 (detaching USB hard drive worked).

SargonofAssyria, 2020-12-20:
I did not fix it. I occasionally have those (and other) kind of errors.
I cannot detach SSD drive, because my system is on it.