Koenkk/Z-Stack-firmware #241: Issue Flash cc2530 router + raspberry pi

pedroxns, 2020-12-05:
Hey, first of all I have to tell that you guys are fantastic, thanks for all the programing and effort to help noobs like me !
I already have a coordinator running on a rPi4 with 19 devices (had 20 but deleted 1 to add the router), in my configuration.yaml I have the permission to join set to true an I'm capable to add devices when needed.

Now...my issue is with the flashing process, I'm following all the instructions for flashing with a rPi.
I have a rPi Zero, a lot of jumper cables, a CC2530 with an external antena that just arrived from china and I'm using a MacBook.
The process I've made was:
- Wiringpi + flash_cc2531 on rPi with DietPi
- Wired the CC2530:
VCC -> pin 1
GRD -> pin 39
RST -> pin 35
P22 –> pin 36 
P21 –> pin 38 

The steps:
```
root@DietPi:~/flash_cc2531# ./cc_erase
  ID = a524.
  erase result = 00a2.
root@DietPi:~/flash_cc2531# ./cc_write router-cc2530-std.hex
  ID = a524.
  reading line 12420.
  file loaded (12422 lines read).
writing page  98/ 98.
verifying page  98/ 98.
 flash OK.
root@DietPi:~/flash_cc2531# 
```

My issue is that nothing happens ... 
In my zigbee2mqtt journal I have nothing indicating that the router was adopted and in my database.db I just have 1 coordinator and 19 end devices.

What am I doing wrong?

closed 2020-12-08 by pedroxns

gleon12345, 2023-08-14:
Добрый день. У меня есть проблема после прошивки сс2530 с raspberry 3, возникает ошибка, как я могу исправить?
[image]
[image]

gleon12345, 2023-08-14:
@Koenkk I have a problem after flashing the ss2530 with an antenna and with raspberry 3 b, an error occurs, how can I fix it?