Koenkk/Z-Stack-firmware #242: how long from develop to master firmware?

bartve66, 2020-12-09:
Just curious: how long does it take for firmware in the develop branch to get promoted to the default/master (production?) branch? In general, and now in particular for the dev version of the ZZH firmware (CC26X2R1_20201113?).
I have some time planned during the holiday season to do some maintenance on my zigbee2mqtt/home assistant system, and finally make the jump from my current cc2530 coordinator to the new ZZH stick, and would prefer to do it right from the start, with the new firmware installed. 
Thanks!

Koenkk, 2020-12-09:
If no problems are found are reported current develop firmware will be released to master on 1 january (together with new Zigbee2MQTT release)

closed 2020-12-09 by Koenkk

ProfDrYoMan, 2021-01-08:
Happy new years. 1.1. passed, any news?

Koenkk, 2021-01-08:
@ProfDrYoMan delayed for 2 feb release for more tests