Koenkk/Z-Stack-firmware #243: CC2531 router not connecting to CC2531 cortinator

bigretromike, 2020-12-21:
I tried and flash CC2531_router_2020_09_29 std version as the diag-usb one.
PANID:1A62,Ch:11,Join:1 is only thing I see via usb diag - Join:1 is when I Permit to Join on cortinator, when I dont I see 0. So it does see the network but can't join it.

I dont see any router (new device) in zigbee2mqtt (1.16.2.2) panel nothing in logs.

The router is flashing red 1flash per second. I tried SW1 and SW2, reseting, reinit joining, chaining mode, and still nothing.

The coordinator is also cc2531 flashed with CC2531_DEFAULT_20190608/CC2531ZNP-Prod.hex, working fine with zigbee2mqtt (1.16.2.2)

I tried to stand next to coordinator with router no difference. This is the only network here (previously there was one with zigbee gateway but I disconnected it and all the devices was moved to my new coordinator)

What more I can do to make it work?

I tried procedure from: https://github.com/Koenkk/zigbee2mqtt/issues/4445

I remove every zigbee device I had, disable and enable permit to join, nothing, restarted zigbee2mqtt permit to join, nothing, restarted HA, permit to join, nothing.

bigretromike, 2020-12-26:
looks like zigbee2mqtt without devices.js is not able to recognize cc2531 router. after adding proper `devices.js` to data_path and adding `zigbee_shepherd_devices: true` to config the cc2531 is recognized as cc2530 and interview correctly.

closed 2020-12-26 by bigretromike

gasparirob, 2021-05-01:
What is the proper device.js?

hakha4, 2021-11-20:
Newbie with this. Same problem with cc2531 with antenna.  Flashed with CC2531_router_2020_09_29. Can someone give some info about syntax to add devices.js to data_path and what 'proper' device.js should be. Thank's in advance

Corinake, 2022-11-30:
> looks like zigbee2mqtt without devices.js is not able to recognize cc2531 router. after adding proper `devices.js` to data_path and adding `zigbee_shepherd_devices: true` to config the cc2531 is recognized as cc2530 and interview correctly.

Hi, can you tell us how you do that? "adding proper `devices.js` to data_path". Thank you

bigretromike, 2022-12-01:
sorry, been some time ago, it used to be this file: https://github.com/Koenkk/zigbee2mqtt/blob/3fdfc6f57f18d26a20a69619c5b51eb5d6d407b2/lib/devices.js I had to add CC2531 to this file to be able to use it, but its been quite some time and im unsure if this is also solution for same problem with current software.

mrneutron42, 2023-10-23:
I've got the same thing going on in 2023.
I've got a CC2531 USB stick with 20211115 version of CC2531ZNP-Prod.hex and it's working for Zigbee devices.
I got another CC2531 USB stick and loaded it with the 2020_09_29 version of router-cc2531-std.hex.
The CC2531 USB board with router firmware in it blinks the red LED once per second when powered up - indicating it's in pairing mode.  I put the CC2531 USB stick into pairing mode from inside Home Assistant and the messages show it's waiting for devices to pair.  The CC2531 USB board with router firmware never pairs!
So, I loaded the CC2531 USB stick with the 2020_09_29 version of router-cc2531-diag-usb.hex.
I put the CC2531 USB board with router firmware into a USB port on my Windwos 10 machine and a COM5 port showed up in the Windows Device Manager....good.
When I open PUTTY to view the serial port info, I see this over and over.
```
PANID:F5B,Ch:15,Join:0

PANID:F5B,Ch:15,Join:0

PANID:F5B,Ch:15,Join:0
```
I'm not sure what to do to force it to pair with the CC2531 USB coordinator??

mrneutron42, 2023-10-24:
I've got the same thing going on in 2023 that @bigretromike had in December 2020.
I've got a CC2531 USB stick with 20211115 version of CC2531ZNP-Prod.hex and it's working for Zigbee devices.
I got another CC2531 USB stick and loaded it with the 2020_09_29 version of router-cc2531-std.hex.
The CC2531 USB board with router firmware in it blinks the red LED once per second when powered up - indicating it's in pairing mode.  I put the CC2531 USB stick into pairing mode from inside Home Assistant and the messages show it's waiting for devices to pair.  The CC2531 USB board with router firmware never pairs!
So, I loaded the CC2531 USB stick with the 2020_09_29 version of router-cc2531-diag-usb.hex.
I put the CC2531 USB board with router firmware into a USB port on my Windwos 10 machine and a COM5 port showed up in the Windows Device Manager....good.
When I open PUTTY to view the serial port info, I see this over and over.
```
PANID:F5B,Ch:15,Join:0

PANID:F5B,Ch:15,Join:0

PANID:F5B,Ch:15,Join:0
```
I'm not sure what to do to force it to pair with the CC2531 USB coordinator??

ptvoinfo, 2023-10-24:
@mrneutron42 Your device may try to connect to a router or another coordinator with a higher signal level. 

Try to move your device closely to the coordinator.

mrneutron42, 2023-10-24:
During this setup testing the router is 7 feet away from the coordinator and there are no physical obstructions.
I am sure this is not an RF signal strengh problem.  I'd say the focus needs to be on the software and firmware.

For example, is there are interoperation problem with the 2 versions of cooridinator and router firmware I've got loaded?

mrneutron42, 2023-10-26:
I got another [CC2531 USB stick](https://www.amazon.com/gp/product/B081WRGL31/ref=ppx_yo_dt_b_search_asin_title?ie=UTF8&psc=1) and loaded the 2020_09_29 version of router-cc2531-diag-usb.hex into it.
SAME results - it has the constant red LED blinking (join mode) but it never joins the CC2531 coordinator's Zigbee network, after clicking the "Add Devices via this device" button in the ZHA Home Assistant.
[image]

```
PANID:F5B,Ch:15,Join:0

PANID:F5B,Ch:15,Join:0

PANID:F5B,Ch:15,Join:0

PANID:F5B,Ch:15,Join:0

PANID:F5B,Ch:15,Join:0

PANID:F5B,Ch:15,Join:0

PANID:F5B,Ch:15,Join:0

PANID:F5B,Ch:15,Join:0
```

bigretromike, 2023-11-18:
Last time I played with my device the issue was also that it had that internal antena (print on mobo, not real wire antena) and the range was very small (1-3meters in plain sight), that was also issue that would prevent me from joining zigbee devices.

mrneutron42, 2023-11-18:
RF strength isn't likely my problem.  I'm using the the USB units with the SMA connector and external antenna.  
[image]
Both the USB CC2531 coordinator and the USB CC2531 cordinator has the external antenna.
And cooridinator is easily communicating with the IKEA TRADFRI repeaters that are 5-10 meters away.