Koenkk/Z-Stack-firmware #244: continuously send status messages im short intervals

ralf-e, 2021-01-08:
Is it normal behavior for a ZigBee router to continuously send status messages at very short intervals (every second)?
I observed that on the router this version was flashed: CC2530_CC2591_router_2020_09_29

13: 24: 11.104 RSL: tele / tasmota_D7F444 / SENSOR = {"ZbReceived": {"0x95A4": {"Device": "0x95A4", "Name": "CC2530CC2591Router_0x95A4", "BinaryDescription": "32814 /
13: 24: 13.726 RSL: tele / tasmota_D7F444 / SENSOR = {"ZbReceived": {"0x95A4": {"Device": "0x95A4", "Name": "CC2530CC2591Router_0x95A4", "Power": 1, "BinaryDescriptio
13: 24: 15.722 RSL: tele / tasmota_D7F444 / SENSOR = {"ZbReceived": {"0x95A4": {"Device": "0x95A4", "Name": "CC2530CC2591Router_0x95A4", "Power": 1, "BinaryDescriptio
13: 24: 16.106 RSL: tele / tasmota_D7F444 / SENSOR = {"ZbReceived": {"0x95A4": {"Device": "0x95A4", "Name": "CC2530CC2591Router_0x95A4", "BinaryDescription": "32814 /
13: 24: 18.726 RSL: tele / tasmota_D7F444 / SENSOR = {"ZbReceived": {"0x95A4": {"Device": "0x95A4", "Name": "CC2530CC2591Router_0x95A4", "Power": 1, "BinaryDescriptio
13: 24: 20.722 RSL: tele / tasmota_D7F444 / SENSOR = {"ZbReceived": {"0x95A4": {"Device": "0x95A4", "Name": "CC2530CC2591Router_0x95A4", "Power": 1, "BinaryDescriptio

Koenkk, 2021-01-11:
AFAIK this only happens with the `-diag` firmware, so try the `-std` firmware.

ralf-e, 2021-01-11:
I'm not 100% sure, was a few days ago and the FW has now been replaced with another. But I think I used the _std.

ptvoinfo, 2021-01-12:
@ralf-e Only the "diag" firmware outputs data from your first message.

sunfos2019, 2021-01-12:
When the problem occurs, steps:

1. Flash E18-MS1PA1-PCB (CC2530+CC2592) with router-cc2530-cc2592-std.bin (latest). In my case I tested 2 routers - based on E18-MS1PA1-PCB (router-cc2530-cc2592-std.bin) and E18-MS1PA2-PCB (router-cc2530-cc2590-std.bin) - both show the same behavior.
2. Join flashed router Zigbee2mqtt network. In my case it is based on CC2538+CC2592 coordinator.
3. Joining Zigbee2mqtt network was needed to confirm that everything works fine - router sends its state every 60 seconds.
4. Unpair router from Zigbee2mqtt network.
5. Join router zigbee network based on D1 mini (Tasmota 9.2.0 core 2.7.4.9) + E18-MS1PA1-PCB (CC2530+CC2592, Coordinator Z-Stack_Home_1.2 Default).
6. After joining Tasmota network router starts sending its state every second.
7. Unpair router from Tasmota network.
8. Join router Zigbee2mqtt network again.
9. Router starts sending its state every second.

It seems that Tasmota reconfigures router from 60 to 1 second. And configuration stays in router flash.
After reflashing E18-MS1PA1-PCB with the same router-cc2530-cc2592-std.bin and joining Zigbee2mqtt again it sends state every 60 seconds as expected.

sunfos2019, 2021-01-12:
> @sunfos2019 
> Confirm that on CC2530_CC2592_router_2020_09_29. Sends its state every second.
> CC2530_router_2020_09_29 sends its state every 60 seconds.

CC2530 was not inside Tasmota network so it works as expected.

sunfos2019, 2021-01-12:
My Tasmota log when sending state every second:

13:03:43 MQT: tele/D1_mini_03/6DAA/SENSOR = {"0x6DAA":{"Device":"0x6DAA","Name":"ZigBee_Router_02","Power":1,"Endpoint":8,"LinkQuality":26}} (retained)
13:03:44 MQT: tele/D1_mini_03/6DAA/SENSOR = {"0x6DAA":{"Device":"0x6DAA","Name":"ZigBee_Router_02","Power":1,"Endpoint":8,"LinkQuality":26}} (retained)
13:03:45 MQT: tele/D1_mini_03/6DAA/SENSOR = {"0x6DAA":{"Device":"0x6DAA","Name":"ZigBee_Router_02","Power":1,"Endpoint":8,"LinkQuality":26}} (retained)
13:03:46 MQT: tele/D1_mini_03/6DAA/SENSOR = {"0x6DAA":{"Device":"0x6DAA","Name":"ZigBee_Router_02","Power":1,"Endpoint":8,"LinkQuality":26}} (retained)

juan11perez, 2021-02-04:
I'm observing the same behaviour. 
I flashed the CC2530 + CC2591 with CC2530_CC2591_router_2020_09_29.zip
After joining the tasmotized sonoff bridge, it reports status every second eventually killing the coordinator.

juan11perez, 2021-02-05:
@ralf-e 
@sunfos2019 

I was able to fix the reporting issue using below  firmware configurator produced by @ptvoinfo and shared below:

https://github.com/Koenkk/Z-Stack-firmware/issues/222

Some guidance on how to configure in this related thread:

https://github.com/ptvoinfo/zigbee-configurable-firmware/issues/8

closed 2021-03-15 by github-actions[bot]

m0m4x, 2022-06-03:
For further reference.

To solve this issue you need to give this in the tasmota console:
`ZbSend {"Device":"0xXXXX","Config":{"Power":{"MinInterval":60,"MaxInterval":3600}}}`

Bye :)