Koenkk/Z-Stack-firmware #247: CC2531 & CC2530

dkarthickch, 2021-01-20:
Hi I have used the HA 1.2 Source routing firmware in CC2531 and CC2530+CC2591 in 2 different hardware, initially i could able to add 30+ lighting modules (Zigbee 3.0), but later on I am unable to add any other modules, even after unpairing some modules I am unable to add further devices. Do you have any suggestion for this?

allofmex, 2021-01-21:
What I noticed (cc2531, 1.2), besides the normal reboots:

- Sometimes I need to unplug cc2531 for longer than just a few seconds (to let it completely power-down internally??)
- pairing often works only during first minutes after startup
- sometimes it seems to help, to power-down some(/most) other router during pairing to reduce network load
- and every few month I need to reflash cc2531, because it does not start up at all anymore (usually after some host crash)

But I agree, pairing is sometimes not very fun :-/

dkarthickch, 2021-01-21:
> What I noticed (cc2531, 1.2), besides the normal reboots:
>
> * Sometimes I need to unplug cc2531 for longer than just a few seconds (to let it completely power-down internally??)
> * pairing often works [...]

For my case if i start adding it keeps on adding and in case of removing and adding again its not happening, the ztool shows some unsupported format messages, seems like the end device is sending some beacon messages but the coordinator unable to pair it.

closed 2021-02-28 by github-actions[bot]