Koenkk/Z-Stack-firmware #249: CC2652R Flashed but wont pair

jimmyeao, 2021-01-27:
Hi, fantastic to see router firmware for the CC2652, however, after flashing i am unable to pair it with my CC2531 coordinator (ZAigbee2mqtt) Nothing appears in the logs of Zigbee2mqtt.

How could I troubleshoot this?

Thanks :)

Koenkk, 2021-01-27:
- Did you click the button your CC2652 and wait 20 seconds after that?
- If that doesn't work sniff the traffic while trying to join: https://www.zigbee2mqtt.io/how_tos/how_to_sniff_zigbee_traffic.html

jimmyeao, 2021-01-27:
Reflashed and clicked (I think I was missing the single click after flashing) and it has now paired, thanks :)

closed 2021-01-27 by jimmyeao

Theminijohn, 2021-09-02:
Which button was that @jimmyeao ?

jimmyeao, 2021-09-02:
The button on the stick, I use this one https://electrolama.com/projects/zig-a-zig-ah/