Koenkk/Z-Stack-firmware #251: CC2652R wont join after a reset

jimmyeao, 2021-01-28:
I moved my CC2652R stick flashed with the 1.0.0 router firmware, and it refused to reconnect to zigbee2mqtt, nothin in the logs showing it even trying.
I have tried:
Removing it an re-adding it (Never shows in logs)
A touch link reset which fails:
```
Jan 28 16:18:40 raspberrypi npm[29463]: Zigbee2MQTT:info  2021-01-28 16:18:40: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"reset_started","meta":{"status":"started"},"type":"touchlink"}'
Jan 28 16:18:51 raspberrypi npm[29463]: Zigbee2MQTT:warn  2021-01-28 16:18:51: Failed to factory reset device through Touchlink
Jan 28 16:18:51 raspberrypi npm[29463]: Zigbee2MQTT:info  2021-01-28 16:18:51: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"reset_failed","meta":{"status":"failed"},"type":"touchlink"}'

```
and re-flashing and resetting, which again fails.
I have also left it unplugged over night to see if it would force the network to 'forget' the device and allow it to be reset.
I don't have a spare CC2531 (yet) to flash with the sniffer firmware, is there anything else I can try?
Willing to re-do from scratch and re-pair everything if that is necessary

thanks :)

Koenkk, 2021-01-28:
If nothing shows in the log the only way to debug is to sniff :(

jimmyeao, 2021-01-28:
Ok, ill get another cc2531 ordered, then Ill report back
any tips on how to sniff/what to look for?

Koenkk, 2021-01-28:
Just send me the whole sniff when pressing the button on the CC2652R: https://www.zigbee2mqtt.io/how_tos/how_to_sniff_zigbee_traffic.html

jimmyeao, 2021-01-28:
Should be here monday, Ill take a look then :) thanks!

copystring, 2021-02-01:
My LAUNCHXL-CC26xR1 does this too. Flashed the router firmware 2-3 times. Also erased entire flash via UniFlash.
Here's my log from sniffing. In the log I pressed the button a few times. Also re-plugged it a lot.

[log.zip](https://github.com/Koenkk/Z-Stack-firmware/files/5906387/log.zip)

Koenkk, 2021-02-02:
@copystring try pairing the device close to the coordinator. In your sniff it's trying to join through a Philips device 0x0017880108556c81 but it's not sending the network key.

copystring, 2021-02-02:
Nice. Works now.

jimmyeao, 2021-02-04:
Mine is now also working thanks, I was unable to get the sniffer to work under windows, but moved it closer to the coordinator and it managed to eventually join :)

closed 2021-02-04 by jimmyeao