Koenkk/Z-Stack-firmware #256: Led(s) on Z-Stack_3.x.0

Gio-dot, 2021-02-05:
Hi, i noticed that in 3.x.0 firmware table (https://github.com/Koenkk/Z-Stack-firmware/tree/master/coordinator/Z-Stack_3.x.0/bin) are reported leds on various boards. I'm working on a cc2652P module and noticed that leds on dio_7 and dio_8 are not working. Tried egony firmware: leds works as expected. I'm missing something or leds on your firmware are not yet implemented?

Koenkk, 2021-02-06:
Indeed not supported for now, similar issue: https://github.com/Koenkk/Z-Stack-firmware/issues/203

Gio-dot, 2021-02-17:
> Indeed not supported for now, similar issue: #203

Do you think you will add LED support in the future?

Koenkk, 2021-03-18:
Let's continue in https://github.com/Koenkk/Z-Stack-firmware/issues/203

closed 2021-03-18 by Koenkk