Koenkk/Z-Stack-firmware #257: Support for Ebyte E79 module (CC1352P with a different RF arrangement)

omerk, 2021-02-05:
Hi @Koenkk,

The Ebyte E79 modules have a different RF arrangement than the current `CC1352P2_CC2652P_other_coordinator` firmware, the PA is used for sub-GHz instead of 2.4GHz (the opposite of the configuration in the "-2" Launchpad") so the RF Switch is used only for the sub-GHz operation (PA output + sub-GHz path) -- E2E discussion [here](https://e2e.ti.com/support/wireless-connectivity/sub-1-ghz/f/156/t/952841?CC1352P-Overrides-for-20dBm-434MHz-Ebyte-E79-module-). 

I think this effectively means that we can just use the CC2652R firmware for this radio (I am going to be testing this over the weekend), except the BSL pin is now a problem as we decided that `DIO_15` was to be used on all CC1352P/CC2652P designs and burning the CC2652R firmware would disable BSL as it is setting the BSL trigger to be `DIO_13`. Not great.

What's the best way to move forward with this, as this particular module kind of breaks the whole "by chip" firmware structure this repo has?

I'll prepare a test setup and report back here.

Koenkk, 2021-02-06:
I think it's unavoidable to have a separate firmware for this right? That means either we don't support it or we have to come up with a different structure of the archive. This also means a deviation in compilation instructions has to be added for this device here: https://github.com/Koenkk/Z-Stack-firmware/blob/master/coordinator/Z-Stack_3.x.0/COMPILE.md .

omerk, 2021-02-06:
Looks like I was wrong in my assumption that the CC2652R firmware would work on this module:

With `CC2652R_coordinator_20210120.hex`:

```
./znp-uart-test.py
Got 0 bytes in response to PING command: b''
FAIL|Expected length mismatch
```

With `CC1352P2_CC2652P_other_coordinator_20210120.hex`:

```
./znp-uart-test.py
Got 7 bytes in response to PING command: b'\xfe\x02a\x01Y\x06='
PASS|OK
```

Not entirely sure why this fails, though? Haven't run it long enough to assess the RF performance but I am guessing it is not going to be optimal as this firmware, as far as I can tell, uses the PA for TX (as the PA is set up for the 2.4GHz on "-2" LP), right? On this module we are limited to the standard (5dBm) 2.4GHz path.

So the current `CC1352P2_CC2652P_<type>_coordinator_<date>.hex` is essentially "Coordinator firmware for CC1352P and CC2652P, with PA set up for 2.4GHz TX (20dBm?)" and we will need another one that would be "Coordinator firmware for CC1352P and CC2652P with PA disabled". Alternatively, and if we can figure out why the CC2652R firmware is not running: A version of that firmware with the BSL pin changed to `DIO_15`?

Bit messy but would appreciate your support on this, as usual 🙂

egony, 2021-02-10:
When TX power sets up to 5 - PA is not used. So you can use any CC1352P2_CC2652P_*_coordinator firmware with 

experimental:
  transmit_power: 5

settings in config (but default settings in CC1352P2_CC2652P_*_coordinator firmware already 5, so you need do nothing).
Why you whant to use 2652R firmware???

closed 2021-03-20 by github-actions[bot]