Koenkk/Z-Stack-firmware #258: CC2652RB does not start with CC2652RB_coordinator_20210120

HVPereira, 2021-02-06:
Hello,

I've flashed the latest coordinator firmware and my zigbee2mqtt is failing to start now. 

I am using zigbee2mqtt version 1.17.1 and Home Assistant core 2021.2.1 
[image]

Below are my findings:

`~ $ ls -la /dev/serial/by-id/
total 0
drwxr-xr-x    2 root     root            60 Feb  6 14:30 .
drwxr-xr-x    3 root     root            60 Feb  6 14:30 ..
crw-rw----    1 root     audio     188,   0 Feb  6 14:30 usb-Silicon_Labs_slae.sh_cc2652rb_stick_-_slaesh_s_iot_stuff_00_12_4B_00_21_CC_42_0D-if00-port0`

I have tried to connect it directly to my pi3 with hass.io and via usb hub (mains powered), below are the logs of the usb hub:

`[ 3385.662763] usb 1-1.5.2: new full-speed USB device number 12 using dwc_otg
[ 3385.800743] usb 1-1.5.2: New USB device found, idVendor=10c4, idProduct=ea60, bcdDevice= 1.00
[ 3385.800756] usb 1-1.5.2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[ 3385.800764] usb 1-1.5.2: Product: slae.sh cc2652rb stick - slaesh's iot stuff
[ 3385.800771] usb 1-1.5.2: Manufacturer: Silicon Labs
[ 3385.800778] usb 1-1.5.2: SerialNumber: 00_12_4B_00_21_CC_42_0D
[ 3385.803836] cp210x 1-1.5.2:1.0: cp210x converter detected
[ 3385.807527] usb 1-1.5.2: cp210x converter now attached to ttyUSB0`

Below you can find my configuration for zigbee2mqtt:

`data_path: /share/zigbee2mqtt
external_converters: []
devices: devices.yaml
groups: groups.yaml
homeassistant: true
permit_join: false
mqtt:
  base_topic: zigbee2mqtt
  server: 'mqtt://core-mosquitto'
  user: XXXXXXXXXXX
  password: XXXXXXXXX
serial:
  port: /dev/serial/by-id/usb-Silicon_Labs_slae.sh_cc2652rb_stick_-_slaesh_s_iot_stuff_00_12_4B_00_21_CC_42_0D-if00-port0
advanced:
  log_level: info
  pan_id: 6754
  channel: 11
  network_key:
    - 1
    - 3
    - 5
    - 7
    - 9
    - 11
    - 13
    - 15
    - 0
    - 2
    - 4
    - 6
    - 8
    - 10
    - 12
    - 13
  availability_blocklist: []
  availability_passlist: []
device_options: {}
blocklist: []
passlist: []
queue: {}
frontend:
  port: 8099
experimental:
  new_api: true
socat:
  enabled: false
  master: 'pty,raw,echo=0,link=/dev/ttyZ2M,mode=777'
  slave: 'tcp-listen:8485,keepalive,nodelay,reuseaddr,keepidle=1,keepintvl=1,keepcnt=5'
  restartdelay: 1
  initialdelay: 1
  options: '-d -d'
  log: false
`
And finally the error logs. 

`[s6-init] making user provided files available at /var/run/s6/etc...exited 0.
[s6-init] ensuring user provided files have correct perms...exited 0.
[fix-attrs.d] applying ownership & permissions fixes...
[fix-attrs.d] done.
[cont-init.d] executing container initialization scripts...
[cont-init.d] done.
[services.d] starting services
[services.d] done.
./run.sh: line 17: [Info] Configuration backup found in /share/zigbee2mqtt/.configuration.yaml.bk. Skipping config backup.: No such file or directory
[Info] Socat is DISABLED and not started
2021-02-06T15:37:55: PM2 log: Launching in no daemon mode
2021-02-06T15:37:55: PM2 log: App [npm:0] starting in -fork mode-
2021-02-06T15:37:56: PM2 log: App [npm:0] online
> zigbee2mqtt@1.17.1 start /zigbee2mqtt-1.17.1
> node index.js
Zigbee2MQTT:info  2021-02-06 15:38:05: Logging to console and directory: '/share/zigbee2mqtt/log/2021-02-06.15-38-01' filename: log.txt
Zigbee2MQTT:info  2021-02-06 15:38:05: Starting Zigbee2MQTT version 1.17.1 (commit #1.17.1)
Zigbee2MQTT:info  2021-02-06 15:38:05: Starting zigbee-herdsman (0.13.59)
Zigbee2MQTT:error 2021-02-06 15:38:28: Error while starting zigbee-herdsman
Zigbee2MQTT:error 2021-02-06 15:38:28: Failed to start zigbee
Zigbee2MQTT:error 2021-02-06 15:38:28: Exiting...
Zigbee2MQTT:error 2021-02-06 15:38:28: Error: SRSP - SYS - osalNvRead after 6000ms
    at Timeout._onTimeout (/zigbee2mqtt-1.17.1/node_modules/zigbee-herdsman/dist/utils/waitress.js:46:35)
    at listOnTimeout (internal/timers.js:554:17)
    at processTimers (internal/timers.js:497:7)
npm
 ERR! code ELIFECYCLE
npm ERR! errno 1`

Sorry for not giving more information but I really don't know what else to share or do.

cobra1037, 2021-02-08:
i had the same problem - unplug the cc2652 -> remove the antenna and plug the cc2652 without antenna in the usb port.

then it was starting! after that you can gave the antenna on the stick.
i dont know why, but here it works.

countrysideboy, 2021-02-09:
I also i had the same problem of cc2652p(E72).

johanvh83, 2021-02-11:
Guys, is it save to flash slaesh CC2652 with CC2652RB_coordinator_20210120?
Want to pair/bind a new device and it doesn't work. Some recommend to reflash the CC2652 but based on above info I'm a bit sceptic ;-)

ERICRAC, 2021-02-11:
Flash done, it work well for me with ZigPy

johanvh83, 2021-02-11:
I spend many hours to find out how to upgrade the firmware but I have a basic education in these things. Can someone please help me? I really have no idea how and where to start... Apologies if this is not the right place to ask support but I don't know where else.

closed 2021-03-22 by github-actions[bot]