Koenkk/Z-Stack-firmware #259: How to update without loosing any config

ProfDrYoMan, 2021-02-08:
Hi,

I know, this is no support forum, but I did not know where to ask the question.

I want to update my coordinator 3.x.0 version to the lastet. What do I need to ensure to keep all config/links/...?

Thanks.

Any pointers to the proper forum would be nice.

Koenkk, 2021-02-08:
If using Zigbee2MQTT; just reflash your coordinator and Zigbee2MQTT will automatically restore a backup (https://www.zigbee2mqtt.io/information/FAQ.html#doesnt-require-repairing). If using anything else please open an issue at their project page.

closed 2021-02-08 by Koenkk

ProfDrYoMan, 2021-02-09:
As info for my use case with HomeAssistant and any other user of zigphy: https://github.com/zigpy/zigpy-znp#nvram