Koenkk/Z-Stack-firmware #260: zzh! CC2652R, problems with router firmware

gabric098, 2021-02-10:
Current setup:
- Zigbee2MQTT version 1.17.1
- zigbee-herdsman (0.13.59)
- 1 zzh! stick as coordinator
```
Coordinator firmware version: 
'{"meta":{"maintrel":1,"majorrel":2,"minorrel":7,"product":1,"revision":20210120,"transportrev":2},"type":"zStack3x0"}'
```
1 zzh! stick as router flashed with with CC2652R_router_20210128.zip

Coordinator is working fine, I can see all my zigbee devices showing up correctly but **I cannot pair the zzh! stick flashed with router firmware**.

These are the steps I follow:
- zzh! router stick is unplugged
- from zigbee2mqtt web interface I permit join.
- I plug the zzh! router stick into a usb power brick and I wait
- I don't see any joining requests coming from the router stick
- I press the little button on the zzh! router stick
- I still don't see any joining requests coming from the router stick

Notes:
I don't see any led on the zzh! router stick (but I believe that's normal)

I flashed the router firmware twice to make sure the flashing process was correct, and in both cases the flashing/verifying message was successful, so I'm excluding a problem with the flashing process.

Does anyone has the same issue?

Any idea on how to debug further to understand where the problem lies?

Thanks,
Gab

zajnic, 2021-02-12:
Same problem here, flashed router firmware and can't see it pairs with the coordinator.
No LED indicator on ZZH adapter,, keen to know how to resolve it. Thanks.

thiscantbeserious, 2021-02-21:
Its also non-functional on slae.sh's CC2652RB ... so its not just one image being non-functional - albeit the led turns green it does never connect, also doesn't react on any button-press.

gabric098, 2021-02-21:
@thiscantbeserious I just got a couple of slaesh sticks and I'm going to give it a go later today with the router firmware, just to confirm the issue.

gabric098, 2021-02-21:
@thiscantbeserious I just tried and it worked for me using the slaesh stick.
I flashed the stick with this firmware:
https://github.com/Koenkk/Z-Stack-firmware/raw/master/router/Z-Stack_3.x.0/bin/CC2652RB_router_20210128.zip
Permitted join from the web interface and plugged the stick into a power supply. 
The join process started straight away and the router showed up in the web interface as you can see in the screenshot below:

[image]

The issue with zzh! stick though, still persist.

Thanks,
Gab

TSKnuhtsen, 2021-02-27:
I've been able to replicate this bug as well.

Firmware flashes just fine via UNIFLASH but I am unable to pair with ZZH coordinator.

My need is urgent so I am instead going to leverage 5 - 6 more IKEA TRADFRI outlets to remediate.

In the future, I would love to be able to slap 1 ZZH Coordinator + 2 ZZH Routers to support my larger installs (e.g. 150+ sensors).

Koenkk, 2021-02-27:
Please provide a sniff while trying to pair (+ network key): https://www.zigbee2mqtt.io/how_tos/how_to_sniff_zigbee_traffic.html

TSKnuhtsen, 2021-03-01:
ZZH is currently on client site. Will provide sniff report for you on Monday when I head back to facility.

adampetrovic, 2021-03-24:
Also adding to the woes. Flashes fine using `cc2538-bsl.py` method, but no LED on the device after plugging into an apple wall charger

makonde-on-git, 2021-03-26:
Same for me. Flashing without any issues, but no LED, and not pairing with z2m :(

gabric098, 2021-04-23:
I was trying to sniff the traffic during the zzh! router join process when, all of a sudden the router decided to join the network successfully so I have 2 sniffs: the first is the failed one, the second is a successful join.
It's the fist time I do this all sniff process so, I'm not 100% sure that the files are correct. 

[zzh_capture.zip](https://github.com/Koenkk/Z-Stack-firmware/files/6367992/zzh_capture.zip)

HTH

gabric098, 2021-04-24:
I managed to successfully and consistently pair the zzh! stick a couple of times with `CC2652R_router_20210128` firmware. So I'm going to close the issue as it seems that it's not happening anymore.

closed 2021-04-24 by gabric098

rwunsch, 2022-01-17:
I am facing the same problem as described above - ZZH not connecting after successful flashing with CC2652R_router_20210128 to Zigbee2mqtt.
The ZZH stick does not seem to have an LED - so one does not know at all what is going on.
I tried every trick in my box to get it paired - but failed to see it so far. 
Sniffing was so far out of my league :) - need to see if I find some time on the week-end.

What was the "trick" which made the ZZH connect?

daha76, 2022-02-10:
Today I tried to ZZH CC2652R as a router. I flashed CC2652R_router_20210128 and I have problem to pair it with coordinator. Behavior is the same as it has been reported by [rwunsch](https://github.com/rwunsch)

Any tips/tricks?

digiblur, 2022-02-10:
> Today I tried to ZZH CC2652R as a router. I flashed CC2652R_router_20210128 and I have problem to pair it with coordinator. Behavior is the same as it has been reported by [rwunsch](https://github.com/rwunsch)
> 
> Any tips/tricks?

Reset the NVRAM with the zig tools script is what I had to do one time.

daha76, 2022-02-11:
> 
> Reset the NVRAM with the zig tools script is what I had to do one time.

Thanks, that helped.

daha76, 2022-02-17:
> > Today I tried to ZZH CC2652R as a router. I flashed CC2652R_router_20210128 and I have problem to pair it with coordinator. Behavior is the same as it has been reported by [rwunsch](https://github.com/rwunsch)
> > Any tips/tricks?
> 
> Reset the NVRAM with the zig tools script is what I had to do one time.

Reset helped, but there is still a problem after power-up. When I simply reconnect zzh! usb stick then it will not join the network, I have to manually push the button. I have no idea how to force it to connect to the network after power cycle.

Any idea?

w0lf3h, 2022-11-10:
Having this issue on my end as well with the CC2652R as a router. @daha76 did resetting NVRAM work for you? I tried to do it earlier but was encountering an error and have to head over to work now, so I'll look into it later as well. 

I've tried reflashing the firmware multiple times as well as changing the secondary IEEE in case of conflict.

w0lf3h, 2022-11-10:
> > Reset the NVRAM with the zig tools script is what I had to do one time.
> 
> Thanks, that helped.

Having this issue on my end as well with the CC2652R as a router. @daha76 did resetting NVRAM work for you? I tried to do it earlier but was encountering an error and have to head over to work now, so I'll look into it later as well.

I've tried reflashing the firmware multiple times as well as changing the secondary IEEE in case of conflict.

daha76, 2022-11-11:
> Having this issue on my end as well with the CC2652R as a router. @daha76 did resetting NVRAM work for you? I tried to do it earlier but was encountering an error and have to head over to work now, [...]

Yes NVRAM cleaning helped.