Koenkk/Z-Stack-firmware #261: Question. When should I reflash the coordinator firmware?

aleksei-panin, 2021-02-12:
Currently, the version of the coordinator firmware installed on my zzh! stick is CC26X2R1_20200805. My question is when and in which case do I need to update the firmware? The version of Zigbee2mqtt is 1.14.4. I am planning the update to the latest version to support the recently added gledopto light stripe GL-MC-001P (support was added in version 1.17.0). Together with Z2M update, do I need to update the zzh! firmware to support new devices?

Koenkk, 2021-02-13:
> firmware to support new devices?

Nope!

Generally speaking only update when you are having problems. (or just like to be up-to-date :))

closed 2021-02-13 by Koenkk

aleksei-panin, 2021-02-17:
Great! Thanks for your reply!