Koenkk/Z-Stack-firmware #263: Corruption in CC2531 coordinator

jorticus, 2021-02-13:
PSA - I have discovered an issue which may be wide-spread among many users of the Z-Stack firmware. I was able to resolve this myself, but I am raising this issue for awareness..

---

I have been running the CC2531_DEFAULT_20190608 firmware for the past few months along with the native HomeAssistant ZHA integration, and suddenly found my CC2531 stick to no longer be responding.

HomeAssistant would time out trying to communicate, and DMESG showed the following in the log:
```
[   38.623909] usb 1-1: USB disconnect, device number 2
[   38.624209] cdc_acm 1-1:1.0: failed to set dtr/rts
[   38.939253] usb 1-1: new full-speed USB device number 5 using xhci_hcd
[   39.085625] usb 1-1: New USB device found, idVendor=0451, idProduct=16a8
[   39.085630] usb 1-1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[   39.085632] usb 1-1: Product: TI CC2531 USB CDC
[   39.085634] usb 1-1: Manufacturer: Texas Instruments
[   39.085636] usb 1-1: SerialNumber: __0X00124B0014DA3D7E
[   39.086680] cdc_acm 1-1:1.0: ttyACM0: USB ACM device
```
(See first two lines - something very wrong is happening inside the USB stack...)

On a hunch (since I deal with this a lot at my job) I dumped the firmware off the failing stick and flashed it onto a known-good stick. This new stick had the exact same symptoms and would not respond. Then I re-flashed the original stock firmware and it started working again, however I lost all my pairings/network keys.

This drove me to find a way to restore operation of the CC2531 while also keeping my pairings/network keys. Sure enough, non-volatile configuration is stored at the end of the flash at offset `C80000h`, so in theory you should be able to take this and append it to a stock HEX file to keep your network intact:
[image]
(left: stock firmware, right: dumped firmware)

Digging a bit deeper I came across the culprit:
[image]

You can see that there is corruption at offset `D92000h`. By simply copying the bytes from left to right I was able to fix my dumped firmware, and after flashing it back onto the CC2531 the stick was back in action as if nothing had gone wrong!

Now I have no idea why this corruption occurred, and it isn't located in the non-volatile region, but it would definitely cause things to go awry inside the firmware and cause all sorts of chaos...

---

Hopefully this information will help others who are also experiencing problems with this firmware/hardware combo. 

It seems this isn't an isolated occurrence, and others may be running into similar corruption on their devices, since there are a lot of anecdotes about only being able to fix things by re-flashing stock firmware:

- https://github.com/Koenkk/Z-Stack-firmware/issues/122
- https://github.com/Koenkk/zigbee2mqtt/issues/2899
- https://old.reddit.com/r/homeassistant/comments/liaoh6/my_zigbee2mqtt_wont_find_my_cc2531_dongle_can_you/

Koenkk, 2021-02-13:
Thanks for analysing this, very interesting!

I however have very low hopes that Texas Instruments will every fix this bug. The CC253X is considered legacy since they moved on to the CC2652/CC1352.

PetersPlace, 2021-02-16:
I have exaly the same problem. reflash it run for 3 days and now iám again see corruption, stick will not start also zigbee2mqtt 

In my VM console i see Failed to set DTR/RTS.    After reflash it is gone for x days.

I did order this with 3.0 preinstalled, i moved to 1.2 beause i good not get all sensors connected.

However where is the old firmware 3.0 for CC2531 ?

barrymossel, 2021-02-16:
Am I correct this is only for the CC2531 and will (probably) not be addressed? What alternative would you suggest then?

Koenkk, 2021-02-17:
@PetersPlace you can find the 3.0 firmware here: https://github.com/Koenkk/Z-Stack-firmware/tree/master/coordinator/Z-Stack_3.0.x/bin . This is the only 3.0 firmware that I've ever released for the CC2530/CC2531.

@barrymossel this only applies to the CC2530 and CC2531 indeed. Why it happens is unclear. I suggest to switch to one of the adapters recommended here: https://www.zigbee2mqtt.io/getting_started/what_do_i_need.html#supported-zigbee-adapter . From a user perspective there are no expected differences between these 2 recommended adapters, however you can find the technical difference here: https://www.zigbee2mqtt.io/information/supported_adapters.html#notes (3rd point). Many users reported a very big improvement in speed, range and stability (rock solid) after switching to one of these adapters.

barrymossel, 2021-02-17:
> @PetersPlace you can find the 3.0 firmware here: https://github.com/Koenkk/Z-Stack-firmware/tree/master/coordinator/Z-Stack_3.0.x/bin . This is the only 3.0 firmware that I've ever released for the [...]

Thanks a lot Koen. I put myself on the waitlist for both and will order the one I'll receive first. For now a re-flash did the trick and everything is working nicely again. Let's hope it won't corrupt again too soon...

Keep up the good work! Especially your great "support"!

timmmmmmmmm, 2021-02-20:
I experienced a similar issue after upgrading my CC2531 coordinator to a source routing firmware. Flashing was successful, however my USB stick wasn't recognized anymore either on my Pi with Home Assistant and Windows PC. I tried to reflash again, but still got the same issue. Also the green LED wasn't lighting up anymore so I thought my CC2531 was fried.

In the flashing process I used the .bin file in [this package](https://github.com/Koenkk/Z-Stack-firmware/blob/master/coordinator/Z-Stack_Home_1.2/bin/source_routing/CC2531_SOURCE_ROUTING_20201128.zip). That resulted in the problems described above. After focussing on the .hex file, and turning it into a .bin with [these instructions](https://www.zigbee2mqtt.io/information/alternative_flashing_methods.html#prepare-the-firmware), I successfully flashed my CC2531 again and everything seems to work again. Maybe this process can prevent the corruption you found?

PetersPlace, 2021-02-21:
> I experienced a similar issue after upgrading my CC2531 coordinator to a source routing firmware. Flashing was successful, however my USB stick wasn't recognized anymore either on my Pi with Home [...]

Iám flashing with raspberry pi, so i use the Hex file
Tested with Z-Stack_Home_1.2 20201127/20201128
- CC2531_DEFAULT_20201127.zip
- CC2531_SOURCE_ROUTING_20201128.zip
- CC2531_20190425.zip  Z-Stack 3.0.x .bin firmware and CC2592. #94 

All 3 flashes same problem , after random 2-3 days HA can not connect with CC2531  rts/cts  error on the terminal window
I have now been advised not to use **ttyACM0** but **/ dev / serial / by-id / usb-Texas_Instruments_TI_CC2531_xxxxxxxxxx**

Because I want to expand anyway, I'm going to use a new version like https://electrolama.com/projects/zig-a-zig-ah/
Since the CC2531 is more then 12 years old  and as mentioned by others this corruption will not be fixed by TI

jorticus, 2021-03-16:
I'm planning on moving to the CC2652RB, since I fully expect this corruption to occur again in the future. @Koenkk Feel free to close this issue.

For those that must stick to the CC2531 for whatever reason, you can use my above documented procedure to re-flash without losing your network bindings.

Tche333, 2021-03-16:
Thanks @jorticus , @PetersPlace , @timmmmmmmmm , @Koenkk  for having taken time to dig and document this issue. There are for sure still some on this legacy stick this information would be useful to them (as I'm for the time being)

emilm, 2021-04-04:
I have a lot of differences between the backup and the image flashed. I first flashed 1.2 and then 3.0 on my CC2531 . Is it corruption or are some regions never cleared?

closed 2021-05-12 by github-actions[bot]