Koenkk/Z-Stack-firmware #265: [Feature request] Higher baud rate for CC2652

m4tek, 2021-02-15:
Hey,

In networks where there is more traffic bringing possibilities of higher interface throughput could help with stability.

CC2652R supports up to 3Mbps (https://www.ti.com/lit/ds/symlink/cc2652r.pdf?ts=1613387764521)

CH340 supports common baud rates: 50, 75, 100, 110, 134.5, 150, 300, 600, 900, 1200, 1800, 2400, 3600, 4800, 9600, 14400, 19200, 28800, 33600, 38400, 56000, 57600, 76800, 115200, 128000, 153600, 230400, 460800, 230400, 1500000, 2000000 baud
(https://www.insidegadgets.com/wp-content/uploads/2016/12/ch340g-datasheet.pdf)

Do you think there is practical sense on increasing baud rate to 230400 or higher?

Koenkk, 2021-02-16:
Pinging @omerk here

omerk, 2021-02-16:
Do we have any indication that the standard baud rate is insufficient and causing stability problems?

m4tek, 2021-02-16:
I have nothing more than anecdotal data.

I'm running a network of 40+ nodes and it is very stable during normal operations, but few times a day I observe glitches that coincide with setting different heating profile on all TRVs. This happens rarely and I need to issue other command at the same time to notice a delay.

Recently I noticed also much delayed reaction to commands while OTA upgrade was running.

I have no means to check if this is MAC layer saturation or serial communication with the dongle itself (running zhh!), but considering that Zigbee MAC is 3Mbps, I asserted issue with the baudrate.

allofmex, 2021-02-16:
Are the TRVs Eurotronic's and are you controlling one of the system-mode settings when the glitches occur? (Window mode,..)
In that case I see glitches too, but I guess it is because each valves value is read first before new setting gets written. Seems to block other operations..

omerk, 2021-03-05:
@m4tek Did you manage to find a solution to this issue? Sounds like it might not be related to comms going with @allofmex's comment?

If you suspect it might be a comms bottleneck, you should be able to compile a version of the ZNP coordinator firmware following @Koenkk's [compilation notes](https://github.com/Koenkk/Z-Stack-firmware/blob/master/coordinator/Z-Stack_3.x.0/COMPILE.md) and setting the `baudrate` you've chosen in your zigbee2mqtt config.

closed 2021-04-12 by github-actions[bot]

Andrik45719, 2024-05-28:
My test build based on Koenkk firmware 20230507
460800bps
https://github.com/Andrik45719/Z-Stack-firmware