Koenkk/Z-Stack-firmware #269: Cc2530 router firmware

mamrai1, 2021-02-21:
Is the watchdog timer enabled in compiled router firmware?

ptvoinfo, 2021-02-21:
@mamrai1 No, the watchdog timer is not used in the router firmware.

mamrai1, 2021-02-21:
Also, why can't i open it with ptvoinfo configurable firmware app?
I want to assign a led to an output. I tried to use ptvinfo configurable firmware and create one but it's not seen as a light in phoscon/deconz..In phoscon in not shown at all. Your compiled firmware is seen as a light and working but really need a led indication to see if it's working

ptvoinfo, 2021-02-21:
PTVO configurable firmware is a different firmware. It can also work as a router, but Deconz may not support it.

mamrai1, 2021-02-24:
i managed to flash a cc2530+c2591 with ptvo configurable firmware and also enabled the watchdog timer along with a gpio for on/off for a led..i had to change the manufacture and manufacure id to 'lumi' and 'lumi.router' respectively. It works as expected now. It seems more stable now because with old firmware sometimes the cc2530 stuck at sending reports every second instead of every minute. Thanks

closed 2021-04-03 by github-actions[bot]