Koenkk/Z-Stack-firmware #273: TX power for CC1352P2 CC2652P router firmware

bkupidura, 2021-03-04:
Hey,
Can you please share what tx power is set in `CC1352P2_CC2652P_other_router_20210128.hex`?

Currently im tessting `CC2652P` as a router, it works stable but i think that it has lower radio tx power than my old `CC2530+CC2591` - which is surprising ;)

Koenkk, 2021-03-06:
5dbm

bkupidura, 2021-03-06:
Ok, will try to build new firmware. As i dont have any experience with CCS (and zigbee firmware), any hints how to adjust tx power before compilation?

r312645, 2021-03-15:
Can you please share your firmware？

bkupidura, 2021-03-15:
Sure, but please remember that this is "test" firmware, im dont know much about this board. I cant guarantee anything (except that it works in my network). But still power is better on `CC2530+CC2591`, can you confirm?

+ I added LED indicator.

[zb_firmware_CC1352P_2.zip](https://github.com/Koenkk/Z-Stack-firmware/files/6139567/zb_firmware_CC1352P_2.zip)

Gio-dot, 2021-03-27:
> Sure, but please remember that this is "test" firmware, im dont know much about this board. I cant guarantee anything (except that it works in my network). But still power is better on `CC2530+CC2591`, can you confirm?
> 
> Changes:
> 
> * TX power set to 20
> * Red LED always turn on
> 
> [zb_firmware_CC1352P_2.zip](https://github.com/Koenkk/Z-Stack-firmware/files/6139567/zb_firmware_CC1352P_2.zip)

Wich hardware are you using?

Alfiegerner, 2021-04-15:
@bkupidura - thanks, appears to be working for me.

I have two of these [CC2652P routers](https://www.tubeszb.com/product/cc_router/4?cp=true&sa=false&sbp=false&q=false&category_id=3), and the one flashed with your firmware is reporting approx double the LQI in the same location.

Finger crossed this change makes its way through.

bkupidura, 2021-04-18:
@Alfiegerner my firmware for sure will not be part of "official" zigbee2mqtt firmware available for CC1352P2. As it have some issues with introspection.

But maybe @Koenkk can just introduce two firmwares created by him (5 and 20 DB tx power).

cjsteward, 2021-04-20:
@Alfiegerner @bkupidura Have either of you had any issues since you've been running the modified firmware? I've installed it on a couple of tubes routers the other day and they seem to be working perfectly so far. And pardon my ignorance here @bkupidura , but what did you mean by "issues with introspection"?

Alfiegerner, 2021-04-21:
No issues from me @cjsteward, looking good. I don't any updates on last seen in the z2m frontend but I don't think I did pre either.  Network map show sitting connections and connected endpoints all good.

bkupidura, 2021-04-21:
@cjsteward board with that firmare is running in my network for last 1.5 month without issue.

Regarding introspection i mentioned, when you are pairing board with zigbee2mqtt, you will see error in log message about missing endpoint.  Except that error, everything else (i hope) is working.

cjsteward, 2021-05-02:
Just wanted to follow up here -  I've been running the modified firmware for over two weeks now and everything is still stable, but I just wanted to call out a few oddities I've noticed:

- As @bkupidura mentioned above, I also receive errors on startup:
`Zigbee2MQTT:error 2021-05-02 15:04:36: Failed to configure 'New Garage Repeater', attempt 1 (Error: ConfigureReporting 0x00124b00229839f4/8 genBasic([{"attribute":"zclVersion","minimumReportInterval":0,"maximumReportInterval":3600,"reportableChange":0}], {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Timeout - 34821 - 8 - 3 - 0 - 7 after 10000ms)
    at Timeout._onTimeout (/app/node_modules/zigbee-herdsman/dist/utils/waitress.js:37:35)
    at listOnTimeout (internal/timers.js:554:17)
    at processTimers (internal/timers.js:497:7))`

- Link quality is only reported on startup, I believe I read in another thread though that that's a known firmware issue.
- Another issue I've noticed is that I haven't been able to pair new devices. I swapped out my CC1352P-2 coordinator for a CC2652P from Tubes around the same time I put these routers into rotation, so I'm not exactly sure what's causing the issue there.

I'm hoping there will be production ready firmware released with the Tx power increase because besides the issues above, I've been really happy with the performance.

bkupidura, 2021-05-02:
@cjsteward regarding pairing issue, i dont think its related to router firmware. As in my network i added new enddevices while router with custom firmware is running. 

You can easly test that by just removeing router from network (and powering it off after that, to be sure).

closed 2021-06-10 by github-actions[bot]

Alfiegerner, 2021-06-10:
Missed stale bot - I think this would be a great feature.

TSKnuhtsen, 2021-06-14:
I'm using a CC1352P-2 LAUNCHPAD. 
The coordinator firmware works perfectly on my device.
But when I load @bkupidura 's firmware or firmware from the [Z2M Z-STACK Firmware Repo](https://github.com/Koenkk/Z-Stack-firmware/blob/master/router/Z-Stack_3.x.0/bin/CC1352P2_CC2652P_launchpad_router_20210128.zip), the device does will not attempt to interview and join the network.

Anyone have any ideas on how I can get my device to interview and pair with my coordinator?

thiscantbeserious, 2021-11-25:
Succesfully flashed the 20dBm variant above and got double the lqi now ... only issue I have is that no matter what I do not a single device will ever join the router even if I specifically put it on top of it ..m

cjsteward, 2021-12-05:
@thiscantbeserious I had the same issue and ended up reverting back to the stable firmware. I hope someone smarter than me can figure this out because besides the pairing issue, the increase in performance and range was really nice. It also seems like a waste having nice routers if they can only perform at 1/4 of their rated power.

ptvoinfo, 2021-12-05:
Can someone test my router firmware? It supports CC2562 and CC1352P-2, custom LED configuration, etc. I am primarily interested in signal strength.
https://ptvo.info/download/ptvo-firmware-latest.zip

thiscantbeserious, 2021-12-05:
> Can someone test my router firmware? It supports CC2562 and CC1352P-2, custom LED configuration, etc. I am primarily interested in signal strength. https://ptvo.info/download/ptvo-firmware-latest.zip

Will do. Give me a week, maybe a day :)

bkupidura, 2021-12-05:
@ptvoinfo can you build firmware which reports rssi to z2m in some (1/5/10m) intervals? Im currently using Conbee as cordinator, and to be honest im not trusting signal strenght reported by z2m.

ptvoinfo, 2021-12-06:
@bkupidura The coordinator receives RSSI with every message from a device. You can define a virtual output with a virtual pin or an output for a LED pin and enable periodic reports with the necessary interval. But I'm not sure if Conbee can receive messages from a DIY device.

ruimarinho, 2021-12-21:
@ptvoinfo what's the difference between `psm-cc2652p.hex` and `cc2652p.hex`? Not sure which one to flash.

ptvoinfo, 2021-12-22:
@ruimarinho You cannot flash these files directly.
It is necessary to configure the firmware for your board.
https://ptvo.info/cc2652r1-cc2652rb-cc2652p-cc1352p2-configurable-router-firmware-748/

irakhlin, 2021-12-22:
> @ruimarinho You cannot flash these files directly. It is necessary to configure the firmware for your board. https://ptvo.info/cc2652r1-cc2652rb-cc2652p-cc1352p2-configurable-router-firmware-748/

I have tested your router firmware on multiple cc2652p sticks, including the sonoff, zigstar and a custom router I am working on with a SHT35 and some other sensors. I am seeing a significant increase in LQI across my entire network. It would be hard to give an exact figure but previously my mesh had some distance end device sensors sitting at ~15, currently the lowest is ~70. I was however experience this issue: [10282](https://github.com/Koenkk/zigbee2mqtt/issues/10282). Earlier today I re-flashed most of the routers with the newest firmware and all are still responding and pinging as expected.

Just a few things I wanted to comment on...
In your configuration utility all settings are saved between opening and closing the application, except "serial bootloader enable SBL". I ran into an issue as I was going back and forth modifying the firmware a few times, enabled the feature, closed the app, reopened it and only changed a setting on the main page without realizing that this setting does not get saved. Ended up having to flash with jtag, not a huge deal just a minor point.
Unrelated to tx power, but it would also be amazing to be able to configure sensors to use the reporting functionality based on change example (https://www.zigbee2mqtt.io/devices/KMPCIL_RES005.html#kmpcil-kmpcil_res005). This would allow me to configure router devices with less frequent "periodic updates" but still get good sensor values. 

As a follow up question, if you do not mind sharing what version of the TI simplelink SDK are you using for these firmwares? I only ask out of interest as I was working on one for the experience.

Hedda, 2021-12-22:
> currently this is not configurable, to change you have recompile: https://github.com/Koenkk/Z-Stack-firmware/blob/master/coordinator/Z-Stack_3.x.0/COMPILE.md

FYI, submitted a separate feature request for Zigbee router firmware with remotely configurable TX power output level -> https://github.com/Koenkk/Z-Stack-firmware/issues/341

ruimarinho, 2021-12-22:
@ptvoinfo would be great to test a pre-compiled version for TubesZB based on these [schematics](https://github.com/tube0013/tube_gateways/blob/main/tube_cc_router/Schematic_cc_Router.pdf).

irakhlin, 2021-12-22:
> @ptvoinfo would be great to test a pre-compiled version for TubesZB based on these [schematics](https://github.com/tube0013/tube_gateways/blob/main/tube_cc_router/Schematic_cc_Router.pdf).

I can confirm I am using the firmware on two of those units for a few day now. With the following configuration.. just note in the second image I've blacked out Manufacturer ID, Name, Model and Icon as I have simply just set my own custom information here to make it easier and identify all units in my network running "custom" firmware I have deployed. You should be able to use anything. After simply hit "save custom converter to Z2M" which will generate a converter file that you will need to set in zigbee2mqtt before joining the device. Also do not forget to enable serial bootloader SBL or you will lock the bootloader and need to reflash via jtag. 
[image]
[image]

ruimarinho, 2021-12-23:
Thanks @ptvoinfo but I use ZHA so it would probably need its own quirk :/ plus I don't have access to a Windows machine..

digiblur, 2021-12-27:
> Can someone test my router firmware? It supports CC2562 and CC1352P-2, custom LED configuration, etc. I am primarily interested in signal strength. https://ptvo.info/download/ptvo-firmware-latest.zip

Where's the setting for the power output or is it hardcoded to 20?

irakhlin, 2021-12-27:
> > Can someone test my router firmware? It supports CC2562 and CC1352P-2, custom LED configuration, etc. I am primarily interested in signal strength. https://ptvo.info/download/ptvo-firmware-latest.zip
> 
> Where's the setting for the power output or is it hardcoded to 20?

[He had confirmed previously on his website that i](https://github.com/ptvoinfo/zigbee-configurable-firmware/issues/75)

> > Can someone test my router firmware? It supports CC2562 and CC1352P-2, custom LED configuration, etc. I am primarily interested in signal strength. https://ptvo.info/download/ptvo-firmware-latest.zip
> 
> Where's the setting for the power output or is it hardcoded to 20?

Set to 20DB atleast for cc2562 https://github.com/ptvoinfo/zigbee-configurable-firmware/issues/75

digiblur, 2021-12-27:
> > > Can someone test my router firmware? It supports CC2562 and CC1352P-2, custom LED configuration, etc. I am primarily interested in signal strength. https://ptvo.info/download/ptvo-firmware-latest.zip
> > 
> > 
> > Where's the setting for the power output or is it hardcoded to 20?
> 
> [He had confirmed previously on his website that i](https://github.com/ptvoinfo/zigbee-configurable-firmware/issues/75)
> 
> > > Can someone test my router firmware? It supports CC2562 and CC1352P-2, custom LED configuration, etc. I am primarily interested in signal strength. https://ptvo.info/download/ptvo-firmware-latest.zip
> > 
> > 
> > Where's the setting for the power output or is it hardcoded to 20?
> 
> Set to 20DB atleast for cc2562 [ptvoinfo/zigbee-configurable-firmware#75](https://github.com/ptvoinfo/zigbee-configurable-firmware/issues/75)

Interesting double jones'ing

daufinsyd, 2021-12-29:
> I have tested your router firmware on multiple cc2652p sticks, including the sonoff, zigstar and a custom router I am working on with a SHT35 and some other sensors. I am seeing a significant [...]

Hi @irakhlin is it now working stable or do you still have disconnections ?
Did you also flashed the coordinator ? or how did you solve the issue with devices routers (eg plugs) ?

cjsteward, 2021-12-30:
Just wanted to report back - I've flashed both Tube's cc2652p router and Sonoff's Dongle Plus with PTVO's router firmware while following the options in @irakhlin post above and everything has been perfectly stable for about a week or so now. I've noticed a significant increase in LQI with my 3 routers (mix of Tube's and Sonoff's), especially after being paired with my Tube's usb cc2652p coordinator that's flashed with the 20 db firmware from Koenkks dev branch. Sensors and routers pair faster and more easily, sensors located at further points in my house and garage are more responsive, Z2M network map loads up quick and accurately, and nothing has dropped off. This is the end game zigbee solution that I've been trying to achieve for literally years.  I can't thank everyone enough for all the hard work that has gone into this project and the firmware. Seriously amazing stuff.

ruimarinho, 2021-12-30:
@cjsteward which firmware are you referring to with +20dBm hardcoded? I believe the one posted on the dev branch was +9dBm only.

irakhlin, 2021-12-31:
> @cjsteward which firmware are you referring to with +20dBm hardcoded? I believe the one posted on the dev branch was +9dBm only.

@ruimarinho I think he is talking about the ptvoinfo custom firmware configuration tool. I posted a screenshot of the settings I am using for generating that firmware and I can also report my network has been incredibly stable. 

@daufinsyd I did not flash my firmware but I did flash all my 'custom' routers and a few usb powered sensors I am working on, all of which are cc2652p chips. The coordinator I am using is tube's ethernet cc2652. For the last week my network has been stable and the firmware provides a very significant increase in LQI. I now believe the issue I was having with routers going offline was due to 3 zigbee powered sensors that where running very old firmware. I am not positive about this but I have removed them from the network and no issues thus far, I will probably know in the next few days.

ruimarinho, 2021-12-31:
Thanks @irakhlin. Would be great if you could do a side-by-side test with the firmware compiled by @bkupidura above ([link](https://github.com/Koenkk/Z-Stack-firmware/files/6139567/zb_firmware_CC1352P_2.zip))!

irakhlin, 2021-12-31:
> Thanks @irakhlin. Would be great if you could do a side-by-side test with the firmware compiled by @bkupidura above ([link](https://github.com/Koenkk/Z-Stack-firmware/files/6139567/zb_firmware_CC1352P_2.zip))!
> 
> 

@ruimarinho I did not use his specific file but about a month or so ago I did make compile the same firmware with the changes he posted above. I saw an increase in LQI but the issue of not responding to pings/network map requests was very annoying. It also caused significant delay on restarts of zigbee2mqtt. It seemed like it would wait for a ping timeout from all routers running thr firmware before the web ui would start to load.

digiblur, 2021-12-31:
Fun with higher power zigbee https://youtu.be/4aLxw48ZCQo

cjsteward, 2021-12-31:
@ruimarinho I may be mistaken, and I have no real way to measure it, but I was under the impression that the coordinator firmware defaults to 9 but allows you to set it higher with "transmit_power" in the config file.

irakhlin, 2021-12-31:
> @ruimarinho I may be mistaken, and I have no real way to measure it, but I was under the impression that the dev branch coordinator firmware defaults to 9 but allows you to set it higher with "transmit_power" in the config file.
> 
> If I'm wrong then I guess I have more improvements to look forward to in the future.

@cjsteward I think the discussion here was regarding various versions of the router firmware. As it stands now that value is compiled into the firmware. There are a few different versions floating around with values from 5db to 20db (for the cc2652p) chips.

cjsteward, 2021-12-31:
@irakhlin  Yeah, the discussion here is for sure focused on the router firmware. I was just responding to ruimarinho because he was asking about the 20db coordinator firmware in the dev branch that I mentioned in my previous post.

ruimarinho, 2022-01-01:
I've switched to @ptvoinfo's firmware as well and my network has been rock solid too. I've never had an issue with @bkupidura's one too. I'm using ZHA and a Tubes coordinator and router.

tube0013, 2022-01-08:
> * Red LED always turn on

@bkupidura can you share how you enabled the Red LED?

mortenmoulder, 2022-01-23:
I'm getting a solid ~190 LQI between my two Sonoff sticks both running 20 dBm firmwares. The router is using PTVO's firmware. I tried a large antenna from a router, as well as the antenna that came with the Sonoff stick, and I couldn't see a difference between them.

The two sticks are between a thick concrete wall and the coordinator is plugged into my server without an extension cable. Still lots of options for better LQI, but 190 is very, very good.

wojciechczyz, 2022-01-29:
> @ruimarinho You cannot flash these files directly. It is necessary to configure the firmware for your board. https://ptvo.info/cc2652r1-cc2652rb-cc2652p-cc1352p2-configurable-router-firmware-748/

Hi @ptvoinfo  can you please help in correct configuration for https://github.com/egony/cc2652p_cc1352p_RF-STAR/tree/main/firmware/router RF-Star Stick? I found pins to be described here: 
https://github.com/Koenkk/Z-Stack-firmware/blob/master/coordinator/Z-Stack_3.x.0/bin/README.md

ZigStar Stick v4	CC2652P
(RFSTAR RF-BM-2652P2)
CC1352P2_CC2652P_launchpad_*.zip	DIO_15	Only for CH340C ver.	DIO_28: 2.4Ghz
DIO_29: 20dBm PA	DIO_6 (Green)
DIO_7 (Red)

Thank you,
Wojciech

KipK, 2022-02-01:
@wojciechczyz , can't we flash fril usb it using koenkk bootloader? Would it overwrite the bl ?

Hedda, 2022-02-01:
> Can someone test my router firmware? It supports CC2562 and CC1352P-2, custom LED configuration, etc. I am primarily interested in signal strength.
> https://ptvo.info/download/ptvo-firmware-latest.zip

@ptvoinfo can you maybe build +10 and +20 dBm TX router firmware for new ITead SONOFF ZBMINI-L Zigbee 3.0 Smart Switch?

https://itead.cc/product/sonoff-zbmini-l-zigbee-smart-switch-no-neutral-wire-required/

https://sonoff.tech/product/diy-smart-switch/zbmini-l/

It has been confirmed its radio board uses the exact same model of CC2652P chip as ITead Sonoff Zigbee 3.0 USB Dongle Plus:

https://community.home-assistant.io/t/sonoff-zbmini-l-zigbee-3-0-smart-switch-by-itead-works-without-neutral-wire-and-based-on-cc2652p-so-should-be-a-good-zigbee-router/384644/

PS: Not recieved mine yet but the latest information is firmware ITead is shipping is configured as "End Device" (and not "Router").

bojanpotocnik, 2022-02-21:
> latest information is firmware ITead is shipping is configured as "End Device" (and not "Router").

@Hedda, this may be because ZBMINI-L does not have guaranteed power budget, but it can be dependent on what load it is switching (check Shelly Bypass for Shelly 1L). Operating as _End Device_ means lower power consumption -> more stable operation in more use cases. Operating as _Router_ - even more with increased output power - would increase the chance of brownout resets, which would also be more impactful/noticeable when operating as a _Router_.

ZBMINI (with neutral wire) uses the same chip and does not have these limitations, but for some strange reason none of the router firmwares in this repository work with it - check last part of [this comment](https://github.com/ptvoinfo/zigbee-configurable-firmware/issues/96#issuecomment-1046124620).

nickwild-999, 2022-03-07:
@ptvoinfo @Hedda Is there a version of the firmware for the regular ZBMINI?

ptvoinfo, 2022-03-07:
@nickwild-999 
https://ptvo.info/custom-firmware-for-sonoff-zbmini-01minizb-and-ewelink-switch-zr-02-923/

nickwild-999, 2022-03-07:
Thanks @ptvoinfo so if we select the chip type to CC2652R it works? Are there any means of getting your firmware working on the Sonoff S26R2ZB plug too? Problems with this plug are on #10282

ptvoinfo, 2022-03-07:
Yes, the firmware for cc2652r works. Sorry, but I don't have info about sonoff s26R2ZB.

Hedda, 2022-03-08:
> Sorry, but I don't have info about sonoff s26R2ZB.

FYI, ITead Sonoff S26R2ZB ZigBee Smart Plug is based on radio module with CC2652P (CC2652 P1F) chip and ceramic chip antenna:

https://github.com/Koenkk/zigbee2mqtt/discussions/10909

https://github.com/Koenkk/zigbee2mqtt/issues/10282

https://itead.cc/product/sonoff-s26r2zb-zigbee-smart-plug-uk-de-fr/

https://sonoff.tech/product/smart-plug/s26r2zb/

[image]

[image]

[image]

[image]

bojanpotocnik, 2022-03-08:
@Hedda, note that the module on your pictures has exactly the same design flaw/optimization as described in https://github.com/ptvoinfo/zigbee-configurable-firmware/issues/96#issuecomment-1056704722 ([ptvo blog post](https://ptvo.info/custom-firmware-for-sonoff-zbmini-01minizb-and-ewelink-switch-zr-02-923/)), therefore enabling the power amplifier will result in working chip/FW, but no transmitted packets.

Hedda, 2022-03-09:
> @Hedda, note that the module on your pictures has exactly the same design flaw/optimization as described in [ptvoinfo/zigbee-configurable-firmware#96 (comment)](https://github.com/ptvoinfo/zigbee-configurable-firmware/issues/96#issuecomment-1056704722) ([ptvo blog post](https://ptvo.info/custom-firmware-for-sonoff-zbmini-01minizb-and-ewelink-switch-zr-02-923/)), therefore enabling the power amplifier will result in working chip/FW, but no transmitted packets.

So to resolve https://github.com/Koenkk/zigbee2mqtt/issues/10282 it then probably very likely that [selecting CC2652R chip type in the PTVO firmware configurator](https://ptvo.info/custom-firmware-for-sonoff-zbmini-01minizb-and-ewelink-switch-zr-02-923/) will result in a working firmware image on the ITead Sonoff S26R2ZB ZigBee Smart Plug similar to the new Sonoff ZBMINI (01MINIZB) as both use CC2652P (CC2652 P1F) chip model and their radio modules have same design "optimizations"?

ruimarinho, 2022-05-03:
@ptvoinfo there have been a few commits on https://github.com/ptvoinfo/zigbee-configurable-firmware but it seems like the [changelog on the blog](https://ptvo.info/zigbee-configurable-firmware-features/changelog/) hasn't been updated in a while.

Are any of the most recent commits relevant for the router firmware (in my case, originally Dec 2021)? Thanks.

ptvoinfo, 2022-05-04:
@ruimarinho All these changes are relevant to my firmware, where you can build a router too.

Hedda, 2022-05-04:
> 5dbm, **currently this is not configurable**, to change you have recompile: https://github.com/Koenkk/Z-Stack-firmware/blob/master/coordinator/Z-Stack_3.x.0/COMPILE.md

See issue https://github.com/Koenkk/Z-Stack-firmware/issues/341 for the request + discussion about having a similar feature in Koenkk's Z-Stack Zigbee Router firmware -> https://github.com/Koenkk/Z-Stack-firmware/issues/341