Koenkk/Z-Stack-firmware #277: Router firmware CC1352P-2 launchpad not connecting

kezmdor, 2021-03-28:
Hello,

I have a problem with the router firmware for CC1352P-2 launchpad version.
The whole story is I switched from using my CC1352P-2 as coordinator to slaesh's stick. That was a very easy process, just swapping the coordinator and change the serial port configuration. Then I tried to re-use my CC1352P-2 as a router and flashed the router firmware to it according to https://www.zigbee2mqtt.io/information/flashing_via_uniflash.html. In fact I already did that twice because router does not connect to my zigbee network. There is nothing to see in the zigbee2mqtt log.
I searched for some documentation on how to debug the problem, but wasn't successful.
Hope someone can help...

kezmdor, 2021-03-29:
This is a duplicate of https://github.com/Koenkk/Z-Stack-firmware/issues/220

closed 2021-03-29 by kezmdor

TSKnuhtsen, 2021-06-14:
I am having this same issue. @kezmdor how did you end up solving this issue and successfully pairing your CC1352P-2 Launchpad?