Koenkk/Z-Stack-firmware #281: CC2531 routers randomly losing connection with coordinator during 10 minutes

elbueno222, 2021-04-06:
I have two CC2531 routers connected to a CC2652R coordinator and they are usually working fine, except sometimes they both lose the connection with coordinator for 10 minutes or so, and during that time every device connected to them are unable to send/receive messages, so the whole network is down.

One of the routers (I will call it Router1) is near the coordinator, and the other one (Router2) is far away, so during normal operation Router2 sends the messages to the coordinator through Router1, and Router1 sends the messages straight to the coordinator. After sniffing the traffic I found out the issue during those 10 minutes that the routers lose connection is they interchange their roles, Router1 tries to send the messages through Router2, and Router2 tries to send the messages straight to the coordinator, so none of the messages reach the coordinator.

I saved the sniff just in case anyone wants to have a look.

ptvoinfo, 2021-04-07:
I don't have any ideas why a router disconnects.

Could you test a router based on my configurable firmware with the following settings:

CC2530 + CC25xx (select your chip)
Router
Output 1 - P30 - Uptime
Expert - Disable resettings of a device by power on/off cycle
Expert - Disable configuring the interval remotely

The Uptime value will allow detecting if a device reboots at that time.

elbueno222, 2021-04-07:
@ptvoinfo the routers don't disconnect, in the sniff I can see they both keep sending the "Report Attributes" message every minute, they just choose the wrong route and they take 10 minutes to realise they are not receiving the ACK from the coordinator and then they fix the route themselves.  I would like to know why they change the route (could it be possible to fix the route once it is stable so it doesn´t change), and if it is possible to update the firmware somehow so they take much less time to realise something is wrong and fix it (could they try to re-route after 1 or 2 minutes not receiving the ACK from the coordinator?)

ptvoinfo, 2021-04-07:
@elbueno222 I cannot change this behavior from my side because it is code inside Zigbee ZStack.

elbueno222, 2021-04-07:
Ok, in that case I guess this issue cannot be fixed. I will try to play with the configurable firmware to see if I can get any extra info.

closed 2021-05-15 by github-actions[bot]