Koenkk/Z-Stack-firmware #282: CC2530 + CC2592 | Better coordinator firmwares Z-Stack_Home_1.2 vs Z-Stack_3.0.x

Boia11, 2021-04-10:
The better coordinator firmwares for CC2530 + CC2592 (E-Byte E18-MS1PA1-PCB/IPX) is Z-Stack_Home_1.2 or Z-Stack_3.0.x .

Consideration:
- I have same device declaired as ZigBee 3.0
- I see that the router firmware is available only as Z-Stack 1.2 HA
- Firmware 3.0.x was updates 2 years ago otherwise the firmware 1.2 HA 4 months

| **Z-Stack**                       | **Device**                                           | **Zigbee** | **Direct children** | **Routes** | **Notes**                                                    |
| --------------------------------- | ---------------------------------------------------- | ---------- | ------------------- | ---------- | ------------------------------------------------------------ |
| Z-Stack_Home_1.2 (default)        | CC2530, CC2530 + CC2591, **CC2530 + CC2592**         | 1.2 HA     | 16                  | 30/0       |                                                              |
| Z-Stack_Home_1.2 (source_routing) | CC2531, CC2530, CC2530 + CC2591, **CC2530 + CC2592** | 1.2 HA     | 5                   | 40/40      |                                                              |
| Z-Stack_3.0.x                     | CC2530, CC2530 + CC2591, **CC2530 + CC2592**         | 3.0        | 10                  | 40/0       | - [Discussion #1445](https://github.com/Koenkk/zigbee2mqtt/issues/1445)                   - Max 40 Zigbee 3.0 devices |

Which is the best for compatibility and security with zigbee2mqtt?

Koenkk, 2021-04-10:
Check https://github.com/Koenkk/Z-Stack-firmware/tree/master/coordinator#im-using-a-cc2530-or-cc2531-which-firmware-should-i-use

closed 2021-04-10 by Koenkk

Boia11, 2021-04-10:
> 
> 
> Check https://github.com/Koenkk/Z-Stack-firmware/tree/master/coordinator#im-using-a-cc2530-or-cc2531-which-firmware-should-i-use

I had read but there is no useful pros and cons to decide...."Zigbee 3.0 firmwares are not recommended for the CC2530 and CC2531 (since those are not powerful enough)" but apart from this detail, nothing else is said, are there any performance problems? coverage range? compatibility?

Koenkk, 2021-04-11:
The 3.0 setup will cover less devices and might crash faster, I've never investigated this properly because TI advices NOT to use 3.0 on the CC2530/CC2531:

Boia11, 2021-04-11:
> 
> 
> The 3.0 setup will cover less devices and might crash faster, I've never investigated this properly because TI advices NOT to use 3.0 on the CC2530/CC2531:

You recommend HA 1.2 default or source_routing version?

Koenkk, 2021-04-11:
Source routing, but make sure you have enough routers because it can handle only 5 direct connections.

Boia11, 2021-04-11:
> 
> 
> Source routing, but make sure you have enough routers because it can handle only 5 direct connections.

Perfect. Thanks for informations.