Koenkk/Z-Stack-firmware #283: Invalid record at line 1

rindlerblabla, 2021-04-20:
First, I am new to all this, so probably the issue is caused by myself, but since I cannot solve it it would be nice with some help. Am trying to flash my zzh-stick from a  Synology NAS. Have installed proper CH340 drivers, pyserial and intelhex. However even if I use latest .hex file or 210102 I get the error _Invalid record at line 1_.

`python3 cc2538-bsl.py -p ../dev/ttyUSB0 -evw CC2652R_coordinator_20210120.hex
Opening port ../dev/ttyUSB0, baud 500000
Reading data from CC2652R_coordinator_20210120.hex
Your firmware looks like an Intel Hex file
ERROR: Hex file contains invalid record at line 1
`

What am I doing wrong?

rindlerblabla, 2021-04-20:
Python 3.8.6
pip 21.0.1
pyserial 3.5
intelhex 2.3.0

omerk, 2021-04-20:
This is very unlikely a firmware issue.

I don't have a Synology device but can't replicate what you're seeing on Windows or Linux. Please make sure your firmware file is not corrupt/damaged in transit and has an md5sum that matches the one on the repo:

```
(bsl-test) C:\Users\Omer\Desktop\bsl-test>md5sums.exe CC2652R_coordinator_20210120.hex

MD5sums 1.2 freeware for Win9x/ME/NT/2000/XP+
Copyright (C) 2001-2005 Jem Berkes - http://www.pc-tools.net/
Type md5sums.exe -h for help

[Path] / filename                              MD5 sum
-------------------------------------------------------------------------------
[C:\Users\Omer\Desktop\bsl-test\]
CC2652R_coordinator_20210120.hex               662bf48ded7f6a24a37c0b5cd01e1eb9

(bsl-test) C:\Users\Omer\Desktop\bsl-test>more requirements.txt
pyserial==3.5
intelhex==2.3.0

(bsl-test) C:\Users\Omer\Desktop\bsl-test>python cc2538-bsl.py -p COM19 -ewv CC2652R_coordinator_20210120.hex
Opening port COM19, baud 500000
Reading data from CC2652R_coordinator_20210120.hex
Your firmware looks like an Intel Hex file
Connecting to target...
CC1350 PG2.0 (7x7mm): 352KB Flash, 20KB SRAM, CCFG.BL_CONFIG at 0x00057FD8
Primary IEEE Address: 00:12:4B:00:21:B7:7D:5D
    Performing mass erase
Erasing all main bank flash sectors
    Erase done
Writing 360448 bytes starting at address 0x00000000
Write 104 bytes at 0x00057F980
    Write done
Verifying by comparing CRC32 calculations.
    Verified (match: 0xc8b96313)
```

rindlerblabla, 2021-04-20:
Thanks for your reply! Think it's a Synology issue. Installed python on Windows and it worked just fine.

closed 2021-04-20 by rindlerblabla