Koenkk/Z-Stack-firmware #284: LP-CC2652RB

fludstank, 2021-04-24:
After testing out [Z2M on my HUSBZB-1](https://github.com/Koenkk/zigbee-herdsman/issues/319) and really liking it, I recently bought a [LP-CC2652RB](https://www.ti.com/tool/LP-CC2652RB) thinking I was purchasing a supported LAUNCHXL-CC26X2R1 (which clearly it is not). I even flashed CC2652R_coordinator_20210120 because at that point I STILL didn't know that I had gotten something different. Once I realized the mistake I made, I flashed the LP-CC2652RB with the [CC2652RB firmware](https://github.com/Koenkk/Z-Stack-firmware/raw/master/coordinator/Z-Stack_3.x.0/bin/CC2652RB_coordinator_20210120.zip) and to my surprise...it worked. Was able to fire up the container and pair devices right away. I'm going to keep testing it over the next few days and will report back any new findings.

@Koenkk Don't know if this is something that you were aware of, but wanted to report back these initial findings - perhaps it will lead to the LP-CC2652RB becoming officially supported one day.

Koenkk, 2021-04-24:
I indeed expect that the CC2652RB firmware will work fine on it

fludstank, 2021-05-02:
Closed this out as everything is working really well. Thanks again!

closed 2021-05-02 by fludstank