Koenkk/Z-Stack-firmware #286: Thread support

kristofferR, 2021-04-26:
From what I can gather from [this](https://github.com/dresden-elektronik/deconz-rest-plugin/issues/4239) ConBee issue, I think it should be possible for this firmware to add Thread support. 

OpenThread has support for CC13x2/CC26x2 already:
https://github.com/openthread/ot-cc13x2-cc26x2

Koenkk, 2021-04-26:
The Texas Instruments SDK indeed supports Thread, but this repository focusses on Zigbee (Z-Stack) firmwares only so this is out of scope for this repo.

closed 2021-04-26 by Koenkk

kristofferR, 2021-04-26:
I don't see why both can't be integrated simultaneously, other routers using other firmwares will support both.

Koenkk, 2021-04-26:
Because my focus is on Zigbee, I'm currently not interested in Thread so I won't invest time in that.

Koenkk, 2021-05-12:
@scobby without gateway software a thread firmware is useless. You need something like Zigbee2MQTT but then for Thread/Matter/CHIP. My only reason to maintain this repo is Zigbee2MQTT.

larstobi, 2023-03-30:
I think it would be very good to be able run Zigbee and Thread simultaneously on the same USB stick, like the Home Assistant SkyConnect with the Si-Labs chip does.

According to this page https://www.ti.com/product/CC2652RB#description, the CC2652RB chip is supposed to be able to run concurrent multiprotocol operation through the Dynamic Multiprotocol Manager (DMM) software driver.

There is work going on to make gateway software for Thread, like this Matter server: https://github.com/home-assistant-libs/python-matter-server

xPakrikx, 2023-07-18:
> I think it would be very good to be able run Zigbee and Thread simultaneously on the same USB stick, like the Home Assistant SkyConnect with the Si-Labs chip does.
>
> According to this page [...]

I think in this case its matter of his personal preference/belief (pun intended). So maybe time for new project :)