Koenkk/Z-Stack-firmware #292: Router w/o routing table on CC2652RB

FlyingDomotic, 2021-05-28:
I've got a problem with a Slaesh stick (CC2652RB) used as router, loaded with CC2652RB_router_20210128 through cc2538-bsl.

Paring has been done successfully as shown:
info  2021-05-28 18:41:03: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"Routeur test","ieee_address":"0xxxxxxxxxxxxxxxxxxxxxxxxxx"},"type":"device_announce"}'
info  2021-05-28 18:41:03: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"announce","meta":{"friendly_name":"Routeur test"},"type":"device_announced"}'

However, asking for routing table fails:
error 2021-05-28 18:43:55: Failed to execute routing table for 'Routeur test'

Wireshark capture:
5596	2767.386991	0x0000	0xxxxx	ZigBee ZDP	58	Routing Table Request
5598	2767.392638	0xxxxx	0x0000	ZigBee ZDP	61	Routing Table Response, Status: Not Supported

FYI, coordinator is also a Slaesh CC2652RB ({"commit":"6484fee","coordinator":{"meta":{"maintrel":1,"majorrel":2,"minorrel":7,"product":1,"revision":20201026,"transportrev":2},"type":"zStack3x0"}), with 80+ devices, and 4 x CC2530 routers.

What did I forgot?

Koenkk, 2021-05-28:
Looks that the current FW does not support the routing table command, should be fixed in the FW (not a fault on your side)

FlyingDomotic, 2021-05-31:
Thanks for the answer.

Is this specific to CC2652RB, or generic (meaning that routers V3.0 don't work at all with V3.0 coordinator due to lack of routing support in routers)?

Any idea of how to fix this problem? People to signal the error?

Koenkk, 2021-05-31:
> meaning that routers V3.0 don't work at all with V3.0 coordinator due to lack of routing support in routers)

That's not what it means. It will function fine as a router, it just doesn't support the Zigbee command to retrieve all routing table.

This needs to be added support for in the router firmware, in case you are interested to do this you can read https://github.com/Koenkk/Z-Stack-firmware/blob/master/router/Z-Stack_3.x.0/COMPILE.md on how to setup a dev environment.

FlyingDomotic, 2021-06-01:
I downloaded the last SDK version (5.10.48), and adapted patch to the LP_CC2652RB project (zr_genericapp_CC2652RB_LAUNCHXL_tirtos_css doesn't exists anymore in V5).

I built the project and downloaded it successfully into stick. This works properly (but still no routing request support).

Now, if I correctly understood, I have to intercept cluster 0x32 request and send back cluster(s) 0x8032.

Is this correct? Any example of this implementation on other stick?

Koenkk, 2021-06-02:
I don't know of any other example, I recommend to ask for support on https://e2e.ti.com/

FlyingDomotic, 2021-06-03:
Thanks for the advise.

I read TI sources (from simplelink_cc13x2_26x2_sdk_5_10_00_48) and found that code already exists but is not activated by default.

To activate it, you may define ZDO_API_ADVANCED in preinclude.h (from Stack/config) and recompile the code.

You then will have a fully functional 2652RB stick router.

I'll let you decide including it (or not) in your patch documentation.

Have fun!

closed 2021-06-03 by FlyingDomotic

Koenkk, 2021-06-04:
Thanks, will add this on my next router update!

wernerhp, 2021-08-15:
I am also experiencing an issue.  I flashed a Slaesh CC2532RB stick with the router firmware and paired it to a Sonoff ZB Bridge (flashed with Tasmota using ZHA).  It shows up in ZHA, but no devices are connecting to the router (0 Children).  Are you still working on adding this to a router update?

FlyingDomotic, 2021-09-08:
A small update on adding routing functions: with 3.x.0 version:

I put the following in /Stack/Config/preinclude.h

>#define ZCL_REPORT_DESTINATION_DEVICE
#define BDB_REPORTING
#define CUI_DISABLE
#define NWK_MAX_DEVICE_LIST     150
#define MAX_NEIGHBOR_ENTRIES    150
#define MAX_RTG_ENTRIES         150
// Increase NV pages to 3 to allow for bigger device tables
#undef NVOCMP_NVPAGES
#define NVOCMP_NVPAGES 3
#define ZDO_API_ADVANCED
#define ZDO_NWKADDR_REQUEST
#define ZDO_IEEEADDR_REQUEST
#define ZDO_MATCH_REQUEST
#define ZDO_NODEDESC_REQUEST
#define ZDO_POWERDESC_REQUEST
#define ZDO_SIMPLEDESC_REQUEST
#define ZDO_ACTIVEEP_REQUEST
#define ZDO_COMPLEXDESC_REQUEST
#define ZDO_USERDESC_REQUEST
#define ZDO_USERDESCSET_REQUEST
#define ZDO_ENDDEVICEBIND_REQUEST
#define ZDO_BIND_UNBIND_REQUEST
#define ZDO_SERVERDISC_REQUEST
#define ZDO_NETWORKSTART_REQUEST
#define ZDO_MANUAL_JOIN
#define ZDO_COMPLEXDESC_RESPONSE
#define ZDO_USERDESC_RESPONSE
#define ZDO_USERDESCSET_RESPONSE
#define ZDO_SERVERDISC_RESPONSE
#define ZDO_ENDDEVICE_ANNCE
#define ZDO_MGMT_NWKDISC_REQUEST
#define ZDO_MGMT_LQI_REQUEST
#define ZDO_MGMT_RTG_REQUEST
#define ZDO_MGMT_BIND_REQUEST
#define ZDO_MGMT_LEAVE_REQUEST
#define ZDO_MGMT_JOINDIRECT_REQUEST
#define ZDO_MGMT_PERMIT_JOIN_REQUEST
#define ZDO_MGMT_NWKUPDATE_REQUEST
#define ZDO_MGMT_NWKDISC_RESPONSE
#define ZDO_MGMT_LQI_RESPONSE
#define ZDO_MGMT_RTG_RESPONSE
#define ZDO_MGMT_BIND_RESPONSE
#define ZDO_MGMT_LEAVE_RESPONSE
#define ZDO_MGMT_JOINDIRECT_RESPONSE
#define ZDO_MGMT_PERMIT_JOIN_RESPONSE
#define ZDO_MGMT_NWKUPDATE_NOTIFY

I also cc13x2_cc26x2.cmd changed with (as I extended NVRAM):

>#define FLASH_SIZE              0x50000
#define FLASH_NV_BASE           0x50000
#define FLASH_NV_SIZE           0x6000

This is working perfectly with 100+ devices on CC2652RB coordinator (with 3.0.x) and 5 routers (with 3.x.0)

Koenkk, 2021-09-10:
@FlyingDomotic did you already solve the xiaomi devices problem? Disabling child ageing may fix this.

FlyingDomotic, 2021-09-11:
@Koenkk I have to check over a longer period, but since I resized tables, I haven't any Xiaomi (and other brands) problems. I'll keep you posted if anything wrong occurs again.

wernerhp, 2021-09-12:
If you can send me a binary I can test also

FlyingDomotic, 2021-09-12:
@wernerhp : files sent!

ra1der75, 2021-10-30:
If you can send me a binary I can test also

FlyingDomotic, 2021-10-30:
@ra1der75 : Drop me a mail address in PM, I'll do it.

8none1, 2021-10-31:
I think I've got this same issue and would be interested in testing this firmware as well, on a zzh! stick.  I'm not sure how to contact you @FlyingDomotic but would you be able to send me a hex file as well?  My email is will at whizzy.org.
Thanks!

FlyingDomotic, 2021-11-01:
@8none1: It seems that zzh! stick is a CC2652R, but my hex files are build for CC2652RB. Don't know if this can works, but I guess that it won't. Perhaps someone more expert than me can tell if this as a chance to do it.

8none1, 2021-11-01:
Thanks @FlyingDomotic, I'll see if I can work out how to build the f/w

ra1der75, 2021-11-01:
> @ra1der75: Напишите мне почтовый адрес в PM, я сделаю это.

> @ra1der75 : Drop me a mail address in PM, I'll do it.

grinko.dmitriy@gmail.com

Koenkk, 2022-01-25:
I've created a fw with this enabled, see https://github.com/Koenkk/Z-Stack-firmware/tree/develop/router/Z-Stack_3.x.0/bin

Shred99, 2022-02-27:
I may put a jinx on my system by commenting, but...

I have a Xiaomi MCCGQ11M Door/Window sensor that has a very marginal signal - it is about 20m from the house and monitors a gate.  The sensor used to be fairly stable (ran for months at a time without needing re-pairing), but due to shrubs and trees growing (I think), it has needed manual re-pairing on a weekly basis lately.

I upgraded both my routers to the patched firmware a bit over two weeks ago.  The sensor has not dropped off the network since.  It's possible that this is related to the higher Tx power on the ZZH! routers, but  I log the 50 minute heartbeat messages to a SIEM and can see that the signal is still very flakey - there are multiple missed heartbeats every day.  Despite this, the sensor has become much more "sticky" and has not dropped off the network with the new firmware.