Koenkk/Z-Stack-firmware #293: cc2530+2591 - child limits

homeoneautomacao, 2021-05-31:
Hi,

     I am using the latest version 20201128 zStack12 source_routing and as you can see he has 6 direct children despite the limit being 5. I even put 8 routers and everyone was connected to the coordinator. In some moments, endpoints were also linked to the coordinator. I'm having problems with some endpoint sensors sometimes losing the connection. Even if I pair with a certain router after a few hours, they get lost. I'm starting to believe that maybe there's a problem with this child limit and it's interfering with the negotiation between the router and the endpoint. For example, will the endpoint not connect to the router because I chose it but after a few hours the 2591 signal is much stronger and ends up bringing it to the coordinator, but the limit of 5 has already been exceeded and that's why it gets lost?

Thanks

[image]

homeoneautomacao, 2021-07-02:
Hi, i´m still waiting an answer to understand whats going on. Why source_routing accept more than five children and if this maybe is the cause of some devices trying to connect to the coordinator and became lost.

cyijun, 2021-08-09:
My CC2530+CC2591 coordinator is unstable too though I use the default firmware. I have tried many ways to find out the source of the problem but I get nothing. Maybe these much endpionts should use a CC2652R/P as coordinator. I have designed one and it is being produced. I will open the design if it works well.

closed 2021-09-16 by github-actions[bot]