Koenkk/Z-Stack-firmware #294: CC2531 does not pair anymore

drschnalli, 2021-06-08:
2 Days ago i had a little crash of my zigbee2mqtt... i dont know if the stick messed up or something... i flashed the newest default firmware (which is also linked on zigbee2mqtt.org) and zigbee2mqtt is starting again but for now i cannot pair my devices anymore. Some devices are trying to but fail to interview. I already deletet all devices in configuration.yaml and also deleted database.db and state.json... does not help... i even got my second CC2531 stick today.. flashed the same firmware... but does not work...  when i try the 3.0.x firmware then zigbee2mqtt does not start giving me an error like:

Error: Network commisioning failed panID collision detected expected=6754, actual 6755....

drschnalli, 2021-06-08:
Edit: I found the "error"... i had two Blitzwolf Current Measuring Devices which were still plugged into the wall... they might messed up all things... it works perfectly again now! Sorry for disturbing!

closed 2021-06-08 by Koenkk

mpkogli, 2021-06-22:
I'm getting the same error (panId collision detected (expected=6754, actual=6755). Haven't flashed any new firmware to my CC2531 and it was working fine until a few weeks ago. I'm not using any Zigbee repeaters. Any recommendations would be appreciated!

Koenkk, 2021-06-22:
@mpkogli add to your configuration.yaml:

```yaml
advanced:
  pan_id: 6755
```

mpkogli, 2021-06-22:
Thanks @Koenkk! That did the trick.

jarny0815, 2022-10-01:
Same problem here. Using two different Homeassistant hardwares. One instance running on a Raspi3 using an zigbee usbstick and one on an old Odroid C2 (only for test purposes). Now I try to get an old zigbee usbstick running on my Odroid but I get those error message (because now I try to start a second zigbee network):
>Zigbee2MQTT:info  2022-10-01 20:10:44: Starting Zigbee2MQTT version 1.28.0 (commit #unknown)
>Zigbee2MQTT:info  2022-10-01 20:10:44: Starting zigbee-herdsman (0.14.62)
>Zigbee2MQTT:error 2022-10-01 20:10:52: Error while starting zigbee-herdsman
>Zigbee2MQTT:error 2022-10-01 20:10:52: Failed to start zigbee
>Zigbee2MQTT:error 2022-10-01 20:10:52: Check https://www.zigbee2mqtt.io/guide/installation/20_zigbee2mqtt-fails-to-start.html for possible solutions
>Zigbee2MQTT:error 2022-10-01 20:10:52: Exiting...
>Zigbee2MQTT:error 2022-10-01 20:10:52: Error: network commissioning failed - panId collision detected (expected=6754, actual=6755)
    at ZnpAdapterManager.beginCommissioning (/app/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/manager.ts:370:19)

I add the recommended line to the zigbee2mqtt addon configuration but it doesnt make any different at startup.
My configuration:
>`data_path: /config/zigbee2mqtt
>socat:
>  enabled: false
>  master: pty,raw,echo=0,link=/tmp/ttyZ2M,mode=777
>  slave: tcp-listen:8485,keepalive,nodelay,reuseaddr,keepidle=1,keepintvl=1,keepcnt=5
>  options: "-d -d"
>  log: false
>mqtt: {}
>serial:
>  port: /dev/ttyACM0
>advanced:
>  pan_id: 6755

maybe the advanced parameter **pan_id** is ignored? 
I am using:
Homeassistant 2022.9.7, Supervisor 2022.09.1, Operating System 9.0
Zigbee2MQTT 1.28.0-1 (fresh installation)
Thx in advance
Jarny

alexxoid, 2023-01-14:
> maybe the advanced parameter **pan_id** is ignored? I am using: Homeassistant 2022.9.7, Supervisor 2022.09.1, 

It's not in zigbee2mqtt addon configuration but in **File Editor** - **Folders** - zigbee2mqtt - configuration.yaml

Just worked for me after hours of trouble with base add-on config.

jbrepogmailcom, 2023-11-12:
> @mpkogli add to your configuration.yaml:
> 
> ```yaml
> advanced:
>   pan_id: 6755
> ```

Hello. This helped, but it is a quite bad error. It caused that all my zigbee devices lost access to network and were unable to autocorrect connection. I had to re-pair all devices manually. How to avoid such problem in the future?