Koenkk/Z-Stack-firmware #304: Z-Stack 3.x.0 router firmwares do not send / trigger last seen messages in z2m

marc-gist, 2021-08-12:
Not sure if this is a firmware issue or a z2m issue.  The C253x routers send a notification about every minute or two.  which is great to ensure that they are still alive.
e.g.

```
MQTT publish: topic 'zigbee/device/CCRouter01', payload '{"device":{"dateCode":"2018-09-26","friendlyName":"CCRouter01","hardwareVersion":1,"ieeeAddr":"0x00124b0018e2ed78","manufacturerID":4447,"manufacturerName":"LUMI","model":"CC2530.ROUTER","networkAddress":20447,"powerSource":"Mains (single phase)","type":"Router","zclVersion":1},"last_seen":"2021-08-12T22:45:52.424Z","led":false,"linkquality":39}'
```

Can the 3.x.0 router firmware do the same?  Thanks!

marc-gist, 2021-08-15:
[image]
Setting the min report period seemd to work.

closed 2021-08-15 by marc-gist

mamrai1, 2021-09-04:
I tried this but still can't get a report at min interval.