Koenkk/Z-Stack-firmware #307: xiaomi devices loosing connection

rvalle, 2021-08-25:
Hi !

I am aware that Xiaomi devices have a limited zigbee stack implementation and that sometimes they disconnect form network.

I built my network from the scratch around zigbee2mqtt, so it has a CC2652RB coordinator and a CC2652RB router. The recommended blue stick from slae.sh.

At first I deployed only the coordinator, but after realizing that coverage was poor I also added the router.

Since I added the router, the xiaomi devices (thermometers and PIRs)  disconnect form network very quickly despite having much better coverage now. (>35)

I am almost convinced that there is something in the router firmware that causes this issue (or perhaps a missing fix for the xiaomi poor implementation), something that does not affect the coordinator.

Xiaomi devices directly connected to the coordinator work just fine. 

Anybody experiencing this? Are xiaomi devices working via this router firmware? (CC2652RB_router_20210128.hex)

FlyingDomotic, 2021-08-29:
I have exactly the same issue as @rvalle with 5 slae.sh CC2652RB sticks, one being coordinator,  and 4 used as routers in a 90+ device configuration. I loose my PIR & thermometers after 2 to 5 days, in waves (few detectors are lost approximately at the same time). Doors contacts and water leak detectors seems less concerned, while they're too.
I was thinking about table sizes (some routers have 30+ devices attached), as internal device table is set to 30 and routing table to 50.
I'm trying to rebuild firmware with higher values, but procedure is not working (I'll open an issue for that).
Did you check the number of devices attached to your routers?

rvalle, 2021-08-30:
@FlyingDomotic I only have 20 devices, and to the router with this firmware perhaps only 7 get attached 2.

I have read many reports that Xiaomi does not implement full zigbee and that as a consequence network connection is lost. I was thinking that would be the main issue here.

are your "offending" devices Xiaomi? or are there other brands loosing connectivity too?

BTW where are the sources of this stack?

FlyingDomotic, 2021-08-30:
I've got mainly Aqara (PIR, door contact, vibration, water leak, temperature, TVOC), but also Heiman, bitron, frient, ptvo).

In my case, I suspect some tables to be silently full, which doesn't mean that Xiaomi can't be bugged. The hard thing is to distinguish btw these 2 issues.

FlyingDomotic, 2021-09-08:
I solved my problems recompiling both coordinators and all routers, with larger (and more coherent) tables. Since it did it, I didn't loose any sensor. I had to re-pair some sensors, but not all of them.

Concerning sources of this stack, they're coming from Texas Instruments, with few modifications Let's have a look to COMPILE.md in each kind of folder to see how to make them.

rvalle, 2021-09-09:
@FlyingDomotic that is interesting. Someone needs to be aware of what the Aqara zigbee incompatibility is and it must be possible to patch the code.
It might be something simple to implement, like not checking in at a specified interval, et.c.

FlyingDomotic, 2021-09-09:
@rvalle That's not exactly what I wanted to say: IMHO, Aqara are properly supported by Z2M. In my case, I was using more devices (30+) on a router than allowed, generating some troubles and loosing devices after few time. I just rebuild both coordinator and (5) routers with larger (150) tables to fix the issue. In addition to that, I made another change to support routing report on CC2652RB sticks, which where not having this option needed by Domoticz Z2M plugin (Z2M own UI doesn't need this).
Details can be found at issue #292.

rvalle, 2021-09-09:
I only have 20 devices, and my issue is on the router, not the coordinator.

Do you think that perhaps #292 is causing my issue? 

#292 seems to be closed in May, and I flashed my devices recently. Is the #292 fix in the current FW?

FlyingDomotic, 2021-09-09:
Depending on the router you have, 20 may be the limit (for CC2531 for example). However, coordinator will also help to support some devices, so globally you may also be on the safe side. One way to tell you is to get a copy of the routing table, provided you have some tools to send/receive MQTT messages. Having a copy of your Zigbee schema may also help, but it'll contain sensitive information like mac addresses. At least, you may draw it and manually count the number of devices attached to the coordinator and the router. Giving us these counts and models of coordinator & router, we may help to give you more details. Don't forget to re-pair all missing/lost devices to have the full scope.

As far I understood, #292 is not yet integrated. Anyway, it'll fix only the routing table dump issue, nothing more.

rvalle, 2021-09-13:
@FlyingDomotic I have 2x CC2652RB as both router and coordinator.

Is anybody successfully running Xiaomi/Aquara devices (PIR/Door) via a router with the Z-Stack-Firmware?

FlyingDomotic, 2021-09-13:
@rvalle: Yes, I'am! Even with current version of firmware available here. The only thing is that I personally have more devices than supported by these versions, and had to increase table sizes to fit 100 devices. Until you keep under limits, this is working well.
I have Aqara PIR/door contact/temperature/water leakage and VOC (plus other non Xiaomi devices).

rvalle, 2021-09-13:
@FlyingDomotic thanks for the information.

And your xiaomis are connecting via router? Cause mine attached coordinator work OK.

Lets look at this issue from a configuration management POV, see if we can debug what is going on.

I am running 2x CC2652RB with CC2652RB_router_20210128.hex, what HW are you running?

The devices are managed by zigbee2mqtt. 

I have consistent failure (log off network) of: PIR, DOOR, and Thermometer, only Aqara and only on router. 

The router continues to work with other devices, mainly IKEA, and one Xiaomi plug. 

Is there any way I can debug the router?

FlyingDomotic, 2021-09-13:
Most of devices are connected to routers (coordinator is something isolated).

> I am running 2x CC2652RB with CC2652RB_router_20210128.hex, what HW are you running?

For routers, I'm using a modified version of 20210128.hex in 2 points:

1. ZDO_API_ADVANCED is defined, to allow routing to be sent to zigbee2mqtt Domoticz extension (this mainly authorize routing table request/response).
2. Tables have been resized for 150 devices.

In my case, issues disappeared when I resized tables (routers where initially sized for 30 devices, some of them had more than this).

You may check count of devices attached to each device (and total count of devices for coordinator), and check them against table size of each of your sticks.

> The router continues to work with other devices, mainly IKEA, and one Xiaomi plug.

It seems to me that all of these devices are also routers, aren't they?

> Is there any way I can debug the router?

You may dump routers routing tables (but you have to use a modified version including ZDO_API_ADVANCED as default firmware seems not supporting this). One way to do this is to send an MQTT request to Z2M like 

`mqtt.publish("zigbee2mqtt/bridge/request/networkmap", '{"type":"raw", "routes":true}')`

You may also use a CC2530/31 sniffer and read exchanged packets into WireShark with zboss_sniffer

rvalle, 2021-10-02:
I am going to make a test before changing the tables... I have replaced the CC2652RB router with an Aqara plug, and all xiaomi devices joined via this plug. Lets see if the connection is lost now, or not. Perhaps that fives us more information about the cause.

FlyingDomotic, 2021-10-04:
Some news about Aqara sensors with CC2652RB. Despite increasing tables, I still have very few devices that are lost after some times (but this time, it's for sure not related to tables anymore).
I'll trace more things to try to isolate the issue.

rvalle, 2021-10-05:
So far, connections are not lost. 

I will wait for a week or so, if they keep stable that would mean that there is something in xiaomi's routers to adapt to their incompatible implementation of zigbee, woudn't it?

[image]

FlyingDomotic, 2021-11-06:
Some news on my side:

After few weeks with large routing tables, I still have some issues with Xiaomi.

I did instrument my installation to get permanent Herdsman logs (as well as Z2M debug logs), plus some Python code to get last seen date and quickly identify newly lost devices.

First analysis didn't show specific messages on Herdsman side (I still should spend more time to more deeply analyze logs). 

I'm thinking capturing ZigBee traffic with a CC2531 stick (knowing it won't cover the full installation, due to large surface).

One remark on re-joining lost devices: this can be done only shortly pushing the pair button on Xiaomi, (I don't need to keep it pushed for few seconds until it blinks 3 times). Then i got "Device xxx announced itself" in Z2M log.

I seems to me I read somewhere that this could be due to a timeout somewhere. Does someone have more details on this?

Stay tuned ;-)

rvalle, 2021-11-10:
On my side, the Aqara power plug as router is rock solid, all xiaomi devices behave correctly now, they did not drop off the network even once.

I actually like the device, provides really good coverage, just like the CC2652RB with a decent antenna, it is visually discreet, and provides turns a power plug into a smart one.

So I think I am going to leave this as my way to go, and perhaps use the extra CC2652RB router as development/sniffer.

Obviously, the aqara plug it must implement some kind of workaround to deal with xiami's non-fully-compatible zigbee implementation, (extended timeouts ?).

Still an interesting fact is that xiami devices that attach directly to a CC2652RB coordinator do not drop of the network, so, the issue should be possible to workout somehow.

FlyingDomotic, 2021-12-17:
Some news about this issue:

My network (99 nodes now) is pretty stable, showing errors only around 10 days.

I tried to debug this at Z2M logs level first, then with herdsman debug log level, without success.

So I put an RPi3 with a sniffer (CC2530) key, and started capturing every frame. I wrote a small python script to extract everything speaking about a node from a Wireshark pcap, making both summary and detailed files.

Since I did this (around 2 weeks ago), I had only 2 occurrences of disconnection, with 2 different devices, connected to 2 different routers.

Each time, scenario is the same: end device are properly feeding the coordinator, through a/some router(s) for a while.

Then, as end device sends a "Report attributes", something goes wrong in routing, ending with lot of "Network Status, 0x3965: Many-to-One Route Failure", making end device to change it's neighbor's router.

When end devices sends next "Report attributes", it receives a "Leave request", and leaves the network as required.

In my case, this seems having nothing to do with end device time-out, but more to topology change (or more probably a router not responding to the end device when requesting the route).

I'll go deeper (when having free time) and keep you posted here.

Meantime, have fun!

zen2, 2021-12-17:
> In my case, this seems having nothing to do with end device time-out, but more to topology change (or more probably a router not responding to the end device when requesting the route).

You confirm what I have suspected for long because this problem have always happened few hours to days after have added new router in the mesh. So my strategy to solve this was extreme: reflash the firmware (to get rid of routes memory on coordinator) and switch off main house power to get rid of routes memory on router devices in order to force a new topology mesh. It's extreme but that have been effective.

zen2, 2022-01-18:
Since I have added several plugs that act as router my network stabilize and everything was working like a charm.
Then yesterday I reflashed my coordinator with latest firmware and upgraded zigbee2mqtt.
All devices come up but not 5 aqara end devices.

Notes:
- they are all situated in the same area of the house on 2nd floor (North).
- There is a really good router that usually handle many end devices on its own right under at 1st floor (18 actually)
- Coordinator is central at first floor and many plug devices are localised on south basement and south 1st floor
- There is HUE light in each room

Globally the mesh is really strong with 36 routers and 58 End devices.

But I wonder about how end devices reconnect to network since they seems to reconnect only to one parent and when you reinitialize the mesh like after flashing fully the firmware it's like they can't connect to parent and so need to rejoin the network.
I wonder if it's possible at this step that the router need the join authorization to accept a new end device.

Is it possible that end devices fight each other to find an eligible parent in some situations ?
More routers you get more the mesh use connections between routers and some routers handle easily a lot of end devices while other one only few, and many are only part of the routers mesh.

So how many connections can have a router ? Is it router dependant ?
Is it possible to get starvation of free router connections for end devices ?
Can you easily do join end devices to specific routers or only coordinator ?

I've already make join end devices to specific coordinator but only specific ones.
From memory, to re-include aqara devices, I can do it only close to coordinator.

A lot of questions that i think are router firmware dependant.
This is literacy that I have found on this topic:
[https://www.sciencedirect.com/topics/computer-science/zigbee-end-device](https://www.sciencedirect.com/topics/computer-science/zigbee-end-device)

nickw444, 2022-02-21:
Definitely don't think this is stale - Am constantly having to go around re-pairing Xiaomi's which keep dropping off the network. 

[image]

Koenkk, 2022-02-21:
@nickw444 probably not something that can be solved from z2m, make sure to check https://www.zigbee2mqtt.io/devices/WXKG01LM.html#troubleshooting-device-stops-sending-messages-disconnects-from-network

the-cyberspace, 2022-03-24:
> Definitely don't think this is stale - Am constantly having to go around re-pairing Xiaomi's which keep dropping off the network.
> 
> [image]

I`ve got the same problem. Just repaired 3 xiaomi sensors.

nickw444, 2022-04-04:
I think I've narrowed this down to being an issue with using a CC2652R stick running the router firmware. I've removed the router from my network, and added a Tradfri smart plug in it's place, and the disconnection problem seems to have disappeared.

closed 2022-05-12 by github-actions[bot]