Koenkk/Z-Stack-firmware #308: Few errors in COMPILE.md of  Z-Stack 3.x.0 router firmware

FlyingDomotic, 2021-08-29:
I found few errors in COMPILE.md explaining how to rebuild firmware for Z-Stack 3.x.0 router.

Firstly, there's no znp.syscfg. The only .syscfg files are named zr_genericapp (one in each folder). Nothing containg the string "znp.syscfg" in any file in these folders too.

Secondly, using zr_genericapp.syscfg, I looked for "Minimal Poll Period", but found nothing. Guessing that the double change could be used to write newer versions of syscfg files, I changed twice "Network Maximum Device List" from 20 to 50 and back to 20 in the 3 folders.

Remaining of procedure worked well.

Hope this helps,

Note: to avoid mistakes, I removed everything related to this issue from my computer (compiler, SDK, workspace), and redone all from scratch when writing this message.

FlyingDomotic, 2021-08-29:
In addition, I found the following errors in Z-Stack-firmware/coordinator/Z-Stack_3.x.0/COMPILE.md

Step 2 required loading from simplelink_cc13x2_26x2_sdk_4_40_00_44/examples/rtos, which generates:

> git apply firmware.patch --ignore-space-change
firmware.patch:272: trailing whitespace.
firmware.patch:274: trailing whitespace.
  // ZDSecMgrPermitJoining is never called. Joining via a router would always fail now since
firmware.patch:567: trailing whitespace.
#define SET_CCFG_MODE_CONF_XOSC_CAP_MOD                 0x0
firmware.patch:1156: trailing whitespace.
    *
firmware.patch:1158: trailing whitespace.
    *
warning: znp_CC1352P_2_LAUNCHXL_tirtos_ccs/cc13x2_cc26x2.cmd has type 100644, expected 100755
warning: znp_CC26X2R1_LAUNCHXL_tirtos_ccs/cc13x2_cc26x2.cmd has type 100644, expected 100755
error: znp_LP_CC2652RB_tirtos_ccs/Application/mt/mt.h: No such file or directory
error: znp_LP_CC2652RB_tirtos_ccs/Application/mt/mt_util.c: No such file or directory
error: znp_LP_CC2652RB_tirtos_ccs/Application/mt/mt_version.c: No such file or directory
error: znp_LP_CC2652RB_tirtos_ccs/Application/mt/mt_zdo.c: No such file or directory
error: znp_LP_CC2652RB_tirtos_ccs/Stack/Config/znp_cnf.opts: No such file or directory
error: znp_LP_CC2652RB_tirtos_ccs/Stack/af/af.c: No such file or directory
error: znp_LP_CC2652RB_tirtos_ccs/Stack/nwk/nwk_globals.c: No such file or directory
error: znp_LP_CC2652RB_tirtos_ccs/Stack/sys/zglobals.c: No such file or directory
error: znp_LP_CC2652RB_tirtos_ccs/Stack/zdo/zd_app.c: No such file or directory
error: znp_LP_CC2652RB_tirtos_ccs/cc13x2_cc26x2.cmd: No such file or directory
error: znp_LP_CC2652RB_tirtos_ccs/znp.syscfg: No such file or directory

As SDK_5.10.00.48 was just downloaded, I tried with this version, giving a more acceptable log:

> git apply firmware.patch --ignore-space-change
firmware.patch:272: trailing whitespace.
firmware.patch:274: trailing whitespace.
  // ZDSecMgrPermitJoining is never called. Joining via a router would always fail now since
firmware.patch:567: trailing whitespace.
#define SET_CCFG_MODE_CONF_XOSC_CAP_MOD                 0x0
firmware.patch:1156: trailing whitespace.
    *
firmware.patch:1158: trailing whitespace.
    *
warning: znp_CC1352P_2_LAUNCHXL_tirtos_ccs/cc13x2_cc26x2.cmd has type 100644, expected 100755
warning: znp_CC26X2R1_LAUNCHXL_tirtos_ccs/cc13x2_cc26x2.cmd has type 100644, expected 100755
warning: znp_LP_CC2652RB_tirtos_ccs/cc13x2_cc26x2.cmd has type 100644, expected 100755
warning: squelched 6 whitespace errors
warning: 11 lines add whitespace errors.

No problem in firmware generation then (but nothing tested yet).

Koenkk, 2021-08-29:
I will recheck and update the instructions when the Q4 version of zstack is released.

zen2, 2021-08-30:
> warning: znp_CC1352P_2_LAUNCHXL_tirtos_ccs/cc13x2_cc26x2.cmd has type 100644, expected 100755
> warning: znp_CC26X2R1_LAUNCHXL_tirtos_ccs/cc13x2_cc26x2.cmd has type 100644, expected 100755
> warning: znp_LP_CC2652RB_tirtos_ccs/cc13x2_cc26x2.cmd has type 100644, expected 100755

I'm almost sure that you can correct this warning by changing permissions on related files from 644 (rw-r--r--) to 755 (rwxr-xr-x) if you are on linux.

FlyingDomotic, 2021-08-30:
Obviously, I'm on Windows :-(

However, this is only a warning, not a big deal...

closed 2021-10-08 by github-actions[bot]