Koenkk/Z-Stack-firmware #310: Add index.json for automatic firmware downloads for board types/roles with llama-bsl

omerk, 2021-09-10:
I started working on a fork of cc2538-bsl.py to add features that will be useful to folks looking at this repo, located at: https://github.com/electrolama/llama-bsl 

First new feature added is automatic download of firmware for specific board/roles. See below:
[image]

The "backend" for this is a simple JSON file that lists available roles->stack versions->releases. This file also contains board definitions and the firmware configuration mappings. Thinking perhaps it could also be used for adding a "BSL lock-out" check ([see discussion](https://github.com/electrolama/llama-bsl/issues/3)) as it is directly linked to firmware configurations anyway. 

The version of `index.json` that is used in the current preview release ([see PyPI](https://pypi.org/project/llama-bsl/)) is located at: https://github.com/electrolama/Z-Stack-firmware/blob/master/index.json The idea, of course, is to move it here once it is tested and ready for general public and support all board types and firmware configurations. Probably makes sense to automate the generation of this file for firmware releases and when new board->firmware config mappings are added.

@Koenkk thoughts?. Are you happy adding `index.json` to the repo?

Probably worth mentioning that help is appreciated, especially on packaging for Windows ([issue](https://github.com/electrolama/llama-bsl/issues/1)) and prettifying PyPi releases via Github Actions ([issue](https://github.com/electrolama/llama-bsl/issues/2)).

Koenkk, 2021-09-11:
Looks like a good idea to me!

What is the purpose of `stack_default`?

omerk, 2021-09-12:
`stack_default` is there in case no `--stack` parameter is specified when running the script, so we know what default stack we should grab ([reference](https://github.com/electrolama/llama-bsl/blob/09b896a5d01f5041cc088d39c1c1b0366f70517c/llama-bsl.py#L1047-L1049))

I don't really see the `--stack` parameter being used a lot tbh, added it in as an option for the sake of completeness.

Koenkk, 2021-09-12:
But no other `stack_default` than `Z-Stack_3.x.0` are allowed?

omerk, 2021-09-12:
You can bypass the default by passing the --stack parameter when you call the script. For example: `./llama-bsl.py -p X                                                  -ewv --download --board zzh --role coordinator --stack Z-Stack_3.0.x` will download the `Z-Stack_3.0.x` coordinator instead of the `stack_default` specified in JSON (Z-Stack_3.x.0).

For router, that is a bit moot as there is only Z-Stack_3.x.0 release but kept it there in case things change in the future.

Koenkk, 2021-09-12:
I see, thanks!

Hedda, 2021-11-19:
Hope this will not go stale?

omerk, 2021-11-19:
Hi @Hedda, this is already committed and working so at this point this issue is more a reference than discussion. See https://github.com/electrolama/llama-bsl/commit/c5dd2a8496086088dba6db8513cc358e146ca274 and https://github.com/Koenkk/Z-Stack-firmware/pull/311

You can grab llama-bsl and run the commands above to download and flash firmware based on board/role.

As far as I'm concerned it is okay to close this issue. Further additions/discussions can happen over a new issue/PR.

closed 2021-11-19 by omerk

pipiche38, 2022-05-03:
I beleive, the index.json https://raw.githubusercontent.com/Koenkk/Z-Stack-firmware/master/index.json is not in sync with the repository.