Koenkk/Z-Stack-firmware #321: Why not using GitHub Releases?

end84, 2021-10-31:
Hello @Koenkk,

i recently bought a "SONOFF Zigbee 3.0 USB Dongle Plus" and flashed the "20210708" firmware. 

To possibly get back to the pre-flashed firmware, i wanted to download the "20210120" firmware and then i see that you are not using the GitHub Releases Feature on this project. 

Why you not using it? Yes, older firmware versions can downloaded by search in the Git History but using the GitHub Releases Feature is easier to use.

It would be good if you could use this feature and make the older versions available later on.

Koenkk, 2021-10-31:
Good idea, will do this for new releases!

closed 2021-10-31 by Koenkk

end84, 2021-10-31:
Why only(?) for new releases?

Koenkk, 2021-10-31:
For older releases, you can download them like the method you described.