Koenkk/Z-Stack-firmware #323: Need Z-Stack 3.x.0 coordinator firmware capable of +20 dBm output power on CC2652P and CC1352P based radio modules

Hedda, 2021-11-02:
Request Z-Stack 3.x.0 coordinator firmware capable of Active mode TX at +20 dBm output power on CC2652P and CC1352P based radios.

CC2652P and CC1352P do have an integrated power amplifier capable of +20 dBm output power, but currently, applications like Zigbee2MQTT and IoBroker (as well as zigpy based applications like the ZHA integration in Home Assistant and Jeedom Zigbee Plugin) can not utilize 20 dBm amplification even if they enable it in their software since it is not yet supported in Koenkk community firmware which most CC2652P and CC1352P based Zigbee coordinator adapters/dongles use.

I understand that issue is that to be able to actually use 20 dBm output power it needs a firmware that allows it to be enabled.

PS: Maybe also a good idea to update both this README.md files with information about utilizing integrated power amplifiers?

https://github.com/Koenkk/Z-Stack-firmware/blob/master/coordinator/README.md

https://github.com/Koenkk/Z-Stack-firmware/blob/master/coordinator/Z-Stack_3.x.0/bin/README.md

Hedda, 2021-11-24:
FYI, Koenkk confirmed that the current +5 dBm confirmation is not a hardware limitation but a setting when building firmware:

Thus current power amplification confirmation if the firmware is the same with 5 dBm for both CC2652R/CC2652RB and CC2652P, therefore, you will currently not see any performance difference between any CC26x2.

Koenkk said that he will likely make a +9 dBm firmware for CC2652P so that they will still be legal to use . 

https://github.com/Koenkk/zigbee2mqtt/discussions/9744 

Configuring the firmware with +20 dBm is technically not harder but it would make it illegal to use in most countries due to international restrictions set on transmission power on the 2.4 GHz frequency band by commercial devices.

Shamshala, 2021-11-24:
Would be nice to get the feature of power settings and just implement a notice in the zigbee2mqtt that anything above 9.61dBm is illegal, instead of hardwiring it into firmware.

I get the idea and problem behind this but I just hate these babysitting stereotypes...

Hedda, 2021-11-25:
> Would be nice to get the feature of power setting anyway and just implement a notice in the zigbee2mqtt that anything above 9.61dBm is illegal, instead of hardwiring it into firmware.

+1 Though alternatively that could be solved by the community by offering several different firmware images which only difference would transmit power dBm configuration, and then offer easy to use OTW flashing of all those via Zigbee2MQTT's UI.

FYI, Koenkk's Z-Stack-firmware repository recently added JSON file with index (index.json) of available/compatible firmware for  each board and maybe that could be extended to add more versions of each firmware for each board(?):

https://github.com/Koenkk/Z-Stack-firmware/blob/master/index.json

https://github.com/Koenkk/Z-Stack-firmware/issues/310

By the way, technical terminology working is that it is "hardcoded" in the firmware (as it is not "hardwired"  the hardware).

Merwenus, 2021-12-15:
Doesn't the default firmware has a maximum of 30 devices?

Hedda, 2021-12-16:
> Doesn't the default firmware has a maximum of 30 devices?

Limitations on amount of connected children devices are not related to this specific issue request in any way. This issue request is for firmware images with a higher RF output power output amplification transmission, as in the radio sending out a stronger signal when transmitting. Changing the RF output power does not affect limitations on connected children or any other configurations.

guozi7788, 2021-12-16:
@Hedda 
We have modified the transmit power table in our ZNP project to make it configurable between 6-20dbm.

Then we will synchronize the transmit table meter to Koenkk .

guozi7788, 2021-12-16:
@Hedda 
We have modified the tx power table in our ZNP project to make it configurable between 6-20dbm.

Then we will synchronize the tx power table to Koenkk .

Hedda, 2021-12-16:
> We have modified the tx power table in our ZNP project to make it configurable between 6-20dbm.
> 
> Then we will synchronize the tx power table to Koenkk .

OK so the TX power output level will in the future be able to configurable in dBm for CC2652P via a software applications like Zigbee2MQTT, Home Assistant's ZHA integration and Jeedom's official Zigbee plugin (via zigpy-znp), and thus there should be no need for Koenkk to make separate firmware images woth different variants RF output power having fixed hardcoded dBm levels?

guozi7788, 2021-12-16:
@Hedda
The TX power can be configured through ZNP's SYS_SET_TX_POWER command.
If software applications provide users with this interface, then the user can configure the TX power of CC2652P..

Since the ZNP firmware maked by Koenkk adapts to a variety of hardware, it may be necessary to distinguish in the firmware whether the current hardware supports 20dBm.

Hedda, 2021-12-16:
FYI, Koenkk has now uploaded an experimental firmware for CC2652P/CC1352P with a default set to 9 dBm (with a max of 20 dBm):

https://github.com/Koenkk/zigbee2mqtt/files/7728605/CC1352P_CC2652P_TX_POWER.zip

He mentions in https://github.com/Koenkk/zigbee2mqtt/issues/8885 that he does not have RF measurement equipment so cannot confirm RF output power.

ruimarinho, 2021-12-16:
What is the effect of `tx_power` on https://github.com/zigpy/zigpy-znp with and without this firmware?

Hedda, 2021-12-16:
> What is the effect of `tx_power` on https://github.com/zigpy/zigpy-znp with and without this firmware?

As it says, now 9 dBm TX power by default on CC2652P and CC1352P based dongle but it is configurable up to 20 dBm TX power.

@puddly Looking at zigpy-znp configuration example it does sound like 19 dBm might be the highest can set for CC2652/CC1352?

https://github.com/zigpy/zigpy-znp/blob/dev/README.md#configuration

Note that other exiting master/stable firmware in this repo is set to only 5 dBm TX power and is not configurable via software.

ruimarinho, 2021-12-16:
That’s precisely my question - with the current stable - because I’m almost certain the software configuretion did manage to improve range in my case (almost double LQI).

Hedda, 2021-12-16:
> That’s precisely my question - with the current stable - because I’m almost certain the software configuretion did manage to improve range in my case (almost double LQI).

Placebo effect? All the other discussions and testings saying opposite. As I understand, with the current stable from the master branch any changes to transmission power output will not be applied. See:

https://github.com/Koenkk/zigbee2mqtt/issues/8885

https://github.com/Koenkk/zigbee2mqtt/discussions/8840

https://community.home-assistant.io/t/sonoff-zigbee-3-0-usb-dongle-plus-by-itead-is-based-on-texas-instruments-cc2652p-can-be-pre-ordered-for-10-99/340705/

Hedda, 2021-12-16:
@notenoughtech Do you Mat have any measuring equipment that could test TX power output on CC2652P dongle's antenna port? I know that you are one of the first to discover this issue with a CC2652P dongle not performing any better than a CC2652R dongle:

https://notenoughtech.com/home-automation/sonoff-zigbee-3-0-usb-dongle-plus/

This RF output power is measured in dBm or dBmW (decibel-milliwatts) is a unit of level used to indicate that a power level is expressed in decibels (dB) with a reference to one milliwatt (mW) so I guess it should be possible to just measure it in milliwatt?

notenoughtech, 2021-12-16:
Sadly I don't. For me the range and transfer wasn't as much of an issue.
Just pairing. It would either pair or ignore pairing requests.

I'm yet to try the latest launchpad firmware. Work took over experiments
and I'll do that over Xmas.

omerk, 2021-12-16:
@sjorge I do have the equipment (used for development/testing of all @electrolama projects) and verifying that the `transmit_power` config option does what it should is on my "things to do when I find the time" list. 

Measuring continuous TX power is very easy as TI's SmartRF Studio has the tools to put the radios into the correct mode but of course z2m does not work that way. To generate enough traffic to observe TX power, my idea is to set up a demo system and perform OTA upgrade on a device and capture ("max hold") the RF power observed on my spectrum analyzer. If anyone has a better test setup proposal, I'd be keen to hear about it.

I will say this though: There appears to be quite the confusion w.r.t to PA output power. This is not only a chip/firmware setting, the RF BOM should also match the reference design from TI depending on what power levels the hardware is targetting. TI has three reference designs for CC1352P/CC2652P, if you compare the RF balun/matching/filtering on the schematics you'll notice that they differ quite a bit:

  - LAUNCHXL-CC1352P1: 868/915 MHz up to 20 dBm, 2.4 GHz up to 5 dBm - **ignore for 2.4GHz PA use**
  - LAUNCHXL-CC1352P-2*: 868/915 MHz up to 14 dBm, 2.4 GHz up to 20 dBm - **"+20dBm"**
  - LAUNCHXL-CC1352P-4*: 433 MHz up to 14 dBm, 2.4 GHz up to 10 dBm - **"+10dBm"**

(You'll also notice that these are all CC1352P based, CC2652P is essentially the same chip as CC1352P without the sub-GHz radio and TI does not have separate CC2652P based designs for this reason)

Why have +10dBm and +20dBm? Because if you care about compliance (specifically: `ETSI EN 300 328` but other standards apply as well) you will find that max. power spectral density becomes a problem with settings over +10dBm or so (depending on what antenna will be used etc, there are a few factors that affect this figure).

Now, with regards to reports of "I increased `transmit_power` yet nothing happened/things got worse" -- welcome to the world of RF! You can't just increase power and expect things to get better as you need your radio hardware and antenna to work properly, in unison. This is where the whole "which RF BOM does this board use?" thing becomes relevant.

I don't mean to turn this into a pitch but to highlight the importance of hardware: here's a sneak peek at the RF section of zzhp, utilising 0201 caps and inductors on a super tight layout for the best possible performance: 
[image]
(ballpoint pen tip for scale) 
Unlike digital circuits, size and layout matters a lot for RF circuits especially if you're keen to push the PA to its max output power.

sjorge, 2021-12-16:
> Now, with regards to reports of "I increased `transmit_power` yet nothing happened/things got worse" -- welcome to the world of RF! You can't just increase power and expect things to get better as you need your radio hardware (including eveything else that sits between the chip and the antenna) and antenna to work properly, in unison. This is where the whole "which RF BOM does this board use?" thing becomes relevant.

That reminds me, just increasing the coordinator's power might make a device that was out of range before hear it. But that doesn't mean the device can send a reply that the coordinator can hear.

Although tweaking hardware component/s device to keep the noise they add on the receiving end as low as possible, just increasing power is not gonna make any difference for that direction.

io53, 2021-12-17:
I've only dabbled in the RF voodoo world, so I might not be correct. But I've understood that antenna is KEY to get good signals going.

Wouldn't a high quality antenna improve the LQ more than just cranking the power?
I've been planning to do some non-scientific test with this one, when/if time and motivation strikes at the same time.
[image]

Hedda, 2021-12-17:
> Sadly I don't. For me the range and transfer wasn't as much of an issue. Just pairing. It would either pair or ignore pairing requests.

@notenoughtech  I think for pairing problems it is most often not an issue of transmission power signal strength sent from the Zigbee Coordinator radio adapter + its antenna) but instead more often rather and the issue of signal reception capability of Zigbee Coordinator radio adapter + its antenna. It is because signal reception quality plays a bigger role why Zigbee coordinator adapters is so sensitive to electromagnetic interference (EMI), also called radio-frequency interference (RFI), and the reason why simply using a long USB extension cable and getting your Zigbee coordinator adapter as far as possible away from any electronic devices/appliances and their electrical cables/wires as possible will have a larger impact on LQI (Link Quality Indicator) and RSSI (Received Signal Strength Indication) values for Zigbee devices than only replacing the Zigbee Coordinator adapter with another model/type. Location and orientation of the Zigbee Coordinator radio adapter + its antenna are really the key factors there to get improved signal reception.

Arakon, 2021-12-17:
> I've only dabbled in the RF voodoo world, so I might not be correct. But I've understood that antenna is KEY to get good signals going.
>
> Wouldn't a high quality antenna improve the LQ more than just [...]

For this to work, all the devices would also need a polarised antenna like this. Otherwise, you actually lose quality. Source: I've been using these kind of antennas in the past for quadcopter flying.
What CAN help is using higher DB antennas of the same type (they're generally longer).

notenoughtech, 2021-12-17:
I'd agree with you if not for the following:

- I'm aware of the cross talk and used usb extension cord kept the
coordinator away.
- WIFI and ZigBee are on the opposite ends of the channel spectrum in my
setup
- same device would pair a device A but not B while cc2531 had no issues in
the same spot with devices A and B
- batteries with voltage over 3.1V were used (tested with multi meter)

Can't say I have the answers but at least I was logical in my approach 🤭🤭

Hedda, 2021-12-17:
BTW, recommend hobbyist enthusiasts interested in inexpensive RF testing equipment tools check out these two blog articles:

https://tinkerman.cat/post/rf-power-monitoring-tools-on-the-cheap/

https://tinkerman.cat/post/analyze-your-antennas-using-an-aai-n1201sa/

notenoughtech, 2021-12-17:
I'll check it out. Since I'm about to kick off my overly complicated ZigBee
heating... I may need it 😆😆

chiakikato, 2021-12-17:
Thanks to Koenkk and all other persons involve this issue.
Here is my initial test result.  I can see SONOFF stick working with over 5dBm performance.

With CC1352P2_CC2652P_launchpad_coordinator_20210708 firmware:   LQI 36, RSSI -73dBm
With Experiential CC1352P2_CC2652P_launchpad_coordinator_TX_POWER firmware: 
  setting of transmit power: 5    LQI 39, RSSI -72dBm    (This is tolerance with old firmware. i.e. same.)
  as above                 : 9    LQI 78, RSSI -58dBm  
  as above                 : 20   LQI 92, RSSI -52dBm  

My test/measurement method is using TI's Packet Sniffer Ver 2.18.1 with TI's original CC2531 USB stick. as a sniffer 
Sniffer located 5meter away from SONOFF coordinator dongle.

I use channel 26 for this test to minimize other 2.4GHz radio signals.
This test measured with zigbee2mqtt version 1.21.0

Hedda, 2021-12-17:
> With CC1352P2_CC2652P_launchpad_coordinator_20210708 firmware: LQI 36, RSSI -73dBm
> With Experiential CC1352P2_CC2652P_launchpad_coordinator_TX_POWER firmware:
> setting of transmit power: 5 LQI 39, [...]

Interesting, so my prediction that a 20 in transmit power would produce worse LQI and RSSI result did not hold true.

> **_My guess is also that setting the TX power to 9 or 14 (and not to 19 or 20) will produce the best LQI and RSSI for CC2652P/CC1352P based adapters. The reason for that prediction is that the higher TX power is then the more its own transmission is also interfering with its own reception capability, as transmitting higher power will produce more signal noise._**

@chiakikato Could you maybe for comparison do a quick test with 14 in transmit power using the exact same setup?

digiblur, 2021-12-17:
> @notenoughtech Do you Mat have any measuring equipment that could test TX power output on CC2652P dongle antenna port? Know that you are one of the first to discover this issue with a CC2652P dongle not performing any better than a CC2652R dongle:
> 
> NotEnoughTech's review of Sonoff CC2652P https://notenoughtech.com/home-automation/sonoff-zigbee-3-0-usb-dongle-plus/
> 
> (Mat also wrote -> https://notenoughtech.com/home-automation/testing-9-external-antennas-with-cc2531-zigbee-coordinator/ )
> 
> The RF output power is measured in dBm or dBmW (decibel-milliwatts) but is a unit of level used to indicate that a power level is expressed in decibels (dB) with a reference to one milliwatt (mW), so I guess it should be possible to just measure it in milliwatt, or?
> 
> https://www.electronicdesign.com/technologies/communications/article/21796484/understanding-wireless-range-calculations
> 
> https://www.controleng.com/articles/radio-math/

I actually found jumping from the CC2652R to the CC2652P iTead one made a difference in LQI a bit.  But to be honest, it all gets irrelevant with a decent mesh of devices which to be honest is pretty damn simple with a couple $5 smart plugs to even jump across the yard to a neighbor's house.  I am curious to see if actually increasing power causes more issues much like I've seen with adding higher gain antennas to things from creating the donut effect, raising the noise floor, to now devices being able to hear the coordinator but they are too weak to talk back to it.

zen2, 2021-12-17:
> 
> Interesting, so my prediction that a 20 in transmit power would produce worse LQI and RSSI result did not hold true.
> 
It think your prediction could be true depending on coordinator type.
I use a TI CC1352P-2 dev board and you need to sold a resistor to use an external antenna.
So by default, on this board, you use the internal antenna present on PCB and I presume that in this case your prediction could be true.
In case I can make the test by using my old CC2531 as a sniffer.

Koenkk, 2021-12-17:
Firmware 20211217 allow settings the transmit power > 5dBm (9dBm by default), ITEAD (Sonoff) confirmed this is working (using a RF analyzer). Link: https://github.com/Koenkk/Z-Stack-firmware/tree/develop/coordinator/Z-Stack_3.x.0/bin

closed 2021-12-17 by Koenkk

Merwenus, 2021-12-17:
Does update for Itead 3.0 Plus looks like the original firmware install? Or do I have to do something else too?
Read somewhere that this thing need extra care but no idea if it is only for the dev firmwares or the stable too?

Koenkk, 2021-12-17:
@Merwenus not sure what you mean, just flash and go 😄 (it of course needs to be tested for some time which I'm already doing for a day now, you can always revert to another firmware)

digiblur, 2021-12-18:
Any router firmware capable of these outputs?  @Koenkk

chiakikato, 2021-12-18:
@Hedda,  Here is the result.

I compared with SONOFF with experimental CC1352P2_CC2652P_launchpad_coordinator_TX_POWER,  SONOFF with 20210708 and zzh!
Sorry, I don't have enough skill to post the graph.   Instead, I post CSV format.  Copy and paste in Excel to see better image.

,SONOFF(new),,,SONOFF(new),,,zzh!,
Setting,LQI,RSSI(dBm),,LQI,RSSI(dBm),,LQI,RSSI(dBm)
20,94,-51,,39,-72,,,
19,94,-51,,,,,,
18,92,-52,,,,,,
17,86,-54,,,,,,
16,89,-53,,,,,,
15,89,-53,,34,-74,,,
14,78,-56,,,,,,
13,84,-55,,,,,,
12,84,-55,,34,-74,,,
11,78,-57,,34,-74,,,
10,76,-58,,42,-71,,,
9,73,-59,,39,-72,,,
8,63,-63,,39,-72,,,
7,63,-63,,28,-76,,,
6,63,-63,,31,-75,,55,-66
5,42,-71,,31,-75,,,
4,39,-72,,28,-76,,,
3,31,-75,,28,-76,,,
2,26,-77,,26,-77,,,
1,21,-79,,23,-78,,,
0,13,-82,,21,-79,,39,-72
-1,18,-80,,,,,,
-2,21,-79,,,,,,
-3,23,-78,,,,,,
-4,5,-85,,,,,,
-5,0,-88,,5,-85,,,
-6,0,-88,,,,,,
-7,0,-93,,,,,,
-8,0,-96,,,,,,
-9,0,-88,,,,,,
-10,0,-93,,0,-88,,,

> Interesting, so my prediction that a 20 in transmit power would produce worse LQI and RSSI result did not hold true.
>
>
> @chiakikato Could you maybe for comparison do a quick test with 14 in transmit [...]

tteck, 2021-12-18:
> I use channel 26 for this test to minimize other 2.4GHz radio signals.

I _thought_ channel 26 was low power?

Koenkk, 2021-12-18:
@digiblur will update router fw later

chiakikato, 2021-12-18:
> > I use channel 26 for this test to minimize other 2.4GHz radio signals.
> 
> I _thought_ channel 26 was low power?
Channel 26 is less chance of interference with WiFi signal as below link band chart. 

https://img.community.ui.com/5f743aa0-9c43-44fb-8889-afd9eccba833/comments/d1ae2c9e-9162-47e2-abcc-12e54862b200/82ccb020-c270-45fc-995c-f837ada4f9c3

codmpm, 2021-12-18:
Thanks @Hedda for pointing me to this. As there are other modules using the "RF-star CC2652P2" module, this should be applicable there also.

As a test setup I've used an innr wall socket and a 2.5dBi antenna connected to the raspberry module.

```
Zigbee2MQTT:info  2021-12-18 11:28:08: Coordinator firmware version: '{"meta":{"maintrel":1,"majorrel":2,"minorrel":7,"product":1,"revision":20210708,"transportrev":2},"type":"zStack3x0"}'
```
Gets me around 120-132 LQI.

Updated the firmware and got 162-172 LQI - nice! Obviously the "default" change of tx_power did do something.
```
Zigbee2MQTT:info  2021-12-18 11:36:05: Coordinator firmware version: '{"meta":{"maintrel":1,"majorrel":2,"minorrel":7,"product":1,"revision":20211217,"transportrev":2},"type":"zStack3x0"}'
```

But contrary to that, the `transmission_power` setting did nothing for me: 
```
Zigbee2MQTT:info  2021-12-18 11:39:00: Set transmit power to '5'
163-174 LQI

Zigbee2MQTT:info  2021-12-18 11:40:54: Set transmit power to '10'
161-173 LQI

Zigbee2MQTT:info  2021-12-18 11:41:51: Set transmit power to '20'
162-173 LQI
```

I'm on Zigbee2MQTT version 1.21.0 (commit #70891eec) - do I need to update? Or is my module missing som externa connections to get this to work?
Happy to test and help where I can.
Please point me in the right direction if I've missed something or had to test in another way.

Keep it up!
Best
Patrik

tteck, 2021-12-18:
> https://img.community.ui.com/5f743aa0-9c43-44fb-8889-afd9eccba833/comments/d1ae2c9e-9162-47e2-abcc-12e54862b200/82ccb020-c270-45fc-995c-f837ada4f9c3

Yeah, I understand that. I was referencing that channel 26 is a "low power channel" that may have skewed the findings (which are unreadable) that you reached.

chiakikato, 2021-12-19:
> Yeah, I understand that. I was referencing that channel 26 is a "low power channel" that may have skewed the findings (which are unreadable) that you reached.

@tteck  Thank you. 
I just know ch25 and 26‘s regulation. 
I’ll test again.

Gio-dot, 2021-12-19:
> @Merwenus not sure what you mean, just flash and go smile (it of course needs to be tested for some time which I'm already doing for a day now, you can always revert to another firmware)

I'm testing this firmware in both launchpad and other versions (on rf-star and e-byte chips). I noticed that no led are working on launchpad version, and on other version red light turn on when are in pairing mode. Is this as expected?

ruimarinho, 2021-12-21:
@Koenkk does this now mean setting TX power via API call will actually set the desired gain? In my case, via https://github.com/zigpy/zigpy-znp/blob/074f33f3832697828bb905d12a547b4b5bf564aa/zigpy_znp/commands/sys.py#L414-L426.

Koenkk, 2021-12-21:
@ruimarinho yes

Hedda, 2021-12-22:
> Any router firmware capable of these outputs?

FYI, submitted a separate feature request for Zigbee router firmware with remotely configurable TX power output level -> #341

Ey3scr34m, 2021-12-22:
> Firmware 20211217 allow settings the transmit power > 5dBm (9dBm by default), ITEAD (Sonoff) confirmed this is working (using a RF analyzer). Link: https://github.com/Koenkk/Z-Stack-firmware/tree/develop/coordinator/Z-Stack_3.x.0/bin

Has anything changed between CC1352P2_CC2652P_launchpad_coordinator_20211217.hex and CC1352P2_CC2652P_launchpad_coordinator_TX_POWER.hex which you posted before for testing purposes?

And do I need to configure anything in configuration.yaml or inside ZHA itself to get the +20 dBm or is it enabled by default? 

I tried:

```
zha:
  zigpy_config:
    tx_power: 19
```

But I can't see any gains on LQI.

Thanks for you work and any help is appreciated! :)

ruimarinho, 2021-12-22:
@Ey3scr34m I think you'd ought to be doing this instead:

```
zha:
  zigpy_config:
    znp_config:
      tx_power: 19
```

However, I don't see any improvement either independently of how high or low I set `tx_power` to.

Hedda, 2021-12-22:
> Has anything changed between CC1352P2_CC2652P_launchpad_coordinator_20211217.hex and CC1352P2_CC2652P_launchpad_coordinator_TX_POWER.hex which you posted before for testing purposes?

Yes and no, depends on which exact version datse you are refering to for comparision, See commit history for the changelog:

https://github.com/Koenkk/Z-Stack-firmware/commits/develop/coordinator/Z-Stack_3.x.0/CHANGELOG.md

* Z-Stack_3.x.0 20211114 firmware image is based on [Texas Instruments SimpleLink 5.30.00.56](https://www.ti.com/tool/download/SIMPLELINK-CC13XX-CC26XX-SDK).
* Z-Stack 3.x.0 20211207 firmware image is based on [Texas Instruments SimpleLink 5.30.01.01](https://www.ti.com/tool/download/SIMPLELINK-CC13XX-CC26XX-SDK)
* Z-Stack_3.x.0 20211217 firmware image is based on [Texas Instruments SimpleLink 5.30.01.01](https://www.ti.com/tool/download/SIMPLELINK-CC13XX-CC26XX-SDK) and added 20 dBm support.

Hedda, 2021-12-22:
> I don't see any improvement either independently of how high or low I set `tx_power` to.

Maybe report to https://github.com/home-assistant/core/issues and https://github.com/zigpy/zigpy-znp/issues/ for znp devs?

ruimarinho, 2021-12-22:
@Hedda yep, reported and cross-linked the issue to this thread.

chiakikato, 2021-12-22:
> @tteck Thank you. I just know ch25 and 26‘s regulation. I’ll test again.

Here is my test results.
I tested with zzh! with 20211217 firmware, SONOFF with 20210708 and 21211217 firmware with CH11 and CH26 settings.

This test result (I tested more than these) shows 
a. SONOFF 20211217 performs much power than its 20210708 and zzh! 20211217.
b. CH11 and CH26 has no difference on both 20211217 firmware.

[image]

sintei, 2021-12-23:
> @digiblur will update router fw later

@digiblur , any progress?

Hedda, 2021-12-23:
> > @digiblur will update router fw later
> 
> @digiblur , any progress?

FYI, there are some experimental router firmware versions posted on #273 that have been hardcoded to 20 dBm TX power output.

Again, also submitted a separate request for Zigbee router firmware with remotely configurable TX power output level -> #341

sintei, 2021-12-23:
> > > @digiblur will update router fw later
> > 
> > 
> > @digiblur , any progress?
> 
> FYI, there are some experimental router firmware versions posted on #273 that have been hardcoded to 20 dBm TX power output.
> 
> PS: Again, also submitted a separate request for Zigbee router firmware with remotely configurable TX power output level -> #341

I saw this but I'm new to Github so have some troubles following all threads :)
Thanks @Hedda for clarifying it to me!
I will be following #341 :)

digiblur, 2021-12-23:
> > @digiblur will update router fw later
> 
> @digiblur , any progress?

Looked yesterday and muat have missed it. Will look again later today.

tannisroot, 2021-12-27:
Does anyone maybe have a version of 20211217 that defaults to 20dbm instead of 9? Due to https://github.com/zigpy/zigpy-znp/issues/119, I can't really see any improvement in LQI with ZHA, and I would really like to try higher tx power without switching to z2m.

puddly, 2021-12-27:
#119 isn't an issue with zigpy-znp, it's just an observation that changing the TX power doesn't affect LQI (as expected).

tannisroot, 2021-12-27:
> Though it could be that the TX power setting needs to be set after the network is started, not before. I'll double check.
Thank you, I actually wanted to check if there is an issue of some sort with tx power setting in zigpy-znp by comparing LQI with `tx_power: 20` and stock 20211217 firmware and modified 20211217 firmware in which the default is 20dbm instead of 9.

tannisroot, 2021-12-27:
I would compile it myself but TI's website is "under maintenance" :(

puddly, 2021-12-27:
@tannisroot Here's what I see when cycling through the power levels on startup and sending a broadcast (repeated ten times):

[image]

My nRF52840 dongle seems to drop packets after they exceed a certain RSSI but the TX power setting definitely has an impact.

That being said, changing the TX power will only impact the link quality of packets being sent from the coordinator to the device, since the coordinator is only "shouting louder", not "listening better". This can be verified empirically with a network with a single router:

[image]

The blue line is what you see in ZHA & Z2M as the device's LQI. It's only the link quality of packets received by the coordinator, coming "from" the device (really the device's last-hop router).

I suspect that increasing the TX power may cause new devices to prefer joining directly to the coordinator, but what will happen when they try sending packets back and the coordinator can't hear them?  Also, will the increased TX power level cause broadcasts/multicasts to propagate slower due to the coordinator's re-transmissions being heard by more of the network at once, causing congestion?

Cranking up the TX power to 100 and focusing on just the LQI may not have a positive impact on network stability.

Hedda, 2022-01-03:
> > > @digiblur will update router fw later
> > 
> > 
> > @digiblur , any progress?
> 
> Looked yesterday and muat have missed it. Will look again later today.

No news on router firmware, keeping watch on https://github.com/Koenkk/Z-Stack-firmware/tree/develop/router/Z-Stack_3.x.0/bin

KameDomotics, 2022-01-23:
I'm sorry, how many db does the antenna supplied with the Sonoff plus 3.0 stick have? 3dbi?

Are there better antennas on the market to use with the CC2652 dongle?

Thanks

[5 more replies]