Koenkk/Z-Stack-firmware #324: Zigbee coordinator firmware with RTS / CTS Hardware Flow Control enabled compatible with Sonoff Zigbee 3.0 USB Dongle Plus adapter by ITead

Hedda, 2021-11-02:
Request Z-Stack 3.x.0 Zigbee coordinator firmware with RTS / CTS Hardware Flow Control enabled and compatible with the new [Sonoff Zigbee 3.0 USB Dongle Plus adapter by ITead](https://github.com/Koenkk/zigbee2mqtt/discussions/8840) which supposedly have hardware support for Hardware Flow Control as per:

https://sonoff.tech/wp-content/uploads/2021/09/Zigbee-3.0-USB-dongle-plus-firmware-flashing-1-1.docx

_2.	Enable hardware flow control and generate firmware (optional)_ section says "_2.1 Import the ZNP project of CC1352P into CCS. Open the .syscfg configuration file in the ZNP project.  Enable serial flow control in the UART option in the .syscfg configuration file._"

Configuration in the image from that document looks like tick the box for "Flow Control" under "UART" and use configuration:

UART Peripheral = Any(UART0)
TX Pin = DIO13/4 (XDS110 UART)
RC Pin = DIO12/3 (XDS110 UART)
CTS Pin = DIO19/19 (Header)
RTS Pin = DIO18/36 (Header)

[image]

[image]

This request was initially brought up as a wish in https://github.com/Koenkk/Z-Stack-firmware/issues/316 but since that issue specifically regarding Zigbee router firmware and nor Zigbee coordinator firmware I thought that I would post a seperate issue as a request for Zigbee coordinator firmware for this.

Would you be willing to compile and host firmware images that support using hardware flow control with this specific dongle?

I understand switching to hardware flow control mode on the new Sonoff Zigbee 3.0 USB Dongle Plus adapter by ITead is done manually via the physical dip-switch on the adapters board but also understand that we need a custom firmware built for it too(?).

So would need custom coordinator firmware before enabling hardware flow control in the software application like Zigbee2MQTT:

[https://www.zigbee2mqtt.io/information/configuration.html](https://www.zigbee2mqtt.io/information/configuration.html)

> `# Optional: RTS / CTS Hardware Flow Control for serial port (default: false)`
> `rtscts: true`

Hardware flow control should in theory make for more stable serial communication since the flow control is off-loaded from the application and the system CPU to the MCU, but not sure if it will really be more stable in the real world for Zigbee2MQTT. 

PS: Know I read that openHAB's Zigbee developer recommended hardware flow control for commercial use for added stability, however then we are talking about Zigbee Coordinators and Zigbee devices, not about dedicated stand-alone Zigbee routers.

Koenkk, 2021-11-02:
I prefer not to provide additional firmwares since it leads to more confusion for users and more maintenance for me **unless** it provides a significant improvement.

Hedda, 2021-11-02:
Would configuring it in firmware actually break functionality if the same firmware was flashed to another adapter without physical hardware flow control pins connected if the optional hardware flow control setting is not enabled in the application?

That is, if a user flash such firmware with Flow Control activated as per ITead's instructions to for example Tube's CC2652P2 USB Coordinator (based on RFSTAR RF-BM-2652P2) but do not enable it Zigbee2MQTT would that then not connect via SW serial?

[https://www.zigbee2mqtt.io/information/configuration.html](https://www.zigbee2mqtt.io/information/configuration.html)

> `# Optional: RTS / CTS Hardware Flow Control for serial port (default: false)`
> `rtscts: false`

Koenkk, 2021-11-02:
Not sure, can you investigate this?

Hedda, 2021-11-02:
Maybe, if you could provide compiled coordinator firmware with flow control activated so we can test it on different dongles?

If we get a test FW image of such firmware then I can at least ask for help from the communities that requested this feature.

Perhaps it could at first be tested on just the Sonoff Plus dongle which if it works should work with the DIP-switch in both positions as long as we set the correct Flow Control configuration in Zigbee2MQTT (and other applications) that matches.

Only other CC2652 dongle I currently own is a zzh (CC2652R) that is in "production" at home so cannot test on it myself now.

tteck, 2021-11-02:
> if you could provide compiled coordinator firmware with flow control activated so we can test it on different dongles?

I'd also be willing to test.

Hedda, 2021-11-16:
@guozi7788 As this issue and questions are related to your comments about pins for Hardware Flow Control in this PR -> https://github.com/Koenkk/Z-Stack-firmware/pull/329

Do you happen to have some connections that you could report this request for firmware to relevant personnel at ITead/Sonoff?

The community would like a pre-compiled firmware image with Hardware Flow Control enabled to test on the SONOFF Zigbee 3.0 USB Dongle Plus by ITead. If someone at ITead/Sonoff or the engineering company it uses could please compile provide such a pre-compiled firmware image then we would get willing testers from the community to test it and report their results.

PS: Tip is that experimental firmware images could be provided by on a GitHub repo similar to the other Sonoff Zigbee 3.0 dongle:

https://github.com/xsp1989/zigbeeFirmware/

Hedda, 2021-11-19:
FYI, got a reply email from ITead's project manager for the Sonoff Zigbee 3.0 USB Dongle Plus who unfortunately said that ITead have no plans on maintaining and distributing their own "official" firmware that supports hardware flow control on it, and continued to say that they currently only plan to provide tutorials on how users themselves can set up and generate the appropriate firmware if needed, meaning that they rely on the community to compile our custom firmware if we want it.

Again, ITead/Sonoff engineer guozi7788 did however at least confirm in https://github.com/Koenkk/Z-Stack-firmware/pull/329 which pins should be mapped to use it:

**_I have added the information you expect to know in this PR:
BSL Trigger Pin: DIO15
Auto-BSL: yes
LED: DIO_7(Not welded) Power LED(Red)
Hardware Flow Control: DIO13: TX DIO12: RC DIO19: CTS(Need DIP-switch ON) DIO18: RTS(Need DIP-switch ON)_**

Merwenus, 2021-11-30:
Great, this is the second Zigbee router I bought for them and they suck in different ways, excellent and they just don't care.

Hedda, 2021-12-01:
> Great, this is the second **Zigbee router** I bought for them and they suck in different ways, excellent and they just don't care.

Know that there is no point in using Hardware Flow Control firmware if used as a Zigbee router as Zigbee routers are not relying on any serial communication on a host while in operation. Zigbee router just needs power to work.

Hardware Flow Control firmware is only of interest for a Zigbee Coordinator to off-load flow control from CPU to MCU in a setup in a non-dedicated system where you might be running a lot of other stuff on the main System CPU, (like maybe on a Qnap or Synology NAS with a virtual machine than run Linux which in turn also runs other applications than Zigbee2MQTT). It should also be noted that at least so far Software Flow Control firmware has been proven to be stable in practically all real-world scenarios.

dkwireless, 2021-12-02:
Do we have benefits on CC2652P side with HW flow control? If yes then we should pursuit this, if no, I think there is no need...

Hedda, 2021-12-03:
My guess is that we will not know the benefits of Hardware Flow Control over Software Flow Control without long term testing. That is, make different stress-test comparisons where Hardware Flow Control versus Software Flow Control is put to their limits.

At least on paper Hardware Flow Control should theoretically create a more robust and solid connection, even if used when the system CPU and/or operating system is under very heavy load, which is when a Software Flow Control is more likely to temporarily lag/delays or even drop the connection that can cause missed messages. So if you hypothetically simply want the best theoretical connection that should be possible then Hardware Flow Control is the best, on paper.

Software Flow Control for Zigbee Coordinator adapters/controllers might perhaps make stable enough in a majority or even all of the real-world daily scenarios for home automation usage in private residents, but I understand that Hardware Flow Control is the norm standard for Zigbee in real-world commercial use such as industry, medical (hospitals and hospice or home care), utility companies (e.g. "Smart Energy" meters for electricity, gas, heating, and water utility metering), as well when targeting IoT within commercial buildings (for example lightning and HVAC heating or cooling).

Note that we know that unstable serial connection can cause issues for Z-Stack, which is why Wi-Fi based remote adapters are not recommended. See -> https://www.zigbee2mqtt.io/guide/adapters/#notes

**_Network adapters connected via WiFi might have reduced stability as the serial protocol does not have enough fault tolerance to handle packet loss or latency delays that can normally occur over WiFi connections. If cannot use a locally connected USB or UART/GPIO adapter then the recommendation is to use a remote adapter that is connected via Ethernet (wired) to avoid issues._**

So based on issues we can see over WiFi we can assume that the Z-Stack ZNP interface protocol is currently not designed with robustness enough to deal with a lot of lag/delays and temporarily dropped connections. Instead, it is designed to work with a direct connected Zigbee serial adapter. You can read more about the concept of in robustness computer science https://en.wikipedia.org/wiki/Robustness_(computer_science)

PS: Disclaimer is that I am a bit biased as find it a little hard to be subjective on this topic as professionally I worked as a server infrastructure engineer for over 20-years and there having redundancy and utilizing robust hardware components/systems for high-availability (as well as having backup systems for safe fail-overs) is one of the essential keys to achieving maximum uptime as few unplanned downtimes as possible. Otherwise, all of the robustness must be achieved by application developers by running several instances of all applications and their dependencies in high availability, which is something that usually costs more and is harder to justify the overhead for.

verjus, 2021-12-06:
I was able to enable hardware flow control in  Koenkk's firmware by making the following modifications to two files, after following the instructions provided [here](https://github.com/Koenkk/Z-Stack-firmware/blob/develop/coordinator/Z-Stack_3.x.0/COMPILE.md).

The two files are: ti_drivers_config.c and ti_drivers_config.h. They reside in the syscfg directory, after applying Koenk's patch.

Here is a diff for each file:

```
diff ti_drivers_config.h ~/syscfg.or/ti_drivers_config.h 
97,104d96
<   /* Owned by CONFIG_DISPLAY_UART as  */
< extern const uint_least8_t CONFIG_GPIO_3_CONST;
< #define CONFIG_GPIO_3 19
< 
< /* Owned by CONFIG_DISPLAY_UART as  */
< extern const uint_least8_t CONFIG_GPIO_4_CONST;
< #define CONFIG_GPIO_4 18
< 
225,226d216
<  *  CTS: DIO19
<  *  RTS: DIO18
```

and
```
diff ti_drivers_config.h ~/syscfg.or/ti_drivers_config.h 
97,104d96
<   /* Owned by CONFIG_DISPLAY_UART as  */
< extern const uint_least8_t CONFIG_GPIO_3_CONST;
< #define CONFIG_GPIO_3 19
< 
< /* Owned by CONFIG_DISPLAY_UART as  */
< extern const uint_least8_t CONFIG_GPIO_4_CONST;
< #define CONFIG_GPIO_4 18
< 
225,226d216
<  *  CTS: DIO19
<  *  RTS: DIO18
PCAXS::syscfg>diff ti_drivers_config.c ~/syscfg.or/ti_drivers_config.c
269,272c269,270
<     /* Owned by CONFIG_DISPLAY_UART as RTS */
<     GPIO_CFG_OUTPUT_INTERNAL | GPIO_CFG_OUT_STR_MED | GPIO_CFG_OUT_LOW, /* CONFIG_GPIO_4 */
<     /* Owned by CONFIG_DISPLAY_UART as CTS */
<     GPIO_CFG_INPUT_INTERNAL | GPIO_CFG_IN_INT_NONE | GPIO_CFG_PULL_DOWN_INTERNAL, /* CONFIG_GPIO_3 */
---
>     GPIO_CFG_NO_DIR, /* DIO_18 */
>     GPIO_CFG_NO_DIR, /* DIO_19 */
307,308d304
< const uint_least8_t CONFIG_GPIO_3_CONST = CONFIG_GPIO_3;
< const uint_least8_t CONFIG_GPIO_4_CONST = CONFIG_GPIO_4;
813,814c809,810
<     .ctsPin             = CONFIG_GPIO_3,
<     .rtsPin             = CONFIG_GPIO_4,
---
>     .ctsPin             = GPIO_INVALID_INDEX,
>     .rtsPin             = GPIO_INVALID_INDEX,
```

The relevant sections of each file should look like this for ti_drivers_config.c

```
void* gpioUserArgs[31];

const uint_least8_t CONFIG_GPIO_1_CONST = CONFIG_GPIO_1;
const uint_least8_t CONFIG_GPIO_2_CONST = CONFIG_GPIO_2;
const uint_least8_t CONFIG_GPIO_3_CONST = CONFIG_GPIO_3;
const uint_least8_t CONFIG_GPIO_4_CONST = CONFIG_GPIO_4;
const uint_least8_t CONFIG_RF_24GHZ_CONST = CONFIG_RF_24GHZ;
```
this
```
    GPIO_CFG_NO_DIR, /* DIO_16 */
    GPIO_CFG_NO_DIR, /* DIO_17 */
    /* Owned by CONFIG_DISPLAY_UART as RTS */
    GPIO_CFG_OUTPUT_INTERNAL | GPIO_CFG_OUT_STR_MED | GPIO_CFG_OUT_LOW, /* CONFIG_GPIO_4 */
    /* Owned by CONFIG_DISPLAY_UART as CTS */
    GPIO_CFG_INPUT_INTERNAL | GPIO_CFG_IN_INT_NONE | GPIO_CFG_PULL_DOWN_INTERNAL, /* CONFIG_GPIO_3 */
    /* Owned by CONFIG_NVS_SPI_0 as Button GPIO */
    GPIO_CFG_OUTPUT_INTERNAL | GPIO_CFG_OUT_STR_MED | GPIO_CFG_OUT_HIGH, /* CONFIG_GPIO_0 */
    GPIO_CFG_NO_DIR, /* DIO_21 */
    GPIO_CFG_NO_DIR, /* DIO_22 */
    GPIO_CFG_NO_DIR, /* DIO_23 */
    GPIO_CFG_NO_DIR, /* DIO_24 */

```
and this
```
static const UARTCC26XX_HWAttrsV2 uartCC26XXHWAttrs[CONFIG_UART_COUNT] = {
  {
    .baseAddr           = UART0_BASE,
    .intNum             = INT_UART0_COMB,
    .intPriority        = (~0),
    .swiPriority        = 0,
    .powerMngrId        = PowerCC26XX_PERIPH_UART0,
    .ringBufPtr         = uartCC26XXRingBuffer0,
    .ringBufSize        = sizeof(uartCC26XXRingBuffer0),
    .rxPin              = CONFIG_GPIO_2,
    .txPin              = CONFIG_GPIO_1,
    .ctsPin             = CONFIG_GPIO_3,
    .rtsPin             = CONFIG_GPIO_4,
    .txIntFifoThr       = UARTCC26XX_FIFO_THRESHOLD_1_8,
    .rxIntFifoThr       = UARTCC26XX_FIFO_THRESHOLD_4_8,
    .errorFxn           = NULL
  },
};
```

and for ti_drivers_config.h

```
/*
 *  ======== GPIO ========
 */
/* Owned by CONFIG_DISPLAY_UART as  */
extern const uint_least8_t CONFIG_GPIO_1_CONST;
#define CONFIG_GPIO_1 13

/* Owned by CONFIG_DISPLAY_UART as  */
extern const uint_least8_t CONFIG_GPIO_2_CONST;
#define CONFIG_GPIO_2 12

  /* Owned by CONFIG_DISPLAY_UART as  */
extern const uint_least8_t CONFIG_GPIO_3_CONST;
#define CONFIG_GPIO_3 19

/* Owned by CONFIG_DISPLAY_UART as  */
extern const uint_least8_t CONFIG_GPIO_4_CONST;
#define CONFIG_GPIO_4 18

/* Owned by /ti/drivers/RF as  */
extern const uint_least8_t CONFIG_RF_24GHZ_CONST;
// #define CONFIG_RF_24GHZ 28

```
Then rebuild the firmware.

I tested that it works with the latest dev z2m, with rtscts enabled in the configuration file:
```
advanced:
  # Optional: Baud rate speed for serial port, this can be anything firmware support but default is 115200 for Z-Stack and EZSP,
  # 38400 for Deconz, however note that some EZSP firmware need 57600.
  #baudrate: 115200
  # Optional: RTS / CTS Hardware Flow Control for serial port (default: false)
  rtscts: true
```
and of course, the with hardware switch activated. I haven't tested the firmware itself but z2m communicates with the stick.

Hope this is useful to someone. Disclaimer, I haven't tested the firmware's performance or even it's ability to pair. I just checked that the firmware allows for the serial port to be open:

tteck, 2021-12-06:
@verjus are you gonna share the firmware for testing?

Hedda, 2021-12-06:
@verjus Can you also test if Hardware Flow Control firmware works in ZHA too?  -> https://www.home-assistant.io/integrations/zha

@verjus Can you in addition as well test if Zigbee2MQTT (and ZHA) can still communicate with the stick when hardware switch is deactivated and default config (rtscts: true)? As brought up previous above, it would be very interesting to know if firmware compiled with Hardware Flow Control support can also be used with via Software Flow Control when not switching over the switch on the Sonoff Plus dongle. If that is possible then should be no need to release separate firmware for Hardware Flow Control and Software Flow Control.

> Would configuring it in firmware actually break functionality if the same firmware was flashed to another adapter without physical hardware flow control pins connected if the optional hardware flow [...]

verjus, 2021-12-07:
@Hedda, I did not have time to test extensively but from what I was able to determine, the firmware appears to require the the hardware switch be enabled and that the configuration be set to rtscts: true in order to work.

@tteck I was nervous about sharing the firmware because I did not have time to test it, but if you are willing to take a chance, I've posted it [here](https://github.com/verjus/Zigbee-3.0-USB-dongle-plus-firmware). I was able to pair and use an EcoSmart 4 button switch that worked fine. Also note that the firmware is based on simplelink_cc13xx_cc26xx_sdk_5_30_01_01, which is more recent that the one used by Koenkk, so I had to make some adjustments to his patches.

If the firmware works, hopefully Koenkk will be able to incorporate the changes in the version he maintains. It would be great if a single firmware worked for both software and hardware flow control, but it doesn't seem to be the case. Maybe he or someone else kind find a way to make it happen.

tteck, 2021-12-07:
Thanks, Koenkk actually has [firmware out based on SimpleLink SDK 5.30.00.56](https://github.com/Koenkk/Z-Stack-firmware/blob/develop/coordinator/Z-Stack_3.x.0/CHANGELOG.md#20211114)

Koenkk, 2021-12-07:
Didn't know there was a new SDK, will update soon.

Koenkk, 2021-12-07:
Done: https://github.com/Koenkk/Z-Stack-firmware/tree/develop/coordinator/Z-Stack_3.x.0/bin

qosmio, 2021-12-10:
@verjus 
I was not able to get it working using the following in my config. Had to remove the ```rtscts: true``` value for it to come up.

Firmware version is showing as 20211114, so I know I definitely flashed it correctly.
```
advanced:
    rtscts: true
```
PS For those on Raspberry/RPi, kernel version 5.10 has a bug when setting rtscts that seems to flood console with:
```
Dec 10 00:33:49 rpi4-1 kernel: [  481.403392] cp210x ttyUSB2: failed set request 0x7 status: -32
Dec 10 00:33:49 rpi4-1 kernel: [  481.560288] cp210x ttyUSB2: failed set request 0x7 status: -32
Dec 10 00:33:49 rpi4-1 kernel: [  481.715677] cp210x ttyUSB2: failed set request 0x7 status: -32
Dec 10 00:34:38 rpi4-1 kernel: [  530.079617] cp210x ttyUSB2: failed set request 0x7 status: -32
```
I updated to release branch `next` of the kernel with `rpi-update`
`UPDATE_SELF=1 SKIP_WARNING=1 BRANCH=next WANT_PI4=1 RPI_UPDATE_UNSUPPORTED=0 RPI_REBOOT=1 rpi-update`

Below is the debug output of Z2M

```
[12/10/21 01:08:00]
hass@rpi4-1 ~/z2m 55s
➤ DEBUG=* npm start

> zigbee2mqtt@1.22.1-dev start
> node index.js

winston:create-logger: Define prototype method for "error"
winston:create-logger: Define prototype method for "warn"
winston:create-logger: Define prototype method for "info"
winston:create-logger: Define prototype method for "http"
winston:create-logger: Define prototype method for "verbose"
winston:create-logger: Define prototype method for "debug"
winston:create-logger: Define prototype method for "silly"
winston:create-logger: Define prototype method for "emerg"
winston:create-logger: Define prototype method for "alert"
winston:create-logger: Define prototype method for "crit"
winston:create-logger: Define prototype method for "error"
winston:create-logger: Define prototype method for "warning"
winston:create-logger: Define prototype method for "notice"
winston:create-logger: Define prototype method for "info"
winston:create-logger: Define prototype method for "debug"
winston:file: written true false
winston:file: logged 78 info  2021-12-10 01:08:15: Logging to directory: 'data/log' filename: log.txt

winston:file: stat done: log.txt { size: 889388 }
winston:file: create stream start data/log/log.txt { flags: 'a' }
winston:file: create stream ok data/log/log.txt
winston:file: file open ok data/log/log.txt
winston:file: written true false
winston:file: written true false
  serialport/bindings loading LinuxBinding +0ms
  serialport/stream .list +0ms
winston:file: logged 889474 info  2021-12-10 01:08:15: Starting Zigbee2MQTT version 1.22.1-dev (commit #3e5a94ea)

winston:file: logged 889537 info  2021-12-10 01:08:15: Starting zigbee-herdsman (0.13.180)

  serialport/stream .list +169ms
  serialport/stream .list +167ms
  zigbee-herdsman:controller:log Starting with options '{"network":{"networkKeyDistribute":false,"networkKey":[REDACTED],"panID":49984,"extendedPanID":[221,221,221,221,221,221,221,221],"channelList":[25]},"serialPort":{"rtscts":true,"path":"/dev/serial/by-id/usb-ITead_Sonoff_Zigbee_3.0_USB_Dongle_Plus_REDACTED-if00-port0","adapter":"auto"},"databasePath":"/srv/hass/z2m/data/database.db","databaseBackupPath":"/srv/hass/z2m/data/database.db.backup","backupPath":"/srv/hass/z2m/data/coordinator_backup.json","adapter":{"disableLED":true,"concurrent":18,"delay":null}}' +0ms
  zigbee-herdsman:adapter:zStack:znp:log Opening SerialPort with /dev/serial/by-id/usb-ITead_Sonoff_Zigbee_3.0_USB_Dongle_Plus_REDACTED-if00-port0 and {"baudRate":115200,"rtscts":true,"autoOpen":false} +0ms
  serialport/stream opening path: /dev/serial/by-id/usb-ITead_Sonoff_Zigbee_3.0_USB_Dongle_Plus_REDACTED-if00-port0 +129ms
  serialport/binding-abstract open +0ms
  serialport/stream _read queueing _read for after open +2ms
  serialport/bindings/poller Creating poller +0ms
  serialport/stream opened path: /dev/serial/by-id/usb-ITead_Sonoff_Zigbee_3.0_USB_Dongle_Plus_REDACTED-if00-port0 +9ms
  serialport/stream _read reading { start: 0, toRead: 65536 } +1ms
  serialport/binding-abstract read +13ms
  zigbee-herdsman:adapter:zStack:znp:log Serialport opened +16ms
[... 83 lines cut ...]
  serialport/bindings/unixRead waiting for readable because of code: EAGAIN +1ms
  serialport/bindings/poller Polling for "readable" +6ms
  zigbee-herdsman:adapter:zStack:znp:SREQ --> SYS - ping - {"capabilities":1} +6s
  zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,0,33,1,32] +6s
  serialport/stream _write 5 bytes of data +4s
  serialport/binding-abstract write 5 bytes +4s
  serialport/bindings/unixWrite Starting write 5 bytes offset 0 bytesToWrite 5 +6s
  serialport/bindings/unixWrite write returned: wrote 5 bytes +1ms
  serialport/bindings/unixWrite Finished writing 5 bytes +0ms
  serialport/stream binding.write write finished +3ms
  zigbee-herdsman:adapter:zStack:znp:SREQ --> SYS - ping - {"capabilities":1} +6s
  zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,0,33,1,32] +6s
  serialport/stream _write 5 bytes of data +6s
  serialport/binding-abstract write 5 bytes +6s
  serialport/bindings/unixWrite Starting write 5 bytes offset 0 bytesToWrite 5 +6s
  serialport/bindings/unixWrite write returned: wrote 5 bytes +1ms
  serialport/bindings/unixWrite Finished writing 5 bytes +0ms
  serialport/stream binding.write write finished +3ms
winston:file: written true false
winston:file: written true false
winston:file: written true false
winston:file: written true false
winston:file: written true false

```

verjus, 2021-12-11:
@qosmio  I did not test the firmware on a Raspberry Pi yet (I have an RPi3+), only on an Ubuntu Desktop (will try this week-end). I am not sure what could be happening. Did you remember to activate the hardware switch on the Dongle? Are you running more than one instance of Z2M? Can you try on different hardware?

On a side note, I am trying to find out what the hardware switch does, but I didn't find a schematic of the dongle. It would be great if the firmware could detect the switch position and activate/deactivate hardware flow control accordingly. It doesn't seem simple, since the option looks hard-coded in the firmware. Does anybody know?

qosmio, 2021-12-12:
@verjus 
> @qosmio  Did you remember to activate the hardware switch on the Dongle?

I didn’t realize the dip-switch was even available until I checked the manufacturer page. Assumed it was on by design. Unscrewing 2 simple screws and there it was... Just switched it on, set the setting back and it’s working!

Just enabled it 5 minutes ago, will test to see how everything goes. Thanks!

Lefuneste83, 2021-12-15:
Hi there.

Just started Z2M using the HWC version of the FW proposed by @verjus. 

Immediately I see MUCH better stability compared with CC1352P2_CC2652P_launchpad_coordinator_20210708 version of FW which struggled to enumerate Tuya devices and could not reliably pair them at all.

With the newest HWC revision all devices come back to life immediately and the coordinator is very responsive. So far I don't have a single error or warning message.

Recipe

0/ Flashed the adapter with TI Flash Programmer 2 using @verjus HWC version. I already had flashed the ieee address in previous flashing session so did not force it this time. Made sure to validate the flash. No error whatsoever.

1/ I enabled rtscts: true in the configuration file AND switched the adapter onboard switch to ON position.

2/ Keep in mind that before restarting Z2M after you flashed the FW, you first need to purge the NVMEM of the coordinator with the script zStackEraseAllNvMem.js you can find on another github (can't remember where... Google is your friend)

node zStackEraseAllNvMem.js /dev/ttyUSB0

3/ Then you must unplug the coordinator and replug it. Otherwise Z2M will not start

4/ Then start Z2M

Good job @verjus !!

Hedda, 2021-12-15:
> On a side note, I am trying to find out what the hardware switch does, but I didn't find a schematic of the dongle. It would be great if the firmware could detect the switch position and [...]

@guozi778 wrote in https://github.com/Koenkk/Z-Stack-firmware/pull/329

**_Hardware Flow Control: DIO13: TX DIO12: RC DIO19: CTS(Need DIP-switch ON) DIO18: RTS(Need DIP-switch ON)_**

The same info is also in https://sonoff.tech/wp-content/uploads/2021/09/Zigbee-3.0-USB-dongle-plus-firmware-flashing-1-1.docx

_UART Peripheral = Any(UART0)
TX Pin = DIO13/4 (XDS110 UART)
RC Pin = DIO12/3 (XDS110 UART)
CTS Pin = DIO19/19 (Header)
RTS Pin = DIO18/36 (Header)_

verjus says he did not do an extensive test to determine if firmware with hardware flow control enable also works when dip-switch for it is disabled and application is configured to use software flow control. But if we can confirm that it not working then can assume that the dip-switch probably just is a physical switch that (presumably) physically switches TX and RX input paths (as in changing from one circuit board trace path to another) between the CP2102N USB-to-UART chip to the CC2652P chip from DIO13 (XDS110 UART TX pin) and DIO12 ( (XDS110 UART RX pin) to DIO19 (CTS Pin) and DIO18 (RTS Pin) respectively(?).

If so then that is not as user-friendly as if both paths could be enabled at the same time as then only the firmware would differ.

@guozi7788 could maybe help explain the function of the dip-switch for enabling hardware flow control more exactly in detail?

Merwenus, 2021-12-15:
> Done: https://github.com/Koenkk/Z-Stack-firmware/tree/develop/coordinator/Z-Stack_3.x.0/bin

For the record I'm a noob, and my question is does this firmware fix the 5vs20dba antenna problem?
I bought this out of curiousity but the signal rate is so low it drops devices after a while.

Hedda, 2021-12-15:
> For the record I'm a noob, and my question is does this firmware fix the 5vs20dba antenna problem?

No, and FYI; amplification power is a completely different issue not related to HW flow control in any way, so instead see -> https://github.com/Koenkk/Z-Stack-firmware/issues/323

Lefuneste83, 2021-12-15:
I am currently testing the HWC FW version of @verjus. It was first a bit fidly to find my devices. I updated to Bullseye and reinstalled the whole compilation environment and it seems to behave fine so far.

I see one issue though. When stopping the service manually, then restarting it, all cleanly waiting for the service to stop, I got this error on restart : 

```
déc. 15 19:47:45 zigbee2mqtt npm[1915]: > zigbee2mqtt@1.22.1-dev start /opt/zigbee2mqtt
déc. 15 19:47:45 zigbee2mqtt npm[1915]: > node index.js
déc. 15 19:47:52 zigbee2mqtt npm[1928]: Zigbee2MQTT:info  2021-12-15 19:47:52: Logging to console and directory: '/opt/zigbee2mqtt/data/log/2021-12-15.19-47-49' filename: log.txt
déc. 15 19:47:53 zigbee2mqtt npm[1928]: Zigbee2MQTT:info  2021-12-15 19:47:53: Starting Zigbee2MQTT version 1.22.1-dev (commit #d875b297)
déc. 15 19:47:53 zigbee2mqtt npm[1928]: Zigbee2MQTT:info  2021-12-15 19:47:53: Starting zigbee-herdsman (0.13.182)
déc. 15 19:48:15 zigbee2mqtt npm[1928]: Zigbee2MQTT:error 2021-12-15 19:48:15: Error while starting zigbee-herdsman
déc. 15 19:48:15 zigbee2mqtt npm[1928]: Zigbee2MQTT:error 2021-12-15 19:48:15: Failed to start zigbee
déc. 15 19:48:15 zigbee2mqtt npm[1928]: Zigbee2MQTT:error 2021-12-15 19:48:15: Check https://www.zigbee2mqtt.io/guide/installation/20_zigbee2mqtt-fails-to-start.html for possible solutions
déc. 15 19:48:15 zigbee2mqtt npm[1928]: Zigbee2MQTT:error 2021-12-15 19:48:15: Exiting...
déc. 15 19:48:15 zigbee2mqtt npm[1928]: Zigbee2MQTT:error 2021-12-15 19:48:15: Error: Failed to connect to the adapter (Error: SRSP - SYS - ping after 6000ms)
déc. 15 19:48:15 zigbee2mqtt npm[1928]:     at ZStackAdapter.start (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/zStackAdapter.ts:102:27)
déc. 15 19:48:15 zigbee2mqtt npm[1928]:     at Controller.start (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/controller/controller.ts:123:29)
déc. 15 19:48:15 zigbee2mqtt npm[1928]:     at Zigbee.start (/opt/zigbee2mqtt/lib/zigbee.ts:58:27)
déc. 15 19:48:15 zigbee2mqtt npm[1928]:     at Controller.start (/opt/zigbee2mqtt/lib/controller.ts:100:27)
déc. 15 19:48:15 zigbee2mqtt npm[1928]:     at start (/opt/zigbee2mqtt/index.js:97:5)
```

Which seems similar to [this post](https://github.com/Koenkk/zigbee2mqtt/issues/9865) 
Apparently the adapter does not reset itself on system restart. 

Any solution from the config line ? Or does someone have any on how to modify the FW to avoid this ?

Hedda, 2021-12-16:
Not tested but wondering if it will continue to always be possible to flash new option `--bootloader-sonoff-usb` from PR https://github.com/JelmerT/cc2538-bsl/pull/114 when first flashing "hardware flow control" firmware from software flow control mode, and then later when flashing a "software flow control" firmware from hardware flow control mode?

Hedda, 2021-12-16:
> I see one issue though. When stopping the service manually, then restarting it, all cleanly waiting for the service to stop, I got this error on restart :
> 
> ...
> 
> Which seems similar to [this post](https://github.com/Koenkk/zigbee2mqtt/issues/9865) Apparently the adapter does not reset itself on system restart.

Did you not see the same issue when using a software flow control firmware and are you using it in the same USB port as then?

guozi7788, 2021-12-16:
@Hedda    @verjus 

The dip-switch only controls the physical on/off of CTS/RTS between CP210x and CC2652P.

> It would be great if the firmware could detect the switch position and activate/deactivate hardware flow control accordingly.

This function requires CC2652 to add GPIO instead of the dip-switch on the hardware, and the firmware needs to be modified accordingly.

verjus, 2021-12-17:
@Hedda, I was able to reflash a new firmware using bsl even with the HFC switch activated and with a firmware with HFC previously enabled flashed, so it doesn't seem to interfere. Others may be able to confirm.

As for the switch, it does indeed appear to connect CTS/RTS from cp2102N to DIO19 (CTS Pin) and DIO18 (RTS Pin). I am assuming that the switch does not affect TX/RX. So where does it route DIO19 and DIO18 when it disconnects them from the CP1202N? Suppose it sets DIO19 or 18 or both to High, the firmware could read the state of the PIN at boot time to determine whether the switch is on or off and configure the uart cts/rts pin in the driver accordingly. This would mean that a single firmware would work for both software and hardware flow control. I've asked this question to Sonoff support but haven't received a response. I may be completely off on this, as I don't have any experience with the cc2652. But it would be a great feature to have.

@guozi7788 I don't understand what you mean by "This function requires CC2652 to add GPIO instead of the dip-switch on the hardware, and the firmware needs to be modified accordingly".

verjus, 2021-12-19:
For those who are interested in testing hardware flow control, I've recompiled Koenk's [latest](https://github.com/Koenkk/Z-Stack-firmware/tree/develop/coordinator/Z-Stack_3.x.0/bin) firmware with HFC enabled. Find it [here](https://github.com/verjus/Zigbee-3.0-USB-dongle-plus-firmware/blob/main/bin/znp_CC1352P_2_LAUNCHXL_tirtos_ccs_HFC_20211217.hex). I've performed some limited testing with it on a RPi 3+ in my home system and every device paired and works fine, so far.

dkwireless, 2021-12-19:
@verjus Thanks a lot. I'm testing it right now with around 30 devices.
There was no issues with previous firmware also.
Will let you know it something comes up.

Lefuneste83, 2021-12-23:
So I am testing latest version of @verjus with HFC. It is working extremely well but I have one major issue. And this is an ongoing subject that I cannot escape when testing Sonoff USB 3 dongle no matter what FW version. Every device in my network is perfectly detected and I get twice the signal strength compared with my Modkam V3 which is awesome. But some specific devices consistently refuse to be detected or repair. I have a pair of Tuya plugs namely  [TS011F_plug_3](https://www.zigbee2mqtt.io/devices/TS011F_plug_3.html#tuya-ts011f_plug_3) which no matter what I do either disappear from the network after a while or refuse to be forcefully repaired. The start of pairing handshake is present (which is already a good thing), but it then fails miserably timing out. The weird thing is that these same Tuya plugs work perfectly well when using my Modkam V3 in place of the Sonoff (same ieee address).  This is frustrating and I don't understand why it is happening. The rest of my network composed of around 15 devices is mainly Xiaomi based and is working flawlessly. If someone has any clue of what is going on I'd be interested to know.

Error message when already paired device loses connectivity : 
`Zigbee2MQTT:warn  2021-12-23 22:33:05: Failed to ping 'Prise Tuya 02 Chambre' (attempt 1/1, Read 0x540f57fffe74bfe5/1 genBasic(["zclVersion"], {"timeout":10000,"disableResponse":false,"disableRecovery":true,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Data request failed with error: 'No network route' (205)))`

Lefuneste83, 2021-12-23:
Also as stated previously, the Sonoff coordinator needs to be unpluged from the USB plug in case I restart Z2M service. otherwise I get : 
```
> zigbee2mqtt@1.22.1-dev start /opt/zigbee2mqtt
> node index.js

Zigbee2MQTT:info  2021-12-23 22:31:28: Logging to console and directory: '/opt/zigbee2mqtt/data/log/2021-12-23.22-31-25' filename: log.txt
Zigbee2MQTT:info  2021-12-23 22:31:28: Starting Zigbee2MQTT version 1.22.1-dev (commit #d875b297)
Zigbee2MQTT:info  2021-12-23 22:31:28: Starting zigbee-herdsman (0.13.182)
Zigbee2MQTT:error 2021-12-23 22:31:49: Error while starting zigbee-herdsman
Zigbee2MQTT:error 2021-12-23 22:31:49: Failed to start zigbee
Zigbee2MQTT:error 2021-12-23 22:31:49: Check https://www.zigbee2mqtt.io/guide/installation/20_zigbee2mqtt-fails-to-start.html for possible solutions
Zigbee2MQTT:error 2021-12-23 22:31:49: Exiting...
Zigbee2MQTT:error 2021-12-23 22:31:49: Error: Failed to connect to the adapter (Error: SRSP - SYS - ping after 6000ms)
    at ZStackAdapter.start (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/zStackAdapter.ts:102:27)
    at Controller.start (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/controller/controller.ts:123:29)
    at Zigbee.start (/opt/zigbee2mqtt/lib/zigbee.ts:58:27)
    at Controller.start (/opt/zigbee2mqtt/lib/controller.ts:100:27)
    at start (/opt/zigbee2mqtt/index.js:97:5)
npm ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! zigbee2mqtt@1.22.1-dev start: `node index.js`
npm ERR! Exit status 1
npm ERR! 
npm ERR! Failed at the zigbee2mqtt@1.22.1-dev start script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.

npm ERR! A complete log of this run can be found in:
npm ERR!     /home/pi/.npm/_logs/2021-12-23T21_32_20_032Z-debug.log
```

Lefuneste83, 2021-12-24:
I have posted a similar issue for the Modkam V4 coordinator which also loses connectivity to a specific class of devices.
https://github.com/egony/cc2652p_E72-2G4M20S1E/issues/32

Lefuneste83, 2021-12-24:
Ok some updates on @verjus version with HFC.
I was so frustrated by this Tuya pairing issue that I scraped my data folder and restarted with a blank page. On restart I saw I had some pan_id error message when provisioning the network. I tried various channels and moved from 11 to 15.
I then could restart Z2M and all devices are now pairing flawlessly !
Total mystery to me but there was some issues with the existing devices db that would block detection and pairing for pre-existing Tuya devices only. Xiaomi devices did not have this issue at all.

So I can say that @verjus HFC version is wonderful. The signal strength on the Sonoff is by far the strongest of all my coordinators including Modkam 4, and HFC seems to add stability to the network. I will continue the tests.

The only remaining issue is the need to unplug the coordinator in case of Z2M service restart. I hope we'll figure out where this is coming from.

dkwireless, 2021-12-24:
So far so good.
Coordinator is super stable. I was already on channel 15. Started with new database and new mqtt topic. Only issues I had are:
1. When I try to pair new device coordinator z2m restarts. I can get device recognized, and everything works fine after that.
2. I had couple of devices dropping out when they loose power. I have removed them, restarted z2m and added them again. There was no issues after.
3. I'm not sure that coordinator is working with 20 db output. I can see no difference when I add transmit_power: 20 option. I have tried adding it both to experimental section and in root.
All in all, signal is good and network is very stable. I know that we should somehow test if hw flow control makes difference, but at least after these test we will know it this configuration can be recommended.

Lefuneste83, 2021-12-24:
Ok so I've reinstalled the whole development framework with latest NodeJS. It "seems" to improve the stability of the build.

I have rebuilt Z2M from scratch, started fresh, reflashed the coordinator, purged NVMEM, then repaired all devices.

They all paired well, except that for most end devices I had to pair them twice : a first time to get the device recognized (I waited until interview was totally finished), then repaired it again to get all the payload published, otherwise I would only get partial data (no battery level for instance). For the first device I repaired I waited 2 hours to see if I was getting complete data, but no it would stay partial. 
Router devices such as plugs did not need to be repaired twice.
Overall pairing works very well for all brands of devices, I only got 1 failed initial interview out of 15+ devices. and it worked the second time.

Now everything is running smoothly except that I must be physically present in case I need to restart Z2M in order to unplug the coordinator.

Bolukan, 2021-12-25:
> For those who are interested in testing hardware flow control, I've recompiled Koenk's [latest](https://github.com/Koenkk/Z-Stack-firmware/tree/develop/coordinator/Z-Stack_3.x.0/bin) firmware with [...]

I test the firmware with 22 devices. Half of the time the pairing of the [IKEA E1743 switch](https://www.zigbee2mqtt.io/devices/E1743.html) devices (7 in total) failed. It left the network before pairing was finished. Putting the device at least 10cm from the antenna seemed to help. So far it functions without any problems.

verjus, 2021-12-27:
I have also been testing this firmware for a week now and found the combination of the Sonoff and @Koenkk 's firmware to be a great improvement over the cc2538 it replaced. I also had a few initial issues that lead me to repair everything (Hue, Ecosmart and Xiaomi nodes), but that was surprisingly smooth (maybe, I should have transferred the old [IEEE](https://www.zigbee2mqtt.io/guide/adapters/flashing/copy_ieeaddr.html)). 

To be honest, I have no idea if HFC adds anything, but it's available so why not play with it? Many thanks to @Koenkk for creating and maintaining such as wonderful home automation system.

tanus10, 2021-12-31:
> Now everything is running smoothly except that I must be physically present in case I need to restart Z2M in order to unplug the coordinator.

I can confirm this. I had to physically disconnect HFC-enabled dongle to restart Z2M.

FYI, I have tried various software based usb port reset methods without any success.
Maybe some extra codes are needed to be issued to the Sonoff dongle to stop HW FC.

dkwireless, 2022-01-16:
Still working rock solid. Should we make it official?

tanus10, 2022-01-16:
> Still working rock solid. Should we make it official?

Have you tried to restart Z2M?
It seems it is not possible without physically taking out the dongle and re-insert back every time you re-run Z2M. The wanrning should be included in the official notice.
That's kind of a deal breaker for remote-access management.

dkwireless, 2022-01-16:
Yes, my bad. There is a problem on restart. I've been running it on RPI4 on a UPS continuously...

closed 2022-02-23 by github-actions[bot]

dkwireless, 2022-02-25:
Any chance fixing that reboot problem or should we go back to software flow control? Everything else is working perfectly.

verjus, 2022-03-06:
Are you sure that the reboot problem is connected to the HFC? All I did was enabling it in the configuration file created by Koenkk. If the problem is related, it would be due to some kind of side effect that would not be trivial to investigate. The issue could also be related or addressed in the Z2M application.

joskfg, 2022-03-11:
I tried the firmware but after flash it, add the flg in the advanced section and turned on the switch on the board, I'm getting and error when starting zigbee2mqtt, it could not connect and says something like: RSPS --> SYS ping 6000 timeout

@verjus any clue about this?

By other hand, every time that I restart, I need to repair all the devices. It says the panId is already in use. I tried to remove the antenna and it started but with 0 devices. Do you have some solution?

verjus, 2022-03-11:
@joskfg Do these errors occur when you use the non-modified firmware? It doesn't seem to be related to HFC. Some good information about your error may be found [here](https://www.zigbee2mqtt.io/guide/installation/20_zigbee2mqtt-fails-to-start.html).  This [issue](https://github.com/Koenkk/zigbee2mqtt/issues/2893) may help too.

joskfg, 2022-03-13:
> @joskfg Do these errors occur when you use the non-modified firmware? It doesn't seem to be related to HFC. Some good information about your error may be found [...]

The RSPS --> SYS ping 6000 timeout error only happens with the HFC firmware. The other issue about restarts and repairs happens always (well i didn't tried with your firmware because it never starts)

joskfg, 2022-03-13:
@verjus I finally fixed it!. I was using an old zigbee2mqtt version. Seems the home assistant addon repository changed the location and I didn't receive any update from a long time ago.

Not all is working as expected with HFC and no more repairing issues in restart. 

The only issue is the decribed in this issue. If you restart zigbee2mqtt it shows the "RSPS --> SYS ping 6000 timeout error". If you replug the USB it works again.

ivdimitro, 2022-06-06:
Hi, now that you had 3 more months for testing - does the HFC makes the things better or worse or no change at all? 
I just bought the Sonof dongle and wondering which firmware to flash :)

Hedda, 2022-06-06:
No noticable change so while still good to have both for reference I would recommend keeping software flow control as default. Hardware flow control is a niche feature that would probably be most useful when combined with low-end computer like the first generations of Raspberry Pi and Raspberry Pi Zero.

Best would probably also be to sell different hardware variants not to confuse end users so that can be sold preflashed and could only be updated with one type of firmware. So manufacturers should instead of making optional via a switch make tqo defferent models, one with hardware flow control and one with software flow control.

verjus, 2022-10-12:
For those interested, I've re-compiled the latest version of @Koenkk 's firmware with HFC enabled. Tested it on my home system for a few days and it seems to work. As before, it is to be considered as experimental, although the differences between the original and this version are minimal (it just enables HFC). You can find it [here](https://github.com/verjus/Zigbee-3.0-USB-dongle-plus-firmware/tree/main/bin_20220928).

Bolukan, 2022-10-12:
I followed this discussion for a few months but got lost between the trees of the forest: [I applied the version of december 2021](https://github.com/Koenkk/Z-Stack-firmware/issues/324#issuecomment-1000999801) . Do I understand it correctly that applying the standard version (not yours) and mark the "RTS / CTS" in the Serial Settings menu will do the same? If not, what are the differences?

verjus, 2022-10-14:
@Bolukan, yes the only difference between Koenkk's original firmware and the version I re-compiled is that hardware flow control is enabled. This means that 2 additional wires are used to perform RTS/CTS flow control instead of the CPU. These wires are enabled when you open up the dongle and activate the switch. In their default position, they are simply not connected and the flow control is performed by the CPU. The only advantage is that, if the CPU is not powerful, HFC reduces the burden on the CPU and presents an advantage. For a powerful CPU (e.g. Raspberry PI 3 or 4), the additional burden associated to software flow control is negligible. Note that you also need to configure rtscts: false in the z2m configuration file, to switch to SFC. The version referenced in my last post is simply enabling HFC in the latest firmware posted by Koenkk.

Bolukan, 2022-10-15:
@verjus, thx for replying. My view is the CP2102 lowers the burden for the CC2652P not the PI. Your explanation is different. Could you explain more why the PI favors of HFC?
Good for all readers: you have to open the stick once and apply the hardware switch to enable the CP2102 chip. Second step is you have to instruct the CC2652P (CPU of the stick) to send the commands to the pins of the CP2102 which can be done with RTS/CTS configuration switch or just applying the compilation of verjus.

verjus, 2022-10-15:
@Bolukan, you are right the flow control pertains to the UART connection between the MCU and CP2102 USB-to-UART bridge. Therefore it is selectable  by the application (enabled/disabled in z2m configuration "rtscts: true/false"), but the software flow control must be performed in the MCU by the firmware. The additional overhead will indeed be on the CC2652P MCU, not the USB host. I will edit my post above.

Veldkornet, 2022-10-16:
> For those interested, I've re-compiled the latest version of @Koenkk 's firmware with HFC enabled. Tested it on my home system for a few days and it seems to work. As before, it is to be considered [...]

Can I just flash this with the SmartRF Flash Programmer? As it keeps giving me the following error:
```
>Initiate access to target: COM3 using 2-pin cJTAG.
>Reading file: C:/Users/veldkornet/Downloads/SONOFF/znp_CC1352P_2_LAUNCHXL_tirtos_ticlang_HFC.hex.
>Unknown record type: 10.
>Reset target ...
>Reset of target successful.
```

Veldkornet, 2022-12-23:
Anyone have a hardware flow control, version of **20221214** ?

verjus, 2022-12-23:
@Veldkornet I will try to recompile the latest version in the coming days.

[24 more replies]