Koenkk/Z-Stack-firmware #326: Zigbee router with ZZH

Koterak, 2021-11-03:
I tried to flash ZZH with router firmware. Everything went fine. But after I tried to pair router with coordinator nothing happened. Also nothing in logs. I tried to flash my new sonoff zigbee 3.0 usb with router and ZZH with coordinator. This time everything is working fine. There is any known issue with zzh as router?

Hedda, 2021-11-19:
It should work. Are you sure that you flashed CC2652R_*.zip to the zzh and not CC2652RB_*.zip

I read many people make the mistake flashing the wrong firmware file because the names are so similar for "CC2652R" verses "CC2652RB". 

https://github.com/Koenkk/Z-Stack-firmware/blob/master/coordinator/Z-Stack_3.x.0/bin/README.md

Koterak, 2021-11-20:
Managed to get it working, but the range is very low. Connected from distance about 2 meters max, moved away to longer distance cause errors, lags and unavability

Hedda, 2021-11-22:
Close then, but a tip is you really need to use a long USB extension cable with Zigbee adapters to get them just a little bit away from anything that could cause electromagnetic interference. Even power supplies and cables or wires can generate interfering EMF.

### General tips on improving Zigbee network range

1. Adding more Zigbee routers between devices far away and the next closest router or your Zigbee coordinator and routers. Zigbee network topology uses a "mesh network" design which means that each device that acts as a Zigbee router will extend the total range and coverage of your Zigbee network as a whole. The solution is to distribute more Zigbee routers in areas with poor reception. Note that while there are exceptions, understand that almost all permanently powered devices will serve as a Zigbee router. Thus adding more permanently powered Zigbee devices will allow greater range better coverage. There are also dedicated Zigbee routers which you can find by doing a community search for "Zigbee signal repeater" or "Zigbee range extender") (an example being IKEA Tradfri Signal Repeater). Devices that can not act as routers are typically battery-operated and known as "end devices". There are some exceptions as some end devices (e.g. Xiaomi/Aqara devices) have issues connecting through routers that are not from the same manufacturer.
2. Use a long USB extension cable with your Zigbee coordinator adapter. This not only enables you to position the Zigbee coordinator adapter for better signal quality, but it also allows you to locate the Zigbee coordinator further away from Wi-Fi access-points/routers or other sources of 2.4GHz signals to avoid signal interference. Optimally in the best area in your home depending on your building's floorplan. Ideally, you want to place the Zigbee coordinator somewhere in the middle of your house/apartment. Building materials influence signal quality, for example, dense/thick concrete, bricks, masonry, etc. dampen all wireless signals. Place the Zigbee coordinator with some distance from walls, ceilings and floors. Also, try different orientations of your Zigbee antenna (or your whole Zigbee coordinator adapter if it has an internal antenna). Some Antennas have a stronger signal in a certain direction. Simply changing orientation can improve signal quality already. Note: A bad USB extension cable may lead to connection issues between the system and the Zigbee coordinator adapter, symptoms of this are disconnection messages.
3. Zigbee and Wi-Fi can operate on various channels in the 2.4GHz spectrum. A busy Wi-Fi access-points/routers that are operating in the same frequency range (overlapping channels) as your Zigbee coordinator will drown out the Zigbee traffic, especially if they are located close to each other. To avoid interference between Zigbee and Wi-Fi try to choose channels without overlap. Check the channel your Wi-Fi access-points/routers are using (either by checking on the router's web interface or using a Wi-Fi analyzer app). Changing the channel of the Zigbee network usually requires reforming it and re-join/re-pair all of your Zigbee devices. It is therefor typically it's a lot easier to just change the channel used by your Wi-Fi router and/or access-points. See [this article for Wi-Fi and Zigbee channels coexistance](https://www.metageek.com/training/resources/zigbee-wifi-coexistence.html) to avoid using overlapping frequency ranges.
4. Check if updating the firmware on your Zigbee coordinator adapter and your Zigbee end devices is possible. Note that depending on your hardware the latest Zigbee coordinator firmware might not always be the one recommended by the community so it is advised to ask before upgrading.
5. If your Zigbee coordinator adapter has a removable antenna (e.g., with an SMA-connector) then you have the option of replacing it with a high-gain antenna. Note that antennas with higher gain usually have directionality: You might have better reception on the same floor, but reception across floors might suffer. In addition, you also have the option to use an antenna extension cable if needed (usually using just a USB extension cable for your Zigbee coordinator adapter is the better alternative). This should really only be needed if you are trying to cover a long distance, like to another building or very dense/thick walls, ceilings and floors.
6. Buy more powerful Zigbee radio hardware with a better radio range, preferably with an external antenna and based on a newer chip that offers up-to-date firmware. If you are not only experimenting with Zigbee but want a permanently stable and healthy Zigbee mesh network with potentially many devices then you should consider upgrading to a more powerful and newer Zigbee coordinator USB adapter or Ethernet to serial gateway/bridge. Generally, Zigbee adapters with an external antenna will have a better range and offer you more flexibility, therefore you will also want to avoid buying an internal Zigbee adapter unless it has a port for an external antenna.

closed 2021-11-22 by Koterak