Koenkk/Z-Stack-firmware #331: Z-Stack_Home_1.2 (default) aqara issues

xsaren11, 2021-12-01:
Hello,

I would like to report some issues with my Aqara devices.
Yesterday I downloaded the newest software for Z-Stack_Home_1.2 (default) CC2531 but now I have more troubles then before

Here is an example.

Aqara button now do not posting action state
[image]

Aqara door sensor is not supported (old version and new version)
[image]

Best Regards
Krzysztof

pfischbach, 2021-12-01:
After installing this release I can not pair IKEA E1524 remote similar to the Aqara obove. Works with last version.
error 2021-12-01 16:46:31: Failed to interview '0x000b57fffeaf6075', device has not successfully been paired
info  2021-12-01 16:46:31: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x000b57fffeaf6075","ieee_address":"0x000b57fffeaf6075","status":"failed"},"type":"device_interview"}'
info  2021-12-01 16:46:31: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"interview_failed","meta":{"friendly_name":"0x000b57fffeaf6075"},"type":"pairing"}'
info  2021-12-01 16:46:59: Zigbee: allowing new devices to join.

xsaren11, 2021-12-02:
With new version of Zigbee2MQTT everything works fine

phretor, 2021-12-27:
@xsaren11 edge or stable? And what firmware version?

xsaren11, 2021-12-27:
This
one coordinator/Z-Stack_Home_1.2/bin/default/CC2531_DEFAULT_20211115.zip
from master branch

closed 2022-02-04 by github-actions[bot]